zephyrproject-rtos/zephyr #50404: Intel CAVS: tests/subsys/logging/log_immediate failed.

shuai1x, 2022-09-20:
**Describe the bug**
On intel_adsp_cavs15/18/25 platform, tests/subsys/logging/log_immediate failed.

And I have done the following two methods to debug it: 
(1) Add extra delay between test functions, it will pass stably.
(2) Use HDA logging, it can't pass.
     Boards overlay:
```
CONFIG_DMA=y
CONFIG_ASSERT=y
CONFIG_LOG=y
CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_LOG_BACKEND_ADSP=n
CONFIG_LOG_BACKEND_ADSP_HDA=y
CONFIG_LOG_BACKEND_ADSP_HDA_SIZE=8192
CONFIG_LOG_BACKEND_ADSP_HDA_PADDING=y
CONFIG_LOG_BACKEND_ADSP_HDA_CAVSTOOL=y
CONFIG_LOG_BACKEND_ADSP_HDA_FLUSH_TIME=100
CONFIG_XTENSA_ENABLE_BACKTRACE=n
CONFIG_LOG_PRINTK=y
```

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W --hardware-map /home/ztest/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/subsys/logging/log_immediate --no-skipped-report -vv
2. See error

**Logs and console output**
Use HDA logging:
```
DEBUG   - DEVICE: [00:00:00.956,000] <inf> test: data:
DEBUG   - DEVICE: 00 00 00 00 00 00 00 00  00 00 00 00 90 f4 01 be |........ ........
DEBUG   - DEVICE: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
DEBUG   - DEVICE: 00 00 00 00 00 00 00 00                          |........
DEBUG   - DEVICE: [00:00:00.970,000] <inf> test: test string printed 1 2 0x9e026000
DEBUG   - DEVICE: [00:00:00.970,000] <inf> test: data:
DEBUG   - DEVICE: 00 00 00 00 00 00 00 00                          |........
DEBUG   - DEVICE: [00:00:00.972,000] <inf> test: test string printed 1 2 0x9e026088
DEBUG   - DEVICE: [00:00:00.973,000] <inf> test: data:
DEBUG   - DEVICE: 00 00 00 00 00 00 00 00  c8 0b 01 be 01 00 00 00 |........ ........
DEBUG   - DEVICE: [00:00:00.976,000] <inf> test: test string printed 1 2 0x9e026110
DEBUG   - DEVICE: [00:00:00.976,000] <inf> test: data:
DEBUG   - DEVICE: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
DEBUG   - DEVICE: 00 00 00 00 00 00 00 00                          |........
DEBUG   - DEVICE: [00:00:00.978,000] <inf> test: test string printed 1 2 0x9e026198
DEBUG   - DEVICE: [00:00:00.978,000] <inf> test: data:
DEBUG   - DEVICE: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ .....[00:00:00.981,000] <inf> test: test string printed 1 2 0x9e026220
DEBUG   - DEVICE: [00:00:00.982,000] <inf> test: data:
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
DEBUG   - Waiting for device intel_adsp_cavs25 to become available
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain - SDK 15.0

henrikbrixandersen, 2022-09-20:
This seems to be a general problem with that platform. See e.g. #50403, #50402, or #50401.

shuai1x, 2022-09-20:
> This seems to be a general problem with that platform. See e.g. #50403, #50402, or #50401.

No, it is not exactly like https://github.com/zephyrproject-rtos/zephyr/pull/50401, https://github.com/zephyrproject-rtos/zephyr/pull/50402, and https://github.com/zephyrproject-rtos/zephyr/pull/50403. Because these three cases will pass after we use the HDA logging. But this case can't pass after we use it.

teburd, 2022-09-20:
The flush timer seems to cause problems here in some way, changing the flush time to 10000 the test passes for me.

Will spend some time investigating

teburd, 2022-09-21:
Seems to be a missing IPC done return

Which can be seen here

https://github.com/teburd/zephyr/tree/hda_missing_ipc_return

I'm now wondering if this is the actual cause of a few things, like sometimes spinning too long in the timer isr when flushing which I've seen before.

teburd, 2022-09-26:
This should be retried with the defaults and #50623 applied

shuai1x, 2022-09-27:
> This should be retried with the defaults and #50623 applied

Yes, I have tested the PR https://github.com/zephyrproject-rtos/zephyr/pull/50623 with the defaults on CAVS platform five times. And it would pass stably, so this PR will fix this issue.

henrikbrixandersen, 2022-10-04:
Closing as fixed by #50623. Please reopen if that is not the case.

closed 2022-10-04 by henrikbrixandersen