zephyrproject-rtos/zephyr #50413: [backport v3.1-branch] backport of #50230 failed

zephyrbot, 2022-09-20:
This issue was created automatically because the backport of #50230 to `v3.1-branch` failed.

The backport to `v3.1-branch` failed:
```
The process '/usr/bin/git' failed with exit code 1
```
To backport manually, run these commands in your terminal:
```bash
# Fetch latest updates from GitHub
git fetch
# Create a new working tree
git worktree add .worktrees/backport-v3.1-branch v3.1-branch
# Navigate to the new working tree
cd .worktrees/backport-v3.1-branch
# Create a new branch
git switch --create backport-50230-to-v3.1-branch
# Cherry-pick the merged commits of this pull request and resolve the conflicts
git cherry-pick -x e1cb0845b49b6a4100c9e5558d37667b92f0d000~2..e1cb0845b49b6a4100c9e5558d37667b92f0d000
# Push it to GitHub
git push --set-upstream origin backport-50230-to-v3.1-branch
# Go back to the original working tree
cd ../..
# Delete the working tree
git worktree remove .worktrees/backport-v3.1-branch
```
Then, create a pull request where the `base` branch is `v3.1-branch` and the `compare`/`head` branch is `backport-50230-to-v3.1-branch`.

erwango, 2022-09-20:
This was expected as code base changed in V3.2. I'll need to provide a specific patch.

closed 2022-12-05 by github-actions[bot]

erwango, 2022-12-05:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/50484