zephyrproject-rtos/zephyr #50426: STM32: using SPI after STOP2 sleep causes application to hang

sense-Jo, 2022-09-20:
**Describe the bug**
After an application on STM32WL accessed STOP2 sleep mode, at the next access to an SPI device the application hangs.
When connecting a debugger, one always catches the application in a loop in `spi_ll_stm32.c` (see screenshots below).

The problem does not occur when STOP2 sleep mode is disabled, by either calling `	pm_policy_state_lock_get(PM_STATE_SUSPEND_TO_IDLE, 3);` (before any sleep), or by removing the declaration of 
`stop2` from `&cpu0 { cpu-power-states = <…> };` in the boards .dts file.

I'm not sure yet whether this issue is specific to STM32WL, or a general issue affecting all STM32 devices with Zephyr.

**To Reproduce**
I think it's not possible to reproduce without an SPI device connected, at least this failed for me.
I can reliably reproduce with a LIS2DW12 accelerometer to a NUCLEO-WL55JC STM32WL Nucleo-64 board over the Arduino SPI, [please see this repo](https://github.com/sense-Jo/SPI) for further information.

**Expected behavior**
SPI access should work after STOP2 sleep as before.

**Impact**
Requires us to disable STOP2 mode for our power-sensitive application, decreasing battery life.

**Logs and console output**
[image]
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04 LTS
 - Zephyr SDK 0.15.0, 
 - Zephyr 8ac135d7a9be2b5e77183ab1a58b5aee1a4ce428
 - STM32-HAL 5b1610c3cf46589d50e2634ba240b7b38bc59c53 (tried 4f26e65070600d946cf1055d791862b12a929166 as well)

**Additional context**
I suspect the reinitialization of SPI after STOP2-mode, which from our experience is required in STM32Cube, is not done properly, or at all, in Zephyr.

erwango, 2022-09-20:
> I suspect the reinitialization of SPI after STOP2-mode, which from our experience is required in STM32Cube, is not done properly, or at all, in Zephyr.

I confirm this is not done. This has to be implemented. Not a bug from our perspective. As always contributions are welcome. Otherwise we'll try to implement based on priority calls.

r2r0, 2022-10-04:
@erwango Is it specific problem of STM32WL?

sense-Jo, 2022-10-05:
I could now solve it by re-initializing and re-configuring the ADC before it is used again, with the help of Device Runtime Power Management. However, I'm not sure whether this is the proper way to do it, and I didn't test it intensively either.
Especially, I'm unsure about whether it would work with DMA as well. I did test with `CONFIG_SPI_STM32_DMA=y`, but is this enough?
Nevertheless, one might want to have a look at [this branch of my fork](https://github.com/sense-Jo/zephyr/tree/fix_stm32_spi_pm).

sense-Jo, 2023-01-23:
@erwango I guess this issue could be listed in #19755 under `Peripheral low power modes`?

erwango, 2023-01-23:
> @erwango I guess this issue could be listed in #19755 under `Peripheral low power modes`?

Ok, but I'm closing current one as we don't need 2 tickets to track a enhancement request.

closed 2023-01-23 by erwango