zephyrproject-rtos/zephyr #50440: Older stm32 (nucleo) boards without 8MHz ST-LINK clock do not work

thedjnK, 2022-09-20:
**Describe the bug**
Older STM32 nucleo board, C-01, do not work with the default zephyr boards files, this is due to the default using an 8MHz clock provided by the ST-LINK chip, which on the C-01 boards is not connected. A possible fix is to have board revisions whereby building for e.g. `nucleo_f401re@C-01` will build using the internal 16MHz clock source.

**To Reproduce**
Build for any C-01 nucleo board and flash

**Expected behavior**
The board to work

**Impact**
Annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.14.2
 - Commit SHA or Version used: 11baf56948c28ee039b9901951c0d49769d01ceb

erwango, 2022-09-21:
Moving this to "Feature Request" as it's coming down to supporting a new board revision.

FRASTM, 2022-09-21:
On the MB1136 board schematic, the reason is probably that MCO signal is open (SB16 and/or SB50 not present on the PCB)
A possible harwdare WA is to close solder bridge.

FRASTM, 2023-01-11:
The [board_porting](https://github.com/zephyrproject-rtos/zephyr/blob/main/doc/hardware/porting/board_porting.rst) does not explicitely mention a format like "C01"
-->  to accept this, the  ./cmake/modules/extensions.cmake  has to be modified to accept a BOARD_REV_FORMAT  == LETTER_AND_NUMBER