zephyrproject-rtos/zephyr #50452: mec172xevb_assy6906: The testcase tests/lib/cmsis_dsp/matrix failed to run.

Kunli123, 2022-09-21:
**Describe the bug**
 The testcase tests/lib/cmsis_dsp/matrix failed to run on the acrn_ehl_crb.

**To Reproduce**
twister -p mec172xevb_assy6906 --device-testing --device-serial /dev/ttyUSB0  -T tests/lib/cmsis_dsp/matrix/ -vv

**Logs and console output**

```

*** Booting Zephyr OS build v3.2.0-rc2-92-g6030b85f71ce  ***
Running TESTSUITE matrix_unary_f64
===================================================================
START - test_arm_mat_cholesky_f64
*** Booting Zephyr OS build v3.2.0-rc2-92-g6030b85f71ce  ***
Running TESTSUITE matrix_unary_f64
===================================================================
START - test_arm_mat_cholesky_f64
PASS - test_arm_mat_cholesky_f64 in 0.055 seconds
===================================================================
START - test_arm_mat_inverse_f64
PASS - test_arm_mat_inverse_f64 in 0.238 seconds
===================================================================
START - test_arm_mat_solve_lower_triangular_f64
Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/matrix/src/unary_f64.c:411: matrix_unary_f64_test_arm_mat_solve_lower_triangular_f64: (output is NULL)
buffer allocation failed
FAIL - test_arm_mat_solve_lower_triangular_f64 in 0.017 seconds
===================================================================
START - test_arm_mat_solve_upper_triangular_f64
Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/matrix/src/unary_f64.c:331: matrix_unary_f64_test_arm_mat_solve_upper_triangular_f64: (tmp2 is NULL)
buffer allocation failed
FAIL - test_arm_mat_solve_upper_triangular_f64 in 0.017 seconds
===================================================================
START - test_op1_arm_mat_trans_f64
Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/matrix/src/unary_f64.c:126: test_op1: (tmp1 is NULL)
buffer allocation failed
FAIL - test_op1_arm_mat_trans_f64 in 0.013 seconds
===================================================================
START - test_op2_arm_mat_sub_f64
Assertion failed at WEST_TOPDIR/zephyr/tests/lib/cmsis_dsp/matrix/src/unary_f64.c:44: test_op2: (tmp1 is NULL)
buffer allocation failed
FAIL - test_op2_arm_mat_sub_f64 in 0.013 seconds
===================================================================
TESTSUITE matrix_unary_f64 failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -  33.33% [matrix_unary_f64]: pass = 2, fail = 4, skip = 0, total = 6 duration = 0.353 seconds
- PASS - [matrix_unary_f64.test_arm_mat_cholesky_f64] duration = 0.055 seconds
- PASS - [matrix_unary_f64.test_arm_mat_inverse_f64] duration = 0.238 seconds
- FAIL - [matrix_unary_f64.test_arm_mat_solve_lower_triangular_f64] duration = 0.017 seconds
- FAIL - [matrix_unary_f64.test_arm_mat_solve_upper_triangular_f64] duration = 0.017 seconds
- FAIL - [matrix_unary_f64.test_op1_arm_mat_trans_f64] duration = 0.013 seconds
- FAIL - [matrix_unary_f64.test_op2_arm_mat_sub_f64] duration = 0.013 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: e3cd10826da5700ef673f66190f17178
PROJECT EXECUTION FAILED

```
**Environment (please complete the following information):**

OS: Linux
Toolchain: Zephyr SDK 0.15.0
Commit: 32f4f8d987f4

Kunli123, 2022-09-21:
I used the git bisect tool to find that there was a problem with the commite.
commite：ccb3927233b6ede872e8a0e77c8137d2501764d9

stephanosio, 2022-09-27:
> buffer allocation failed

this means that it is running out of memory. updating the test filter (min_ram) should fix this.

hunterhu3000, 2022-09-29:
> > buffer allocation failed
> 
> this means that it is running out of memory. updating the test filter (min_ram) should fix this.

The SRAM for data access is only64KB, so we cannot enlarge the min_ram
As the malloc/calloc takes continuous memory, we alloc 24.9KB first and then 2 12.8KB small memory, the problem can be fixed.

stephanosio, 2022-09-29:
> The SRAM for data access is only64KB, so we cannot enlarge the min_ram

why not? you are setting the minimum required RAM filter for the test.

> As the malloc/calloc takes continuous memory, we alloc 24.9KB first and then 2 12.8KB small memory, the problem can be fixed.

that is completely up to the libc malloc implementation and not a reliable fix.

hunterhu3000, 2022-09-29:
Please refer to MEC17 data sheet chapter 7.1 https://github.com/MicrochipTech/CPGZephyrDocs/blob/master/MEC172x/MEC172x-Data-Sheet.pdf

Depending on the device, the first SRAM, which is optimized for code access, is
• 352kB
The second SRAM, which is optimized for data access, is
• 64KB

Correct me if my understanding is wrong, I'm not sure if there are some other ways to enlarge the data segment

stephanosio, 2022-09-29:
See https://github.com/zephyrproject-rtos/zephyr/pull/50804#pullrequestreview-1125471350

hunterhu3000, 2022-11-08:
fixed by https://github.com/zephyrproject-rtos/zephyr/pull/51883

closed 2022-11-08 by hunterhu3000