zephyrproject-rtos/zephyr #50455: Intel CAVS15/25: tests/subsys/shell/shell failed.

shuai1x, 2022-09-21:
**Describe the bug**
On intel_adsp_cavs15/25 platform, tests/subsys/shell/shell failed.

Tried:
(1) Add extra delay between test functions, it will pass stably.
(2) Use HDA logging, it will fail sporadically.
Boards overlay:
```
CONFIG_DMA=y
CONFIG_ASSERT=y
CONFIG_LOG=y
CONFIG_LOG_MODE_IMMEDIATE=y
CONFIG_LOG_BACKEND_ADSP=n
CONFIG_LOG_BACKEND_ADSP_HDA=y
CONFIG_LOG_BACKEND_ADSP_HDA_SIZE=8192
CONFIG_LOG_BACKEND_ADSP_HDA_PADDING=y
CONFIG_LOG_BACKEND_ADSP_HDA_CAVSTOOL=y
CONFIG_LOG_BACKEND_ADSP_HDA_FLUSH_TIME=100
CONFIG_XTENSA_ENABLE_BACKTRACE=n
CONFIG_LOG_PRINTK=y
```

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W --hardware-map /home/ztest/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -T tests/subsys/shell/shell/ --no-skipped-report -vv
2. See error

**Logs and console output**
Using HDA logging fail sporadically:
```
DEBUG   - DEVICE: shell_execute_cmd(colors on): 0
DEBUG   - DEVICE: shell_execute_cmd(colors): -8
DEBUG   - DEVICE: shell_execute_cmd(shell colors on): 0
DEBUG   - DEVICE: PASS - test_set_root_cmd in 0.002 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_shell_fprintf
DEBUG   - DEVICE: PASS - test_shell_fprintf in 0.001 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_shell_module
DEBUG   - DEVICE: shell_execute_cmd(test_shell_cmd): 0
DEBUG   - DEVICE: shell_execute_cmd(test_shell_cmd -h): 1
DEBUG   - DEVICE: shell_execute_cmd(test_shell_cmd --help): 1
DEBUG   - DEVICE: shell_execute_cmd(test_shell_cmd dummy): -22
DEBUG   - DEVICE: shell_execute_cmd(test_shell_cmd dummy dummy): -22
DEBUG   - DEVICE: shell_execute_cmd(): -8
DEBUG   - DEVICE: shell_execute_cmd(not existing command): -8
DEBUG   - DEVICE: PASS - test_shell_module in 0.003 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_shell_wildcards_dynamic
DEBUG   - DEVICE: shell_execute_cmd(test_dynamic): 0
DEBUG   - DEVICE: shell_execute_cmd(test_dynamic d*): 1
DEBUG   - DEVICE: shell_execute_cmd(test_dynamic c*): 1
DEBUG   - DEVICE: shell_execute_cmd(test_dynamic d* c*): 2
DEBUG   - DEVICE: PASS - test_shell_wildcards_dynamic in 0.002 seconds
DEBUG   - DEVICE: ===================================================================
DEBUG   - DEVICE: START - test_shell_wildcards_static
DEBUG   - DEVICE: shell_execute_cmd(test_wildcard): 0
DEBUG   - DEVICE: shell_execute_cmd(test_wildcard argument_1): 1
DEBUG   - DEVICE: shell_execute_cmd(test_wildcard argument?1): 1
DEBUG   - DEVICE: shell_execute_cmd(test_wildcard argu?ent?1): 1
DEBUG   - DEVICE: shell_execute_cmd(test_wildcard a*1): 1
DEBUG   - DEVICE: shell_execute_cmd(test_wildcard ar?u*1): 1
DEBUG   - Timed out while monitoring serial output on intel_adsp_cavs25
DEBUG   - Process /home/ztest/zephyrproject/zephyr/soc/xtensa/intel_adsp/tools/cavstool_client.py,-s,192.165.1.58,-l terminated outs: None errs None
DEBUG   - run status: intel_adsp_cavs25/tests/subsys/shell/shell/shell.core error
INFO    - 15/15 intel_adsp_cavs25         tests/subsys/shell/shell/shell.core                FAILED No Console Output(Timeout) (device 60.565s)
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain - SDK 15.0

teburd, 2022-09-23:
Please try with #50450

nashif, 2022-09-24:
> Tried: (1) Add extra delay between test functions, it will pass stably. (2) Use HDA logging, it will fail sporadically. Boards overlay:
>
> ```
> CONFIG_DMA=y
> CONFIG_ASSERT=y
> CONFIG_LOG=y [...]

You actually do not need to use HDA backend at all, you can just use the default and that should work:

```
CONFIG_LOG_PRINTK=y
CONFIG_LOG=y
```

But the still not the proper solution for the issue, we can solve the problem by adding the delays to ztest directly and make the console output (without logging being enabled) a bit slower...

shuai1x, 2022-09-26:
I tested the PR https://github.com/zephyrproject-rtos/zephyr/pull/50623 on CAVS platform five times. And it would pass stably, so this PR will fix this issue.

PR #50623 (2022-09-26): ztest: add a delay between tests
Some testsuites dump lots of output very fast where some systems are not
able to capture the complete output from the tests. Add a slight delay
between each test in the suite.

Fixes #50455
closed 2022-09-26 by fabiobaltieri