zephyrproject-rtos/zephyr #50464: Boot banner can cut through output of shell prompt

nordicjm, 2022-09-21:
**Describe the bug**
Whilst testing `smp_svr` with a build included nearly all of the overlay files, this was seen at startup on a module:
```
*** Booting Zephyr OS build v3.2.0-rc2-88-gf7251073c4b6  ***
I: Starting bootloader
I: Primary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Secondary image: magic=unset, swap_type=0x1, copy_done=0x3, image_ok=0x3
I: Boot source: none
I: Swap type: none
I: Bootloader chainload address offset: 0x10000
I: Jumping to the first image slot

ua*** Booting Zephyr OS build v3.2.0-rc2-92-g6030b85f71ce  ***
rt:~$ 
```
The boot banner is output part the way through the shell prompt output.

**To Reproduce**
Steps to reproduce the behavior:
`cmake -GNinja -DBOARD=nrf5340dk_nrf5340_cpuapp -DOVERLAY_CONFIG="overlay-shell.conf;overlay-shell-mgmt.conf;overlay-bt.conf;overlay-udp.conf" ..` (note: also requires mcuboot for cpuapp and hci_rpmsg for cpunet) then run

**Expected behavior**
The output of the boon banner to not obscure the shell, and ideally, output before the shell prompt starts

**Impact**
Annoyance

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.14.2
 - Commit SHA or Version used: 6030b85f71ce05a69d6953f265ae1831a6c67548

nordicjm, 2022-09-21:
@nordic-krch Is there any reason `CONFIG_LOG_PRINTK` isn't enabled by default if logging is enabled and `CONFIG_PRINTK` is enabled? That avoids this issue

nordic-krch, 2022-09-21:
The main reason was that logging behaves differently, especially deferred mode where output is delayed. Imo, `CONFIG_LOG_PRINTK=y` could be the default but I guess that  we need broader audience to make that decision. Maybe Architecture working group meeting (@carlescufi).

nordicjm, 2022-09-21:
I've just tried enabling `CONFIG_LOG_PRINTK` on this issue https://github.com/zephyrproject-rtos/zephyr/issues/38947 and it's magically resolved, so it does seem like defaulting to this option when logging is enabled would help with a few bugs

carlescufi, 2022-09-21:
> The main reason was that logging behaves differently, especially deferred mode where output is delayed. Imo, `CONFIG_LOG_PRINTK=y` could be the default but I guess that we need broader audience to make that decision. Maybe Architecture working group meeting (@carlescufi).

Can you please post a PR changing the default @nordic-krch? Then we'll drive the discussion in that PR.

carlescufi, 2022-09-21:
Closing this issue since it's a matter of enabling `CONFIG_LOG_PRINTK`.

PR #50594 (2022-11-29): logging: Enable LOG_PRINTK if PRINTK is enabled
This change will enable the logging system to handle printk() calls by default if printk support is enabled, which improves output resiliency when logging, mcumgr and printk calls are used.

Fixes #38947
Fixes #50464
closed 2022-09-21 by carlescufi