zephyrproject-rtos/zephyr #50515: Test result reported as "Skipped" with reason  “No results captured, testsuite misconfiguration?”

mengxianglinx, 2022-09-22:
**Describe the bug**
Some test cases reported as Skipped even they run passed or never been compiled into binary.
It happens on all platforms.

Steps to reproduce the behavior:
1. twister -T tests/
2. cd twister-out; junit2html twister_reprot.xml
3. open twister_report.xml.html, search “No results captured, testsuite misconfiguration?”, find the corresponding test cases.
4. search the handler.log for those test cases: 1) there is no handler.log for that test case, means that test case never been run. 2) in handler.log, that test case actually passed.

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK-0.15.0)
 - Commit SHA or Version used: main line

mengxianglinx, 2022-09-22:
As of now, there is three kind of cases cause this kind of skip:
1) Test log is lost in handler.log. for example, test case shell.core.cmd_history on acrn_ehl_crb, output of this case is not in handler.log
2) If test cases call k_oops(), panic() which cause fatal error, test case actually passed, but the handler.log is not complete, the test report is not ended as expected.
3) If test case source file is orgnized as
#ifdef CONFIG_XXX
test_a();
test_b();
#else
test_c();
test_d();
#endif
for test suite with CONFIG_XXX, test_c(), test_d() will be reported as skipped, even there are never run. likewise for test suite without CONFIG_XXX, test_a(), test_b() are reported as skipped.

nordicjm, 2022-09-22:
CC @PerMac

smrtos, 2022-09-27:
Should be fixed by https://github.com/zephyrproject-rtos/zephyr/pull/49619

closed 2022-10-13 by smrtos