zephyrproject-rtos/zephyr #50538: lpcxpresso55s69_cpu0 samples/subsys/usb/dfu/sample.usb.dfu build failure

fabiobaltieri, 2022-09-22:
**Describe the bug**

Build failure for:

`west build -p -b lpcxpresso55s69_cpu0 samples/subsys/usb/dfu`

fails with

```
/zephyrproject/bootloader/mcuboot/boot/bootutil/zephyr/../include/bootutil/bootutil_public.h:86:1: error: static assertion failed: "Unsupported value for MCUBOOT_BOOT_MAX_ALIGN"
   86 | _Static_assert(MCUBOOT_BOOT_MAX_ALIGN >= 8 && MCUBOOT_BOOT_MAX_ALIGN <= 32,
      | ^~~~~~~~~~~~~~
```

The check was introduced in

95292e85aa manifest: mcuboot update

@nvlsianpu can you take a look?

**To Reproduce**
`west build -p -b lpcxpresso55s69_cpu0 samples/subsys/usb/dfu` on current main.

**Expected behavior**
Builds

**Impact**
CI failure

fabiobaltieri, 2022-09-22:
(@stephanosio fyi, from the yesterday's full run)

nordicjm, 2022-09-23:
This platform should be excluded from the test:
> Up to 640 KB on-chip flash program memory with flash accelerator and 512 byte page erase and write.

mcuboot cannot work on this device

fabiobaltieri, 2022-09-23:
@dleach02 seems like the various lpc55 platforms cannot support mcuboot right now due to limitations on the page write size:

```
dts/arm/nxp/nxp_lpc55S0x_common.dtsi:                   write-block-size = <512>;
dts/arm/nxp/nxp_lpc55S1x_common.dtsi:                   write-block-size = <512>;
dts/arm/nxp/nxp_lpc55S2x_common.dtsi:                   write-block-size = <512>;
dts/arm/nxp/nxp_lpc55S3x_common.dtsi:                   write-block-size = <512>;
dts/arm/nxp/nxp_lpc55S6x_common.dtsi:                   write-block-size = <512>;
```

See:

https://github.com/zephyrproject-rtos/mcuboot/blob/2fcae1c1a05fe3a0892ba3c649f93cf3e1077949/boot/bootutil/include/bootutil/bootutil_public.h#L86-L87

I'll send a PR to exclude the board from the test but maybe it should not have the mcuboot partitions defined in the first place if it's not supported:

https://github.com/zephyrproject-rtos/zephyr/blob/1bbe209a30fcdab037c8ac9b96c33ea2249665e0/boards/arm/lpcxpresso55s69/lpcxpresso55s69.dtsi#L120-L140

Is there' something I'm missing?

cc @mmahadevan108

closed 2022-09-23 by fabiobaltieri