zephyrproject-rtos/zephyr #50546: drivers: can: rcar: likely inconsistent behavior when calling can_stop() with pending transmissions

henrikbrixandersen, 2022-09-22:
**Describe the bug**
The Renesas R-Car CAN controller drivers likely exhibits inconsistent behavior when calling `can_stop()` with pending CAN frame transmissions similar as to what was reported in #50545. I have not been able to find the behavior of the Renesas R-Car CAN driver as I do not have access to the hardware.

**To check**
Steps to reproduce the behavior:
1. Disconnect the CAN controller from the CAN bus.
2. Using the CAN shell, call `can start <device>`.
3. Using the CAN shell, call `can send <device> 010`.
4. Using the CAN shell, call `can stop <device>`.
5. Connect the CAN controller to a CAN bus with at least one other node for ACK'ing frames and monitor any received frames on that node (e.g. use `candump` on GNU/Linux).
6. Using the CAN shell, call `can start <device>`.
7. Observe the different behavior across the Zephyr CAN drivers as described above.

**Expected behavior**
The behavior should be the same across the Zephyr CAN drivers. Since some CAN controllers automatically abort any pending transmissions, the common behavior should be to abort any pending transmissions and notify the senders of this.

**Impact**
Unknown.

**Environment (please complete the following information):**
 - Commit SHA: 7e6e4579c29f269626d52bb70d5eed30113d90ee

**Additional context**
The `can_stop()` API was introduced in #50102.

henrikbrixandersen, 2022-10-28:
@aaillet Ping

aaillet, 2023-04-20:
Hi @henrikbrixandersen,

Sorry for this big delay...

I just verified `can_stop()` behavior on a random late Feb 2023 sha (dca33a126d1765cb647c10eabeea36eafcbd0f92) on H3ULCB board and got these results:
```shell
#Board not plugged to CAN bus
uart:~$ can start can@e6c30000
starting can@e6c30000
uart:~$ can send can@e6c30000 01 DE AD BE EF
enqueuing CAN frame #0 with standard (11-bit) CAN ID 0x001, RTR 0, CAN-FD 0, BRS 0, DLC 4
uart:~$ can stop can@e6c30000 
stopping can@e6c30000
#Pluging board to CAN bus
uart:~$ can start can@e6c30000
starting can@e6c30000
CAN frame #0 successfully sent
```

I can confirm that the frame has successfully been received by the other CAN device (a computer using CAN USB adapter) as soon as `can start()` has been called.
I think that this is the behavior we are waiting for but I just wanted you to validate that !

Thank you in advance !

aaillet, 2023-04-21:
> Unfortunately, that is the _opposite_ of the desired behaviour. Calling `can_stop()` should abort any pending TX frames.

Oh ok, so we want the controller to empty FIFOs and clear everything that has to be cleared !
I'll work on fixing that, thanks for the answer !

PR #57344 (2023-05-24): drivers:can: Fix R-Car driver can_stop behaviour
Calling for can_stop() method on R-Car boards was not emptying FIFOs and tx_msgq.

We use controller reset mode to empty controller FIFOs.
We are emptying tx_msgq with "-ENETDOWN" return code.

Tested using can shell module ans obtained the following behaviour:
```
#Board disconnected from CAN bus
uart:~$ can start can@e6c30000
starting can@e6c30000
uart:~$ can send can@e6c30000 001 DE AD BE EF
enqueuing CAN frame #1 with standard (11-bit) CAN ID 0x001, RTR 0, CAN-FD 0, BRS 0, DLC 4
uart:~$ can send can@e6c30000 001 DE AD BE
enqueuing CAN frame  [... cut]
commit 8959b838 (2023-05-24): drivers: can: Fix R-Car driver can_stop behavior

Calling for can_stop() method was not emptying FIFOs
and tx_msgq.
Resetting TX FIFO to empty it.
Emptying tx_msgq with "ENETDOWN" return code.

Fixes #50546
closed 2023-05-24 by nashif