zephyrproject-rtos/zephyr #50554: Test uart async failed on Nucleo F429ZI

gautierg-st, 2022-09-22:
**Describe the bug**
The test `tests/drivers/uart/uart_async_api/drivers.uart.async_api` fails on Nucleo F429ZI with the following error
`Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:526: uart_async_write_abort_test_write_abort: (sent not equal to received)
Sent is not equal to received.`

gautierg-st, 2022-09-29:
After investigation, it appears that it is a case of unstable test.
A git bisect has shown that commit ad1d6d passes and following commit 32ad01 fails, but the only difference between the two is in the MAINAINERS.yml file.
A simple modification of the error message (to display the number of bytes sent and received) makes the test PASS.
Similarly, a repetition of an instruction makes the test PASS.

FRASTM, 2022-10-24:
not reproduced anymore since a while

closed 2022-10-24 by FRASTM