zephyrproject-rtos/zephyr #50562: RFC: AP Power Sequence Driver

perezpri, 2022-09-22:
## Introduction

An Embedded Controller (EC) is essentially a microcontroller installed on a mobile PC platform that performs important system level operations such as: fan control, thermal measurement, LED indication toggling, and communication ports enablement, among others.

One of the key roles of EC is to monitor power rails and supply Application Processor (AP) with required signaling to drive its power state and achieve optimal platform performance.

### Problem description

There is currently no support for AP Power Sequence driver in zephyr. This is not to be confused with Power Management (PM) module, the purpose of this driver to allow host microcontroller to manipulate signals wired to main CPU in the platform and achieve power state transitions.

### Proposed change

Proposing a driver that allows interaction with AP Power Sequence subsystem and sets an underlying framework to easily add or extend AP Power Sequence routines.

[image]

This driver is composed of the following parts:

* Driver API
   This will export functionality to application layer. Given that trivial driver operations are exported, it does not require to use pointer to function structure.

* Driver Instance
   Driver will have its own thread were state actions will be executed along with notifications to clients.
   
* State machine API
   This is for internal driver usage only, exports the methods to do state transitions and state action handler execution.

* Power State Action Handlers wrapper
   A whole set of macros that facilitate state action handler addition.

#### ACPI

The ACPI Specification defines power states for an ACPI-compliant computer system. AP Power Sequence Driver implementation consists of defining operations to be performed on every power state handler and specify state transition utilizing mechanism provided by state machine API.

Power states are defined as below:
   **S0**: the run state. In this state, all the power rails are up and the machine is fully running.
   **S1**, **S2**, **S3**: a low wake-latency sleep state. In this state, memory contexts are held (RAM on) but CPU contexts are lost.
   **S4**: the lowest-power sleep state, longest wake-latency sleeping state supported by ACPI, AP is suspended to non-volatile disk.
   **S5**: the soft-off state. All activity will stop, and all contexts are lost.
   **G3**: soft-off state, the system is completely off and consumes no power.

#### State Machine

State machine is integrated into the driver through the use of Zephyr State Machine Framework (SMF), each SMF state is represented by three functions or action handlers that define operations performed on state entry, run and exit.

ACPI’s global state (G3) and its six sleep power states (S0, S1, S2, S3, S4, S5) are present within the driver state machine domain. All these ACPI states are divided in three levels, each level is a SMF state with a hierarchical relation with others action handlers of the same ACPI state, state handlers at higher levels accomplishing the most common processing of corresponding ACPI power states.

**Architecture** is the highest level of the hierarchy, SMF states at this level drive signals that are  specific to AP CPU architecture, example: X86 (intel), ARM.

In the middle is the **chipset** level; these SMF states carry out operations to drive power of components that are required for the AP chip. Any bus signal or internal power rail that is vital for chip execution is a good fit to be handled in these SMF states. Examples of chipsets are: Tiger Lake and Jasper Lake, these are Intel chipsets that use X86 architecture, and MT8186 and MT8192 are Mediatek chipsets using ARM architecture.

**Application** is at the bottom level of the hierarchy, and these SMF states are reserved for action handlers intended to address board or application specific computations. Example: Hatch, jacuzzi, volteer, asurada, etc…

Hierarchical SMF feature will coordinate execution of entry, run & exit functions accordingly; given that implementation is responsible for doing state transitions, the following considerations should be taken when implementing action state handlers:
* Higher level `entry` actions are executed before the lower level `entry` actions.
* Transitioning from one substate to another with a shared upper level state does not re-execute the upper level `entry` action or execute the `exit` action.
* Upper level `exit` actions are executed after the substate `exit` actions.
* Lower level `run` actions are executed before upper level.
* Upper level `run` actions only executes if no state transition has been made from lower level `run` action.

Please refer to Zephyr [SMF](https://docs.zephyrproject.org/3.1.0/services/smf/index.html) documentation.

State transitions are dictated by the implementation. It is not possible to advance directly into architecture or chipset level SMF state handler within an ACPI state without following parent child execution flow mentioned earlier, this means, transitions are made into application level SMF state handler, in case is not defined, upper level will be called instead.

**Transitioning into an ACPI state with no SMF state action handlers implemented for any level will result in blocking state machine execution indefinitely.**

#### State Implementation

For ACPI states, SMF action handlers are defined using the following macros:
```
[... 171 lines cut ...]
In order to promote this pattern, It is recommended for upper level action handlers to query if state has been handled.

#### Driver

Drivers mediate client communication with the underlying state machine and manage its execution. Upon initialization, the driver will create its own thread. States execution, state transition and any required callback invocation will be performed within the driver’s thread context. Driver’s thread is limited to only execute the `run` action of the current state while implementation is responsible to do state transitions to match AP requirements.

If no state transition has occurred during a thread iteration, the thread will block indefinitely waiting for any event to be posted by one of the driver's clients. Events posted will be passed to the state machine to be processed by the current state action handler, after this, all events received will be discarded prior to the next iteration.

#### Driver Interface
Driver does not require to have any API structure given that its functionality is trivial, on the other hand state machine implementations will vary. For this reason, driver does not provide implementation to any API structure, instead its interface is fixed.

Driver interface allows clients to post events to be processed by state machine implementation, and register callbacks to receive notifications associated with certain power state upon entrance, exit or execution.

#### Driver PR

Driver code and examples on how to add state action handlers can be found in the following draft PR #44481.

### Dependencies

None, however this driver heavily depends on SMF module.

### Concerns and Unresolved Questions

* I did not see the need to have a driver API structure, fixed API is exposed instead, I hope this is acceptable.
* In order to successfully build hierarchical state machine, empty SMF structures needed to be created, I could not find another way to achieve this.

stephanosio, 2022-09-22:
Why does this need to be part of the upstream Zephyr?

Should it not be part of the Embedded Controller firmware "app"?

albertofloyd, 2024-02-22:
> Why does this need to be part of the upstream Zephyr?
> 
> Should it not be part of the Embedded Controller firmware "app"?

Agree , this belongs in EC FW app

closed 2024-06-25 by albertofloyd