zephyrproject-rtos/zephyr #50565: Fatal error after `west flash` for nucleo_l053r8

mm40, 2022-09-23:
**Describe the bug**
I've been following the [getting started guide](https://docs.zephyrproject.org/latest/develop/getting_started/index.html), and everything went well until the `west flash` command. That command fails with a fatal error.

Please also mention any information which could help others to understand 
the problem you're facing:
 - What target platform are you using? *nucleo_l053r8*
 - What have you tried to diagnose or workaround this issue? *tried using st-flash to flash image, failed, see below*
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? *Didn't try previous versions*

**To Reproduce**
Steps to reproduce the behavior:
1. Follow the getting started tutorial step by step on `nucleo_l053r8`, and `west flash` fails.

**Logs and console output**
This is the output of the `west flash` command :
```
(.venv) > west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.11.0+dev-00724-g42b6471c1 (2022-08-17-18:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 300 kHz
Info : STLINK V2J40M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.251613
Info : [stm32l0.cpu] Cortex-M0+ r0p1 processor detected
Info : [stm32l0.cpu] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for stm32l0.cpu on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* stm32l0.cpu        hla_target little stm32l0.cpu        running

Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x080007b4 msp: 0x20002000
Info : Device: STM32L0xx (Cat. 3)
Info : STM32L flash size is 64kb, base address is 0x8000000
auto erase enabled
wrote 16384 bytes from file /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex in 3.245303s (4.930 KiB/s)

Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x08000e2c msp: 0x200007a0
verified 13180 bytes in 0.568584s (22.637 KiB/s)

Info : Unable to match requested speed 300 kHz, using 240 kHz
Info : Unable to match requested speed 300 kHz, using 240 kHz

FATAL ERROR: command exited with status 1: /home/theuser/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/theuser/zephyrproject/zephyr/boards/arm/nucleo_l053r8/support -s /home/theuser/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/theuser/zephyrproject/zephyr/boards/arm/nucleo_l053r8/support/openocd.cfg '-c init' '-c targets' -c 'reset halt' -c 'flash write_image erase /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

So, I've executed the very same command with `-d` flag, and this is the output. *Note : after every `west flash` attempt I properly re-flash using a dummy project in stm32cubeide, because until I do that, it's completely unresponsive*

```
(.venv) > /home/theuser/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -d -s /home/theuser/zephyrproject/zephyr/boards/arm/nucleo_l053r8/support -s /home/theuser/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/theuser/zephyrproject/zephyr/boards/arm/nucleo_l053r8/support/openocd.cfg '-c init' '-c targets' -c 'reset halt' -c 'flash write_image erase /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```
[logfile.txt](https://github.com/zephyrproject-rtos/zephyr/files/9630099/logfile.txt)

**Environment (please complete the following information):**
 - OS: 5.19.8-arch1-1
 - Zephyr SDK 0.15.0

**Additional context**
Right after the first failure of `west flash`, if I try it again, I get a different error output :
```
(.venv) > west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.11.0+dev-00724-g42b6471c1 (2022-08-17-18:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 300 kHz
Info : STLINK V2J40M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.256353
Error: init mode failed (unable to connect to the target)

FATAL ERROR: command exited with status 1: /home/theuser/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/bin/openocd -s /home/theuser/zephyrproject/zephyr/boards/arm/nucleo_l053r8/support -s /home/theuser/zephyr-sdk-0.15.0/sysroots/x86_64-pokysdk-linux/usr/share/openocd/scripts -f /home/theuser/zephyrproject/zephyr/boards/arm/nucleo_l053r8/support/openocd.cfg '-c init' '-c targets' -c 'reset halt' -c 'flash write_image erase /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

And I keep getting that output until I flash using a dummy project in stm32cubeide. Only after that being successful, can the original error of `west flash` be reproduced again.

Also, I tried to do `st-flash --format ihex write /home/theuser/zephyrproject/zephyr/build/zephyr/zephyr.hex`, but that resulted in :
```
(.venv) > st-flash --format ihex write /home/debkorisnik/zephyrproject/zephyr/build/zephyr/zephyr.hex
st-flash 1.7.0
2022-09-23T01:52:37 INFO common.c: L0x3: 8 KiB SRAM, 64 KiB flash in at least 128 byte pages.
2022-09-23T01:52:37 INFO common.c: Attempting to write 13180 (0x337c) bytes to stm32 address: 134217728 (0x8000000)
2022-09-23T01:52:37 INFO common.c: Flash page at addr: 0x08000000 erased
2022-09-23T01:52:37 INFO common.c: Flash page at addr: 0x08000080 erased

...

2022-09-23T01:52:38 INFO common.c: Flash page at addr: 0x08003280 erased
2022-09-23T01:52:38 INFO common.c: Flash page at addr: 0x08003300 erased
2022-09-23T01:52:38 INFO common.c: Finished erasing 103 pages of 128 (0x80) bytes
2022-09-23T01:52:38 INFO common.c: Starting Flash write for L0
2022-09-23T01:52:38 INFO common.c: Starting Half page flash write for STM32L core id
2022-09-23T01:52:38 INFO flash_loader.c: Successfully loaded flash loader in sram
2022-09-23T01:52:38 INFO flash_loader.c: Clear DFSR
2022-09-23T01:52:39 ERROR flash_loader.c: Flash loader run error
2022-09-23T01:52:39 WARN flash_loader.c: Loader state: R2 0xFFFFFFFF R15 0x8000E2C
2022-09-23T01:52:39 WARN flash_loader.c: MCU state: DHCSR 0x1080009 DFSR 0x0 CFSR 0x0 HFSR 0x0
2022-09-23T01:52:39 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2022-09-23T01:52:39 WARN common.c: 
write_half_pages failed == -1
```
And the process never ends until I kill it.

Also, I saw #50306, and tried a quick and dirty solution to change `reset_config srst_nogate` to `reset_config srst_only srst_nogate` in `stm32l0.cfg`, didn't work.

erwango, 2022-09-23:
It seems there is a regression coming from the openocd version packed with the latest SDK. This is under investigation.

Here are a few workarounds:
1- You should be able to flash by pressing the reset button.

2- Alternatively, you can edit the board's `board.cmake` file and add:
```
board_runner_args(stm32cubeprogrammer "--port=swd" "--reset=hw")
include(${ZEPHYR_BASE}/boards/common/stm32cubeprogrammer.board.cmake)
```
and flash using `--runner=stm32cubeprogrammer`.
Of course, this requires stm32cubeprogrammer installation, and that it is available in you environment ($PATH on Linux).

mm40, 2022-09-23:
Hey thanks, that works for flashing.  However, nothing is blinking on my board when I flashed blinky. Any tips on how to get debugging to work? `west debug` uses openocd too which doesn't work.

erwango, 2022-09-23:
Ok, it seems there is something quite weird about this board. I don't have explanation so far.
I can get things working nicely on similar `nucleo_l073rz`.
I've tried to use previous Zephyr version as well. No more luck, which is even more strange.

I'll keep you posted, but it seems you picked up the ugly duckling, unfortunately.

erwango, 2022-09-23:
Moving priority to low as only this board is impacted.

erwango, 2022-09-23:
Note: `nucleo_l011k4` working as expected.

FRASTM, 2022-11-02:
building for the nucleo_l053r8 `west build -p auto -b nucleo_l053r8 samples/basic/blinky
and flashing (west flash) on a **nucleo_l073rz** 
works on this target board : LD2 is blinking

closed 2023-01-17 by github-actions[bot]