zephyrproject-rtos/zephyr #50567: Passed test cases are reported as "Skipped" because of incomplete test log

mengxianglinx, 2022-09-23:
**Describe the bug**
Passed test cases are reported as “Skipped" with reason “No results captured, testsuite misconfiguration?”.

**To Reproduce**
Steps to reproduce the behavior:
On frdm_k64f: 
1. twister -p frdm_k64f --device-testing --hardware-map map.yml -T tests/
2. In twister-out/twister_report.xml, "coredump.backends.logging.coredump_backend" and "drivers.coredump.api.register_memory" are reported as "skipped" with reason "No results captured, testsuite misconfiguration?", but these two case are passed.
``
In the handler.log of "drivers.coredump.api.register_memory": 
Booting Zephyr OS build v3.2.0-rc2-178-g55c4e06b3563
Running TESTSUITE coredump_tests
===================================================================
START - test_register_memory
E: r0/a1:  0x00000004  r1/a2:  0x00005cba  r2/a3:  0x000059a4
E: r3/a4:  0x00000062 r12/ip:  0x00000000 r14/lr:  0x00000893
E:  xpsr:  0x61000000
E: Faulting instruction address (r15/pc): 0x0000089e
E: >>> ZEPHYR FATAL ERROR 4: Kernel panic on CPU 0
E: Current thread: 0x20000090 (test_register_memory)
E: #CD:BEGIN#
E: #CD:5a4501000300050004000000
E: #CD:4102004400
E: #CD:04000000ba5c0000a45900006200000000000000930800009e08000000000061
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:00000000
E: #CD:4d01009000002030010020
E: #CD:d001002050030020000000000880ff0000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:000000000000000000000000000000002008002000000000e8000020e8000020
E: #CD:746573745f72656769737465725f6d656d6f7279000000000000000000000000
E: #CD:0000000040040020000400000000000000000000000000000000000000000000
E: #CD:4d01004004002040080020
E: #CD:000f01d0c8f80040baf1000f02d00020caf8000004e03f1f2d1d241d002fdad1
E: #CD:00bf3046c9e70000740000000000024000000000000008000000100000002000
E: #CD:0000400000008000000000010000000104000240000000000000000000001000
E: #CD:0200000000100000000000000000000000000000010000000000000000000000
E: #CD:01000000010000000100000000000000000000000100000001000000dc040020
E: #CD:dc0400200000000002000000ec040020ec040020000000000200000000000020
E: #CD:200000009ee77900401300204000000008822000801d00204000000008822000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:000000000000000001000000c949ba0700000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:000000000000000000000000f0f00000396000002c0600200000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000031310000040000000000000000000000
E: #CD:00000000e2060000040000000000000000000000000000000000000000000000
E: #CD:0000000000000000000000000000000000000000d4060020d406002020000020
E: #CD:02000000ffffffff000054150000000031310800000054150403020108070605
E: #CD:00000000000000000000a81500000000010000004a580000f055000028000000
E: #CD:bd5a000009000000b65a0000db070000bd5a000071580000080000002d070000
E: #CD:0000000000ed00e0f0020020000000000000000033410000c012000000000061
E: #CD:e11d0000c000002000000000c000002000000000e547000000000000e5470000
E: #CD:0000000000000000000000008c0700208c0700200b0000000b0000009c070020
E: #CD:9c07002000000000010000000000000000000800000058150403020108070605
E: #CD:000000000000000004000000ba5c0000a4590000620000000000000093080000
E: #CD:9e08000000000061275d00003b5c00004800002000000000fc56000010570000
E: #CD:e0560000f60300201057000000000000000000004f200000000000001d200000

**Environment (please complete the following information):**
 - OS: (Linux)
 - Toolchain (Zephyr SDK-0.15.0)
 - Commit SHA or Version used: Main line

mengxianglinx, 2022-09-23:
It seems like twister prefer a complete handler.log ends with
RunID: xxxxxxxxxxxxxxx
PROJECT EXECUTION XXXXXXXX
if the test case call panic() or k_oops(), there is no this kind of info output to handler.log

enjiamai, 2022-09-23:
The kinds of test case shows “No results captured, testsuite misconfiguration?” due to it will not print out the "pass" message as test result because it is expected to trigger a fatal error.  In the twister.json(report) it's status is marked us null. 
```
            "name":"tests/drivers/coredump/coredump_api/drivers.coredump.api",
            "arch":"arm",
            "platform":"frdm_k64f",
            "run_id":"334c59e4ead42bd7cc1becac722c7daf",
            "runnable":true,
            "status":"passed",
            "execution_time":"3.92",
            "testcases":[
                {
                    "identifier":"drivers.coredump.api.register_memory",
                    "execution_time":"0.00",
                    "status":null
                },
```

enjiamai, 2022-09-23:
Still checking if this is a bug.

nashif, 2022-12-02:
cant reproduce

closed 2022-12-02 by nashif