zephyrproject-rtos/zephyr #50587: Regression in Link Layer Control Procedure (LLCP)

caco3, 2022-09-23:
As requested in https://discord.com/channels/720317445772017664/733036956392357908/1022831217969733712 I create following bug report.

We have an nrf52840 system where we use BLE to transfer data at a high througput.
After we updated from https://github.com/zephyrproject-rtos/zephyr/commit/e1c2c36f569a8e06184a2c5574f1d25ba00df8d7 to the next commit https://github.com/zephyrproject-rtos/zephyr/commit/e1c2c36f569a8e06184a2c5574f1d25ba00df8d7, we see following log messages after connecting (device is peripheral):
```
<wrn> bt_hci_core: opcode 0x2016 status 0x0c
err> bt_hci_core: Failed read remote features (-5)
```

Beside of those 2 log entries we did not see any negative impact!

As suggested by @cvinayak, with `CONFIG_BT_LL_SW_LLCP_LEGACY` enabled, the log entries are gone again.

**To Reproduce**
Steps to reproduce the behavior:
1. have an application with BLE
2. connect to it, eg using the [nRF Connect for Desktop](https://www.nordicsemi.com/Products/Development-tools/nrf-connect-for-desktop)
3. After connecting to the device, the mentioned log entries are shown in the device's log.

**Impact**
None as far as I can see

**Environment (please complete the following information):**
Target: nrf52840
OS: Linux (Ubuntu)
Toolchain: Zephyr SDK 0.15.0
Commit SHA or Version used: v3.1.0 (https://github.com/zephyrproject-rtos/zephyr/commit/e1c2c36f569a8e06184a2c5574f1d25ba00df8d7)

kruithofa, 2022-09-29:
I haven't been  able to reproduce the behaviour yet, but I think the problem may be solved by increasing the number of available local procedure contexts, for example by adding the following lines to your prj.conf file:

> CONFIG_BT_CTLR_ADVANCED_FEATURES=y
> CONFIG_BT_CTLR_LLCP_LOCAL_PROC_CTX_BUF_NUM=8

Or by running 'west build -t menuconfig':

- under 'sub Systems and OS Services -> Bluetooth' enable 'show advanced features', 
- under 'sub Systems and OS Services -> Bluetooth -> Advanced features' increase the number of local controle procedure contexts available to 8 (default = 4)

@caco3 is this something you can try?

caco3, 2022-09-30:
Dear @kruithofa 

Thank you for looking into it.

I have  `CONFIG_BT_CTLR_ADVANCED_FEATURES` already activated because we need it for `CONFIG_BT_CTLR_ULL_HIGH_PRIO` and `CONFIG_BT_CTLR_XTAL_THRESHOLD`.

I now disabled `CONFIG_BT_LL_SW_LLCP_LEGACY` again and instead set `CONFIG_BT_CTLR_LLCP_LOCAL_PROC_CTX_BUF_NUM=8`.

The log entries then no longer show up and I didn't find any regression.

kruithofa, 2022-09-30:
Thanks! I suggest that for the time being you keep this setting (i.e. CONFIG_BT_CTLR_LLCP_LOCAL_PROC_CTX_BUF_NUM=8). 
I'll leave the issue open until I've discussed/thought more about whether we need a better fix for this (or if increasing the number of buffers is sufficient)

kruithofa, 2022-10-13:
@kruithofa increase the default to 6 (from 4)

caco3, 2022-10-13:
I will test it in 2 weeks after my holidays

caco3, 2022-10-31:
I can confirm that `CONFIG_BT_CTLR_LLCP_LOCAL_PROC_CTX_BUF_NUM=6` also is fine.

Note that in my case (on `3.2.0`), the default value is on `2`. 
Setting it to `4` also worked.

cvinayak, 2022-10-31:
Thank you for confirming.

caco3, 2022-10-31:
IMO we can close this ticket now.

kruithofa, 2022-11-17:
Issue will be closed when PR #51300 is merged

PR #51300 (2022-11-23): Bluetooth: controller: llcp: increase memory for procedures
The default number of buffers (contexts) for locally initiated procedures was 4. Per default the host initiates 3 procedures when opening a connection, allowing the application to initiate only 1 additional procedure in parallel. This commit increases the default to 6, if auto phy-update and dle is enabled,
 increasing the robustness for an application.

Fixes #50587
closed 2022-11-23 by carlescufi