zephyrproject-rtos/zephyr #50596: Documentation: Broken links in the documentation at docs.zephyrproject.org/

Thalley, 2022-09-23:
**Describe the bug**
Using a linkchecker (https://github.com/linkchecker/linkchecker) I've been able to produce a list of broken URLs/links found the in documentation. 

The linkchecker was run with `linkchecker --no-robots --check-extern --no-warnings`. 

I've run it twice: Once locally on locally built and hosted documentation (SHA: `dca9c2b165b92e90d38a5002ae7e6ad013daf04d`), and once on the actual documentation found at https://docs.zephyrproject.org. 

The result for the local run is attached here.

The result for the remote run is still being compiled, and will be added later. 

**To Reproduce**
Steps to reproduce the behavior:
1. Download and install linkchecker
2. Run with `linkchecker --no-robots --check-extern --no-warnings https://docs.zephyrproject.org.`

**Expected behavior**
Ideally no URLs should return an error. 

**Impact**
Annoyance or missing information. 

**Logs and console output**
See attached files

**Environment (please complete the following information):**
 - Commit SHA or Version used: `dca9c2b165b92e90d38a5002ae7e6ad013daf04d`

**Additional context**
See attached files

To make the links in the txt-file directly usable, I would suggest do a simple search-and-replace of `http://localhost:9000` to `https://docs.zephyrproject.org`

File from local run (only on latest docs) containing 119 `404`s:
[linkchecker_local.txt](https://github.com/zephyrproject-rtos/zephyr/files/9634256/linkchecker_local.txt)

Thalley, 2022-09-23:
It seems that we have a lot of dead links for Kconfig options that were removed in newer versions, as the older versions still have a link with `The latest development version of this page may be more current than this released 2.7.0 version.` when there is not latest version with the Kconfig in it. 

I do not know if that goes for all the Kconfig ones, or if the links for some of them just don't work, but it seems that roughly 80% of the errors (8806) founds are from Kconfig links not working.

Thalley, 2022-09-29:
@stephanosio Did your PR fix all broken links from this issue? I think not, right?

stephanosio, 2022-09-29:
> @stephanosio Did your PR fix all broken links from this issue? I think not, right? 

It did, for the latest documentation.

Thalley, 2022-09-29:
> > @stephanosio Did your PR fix all broken links from this issue? I think not, right?
> 
> It did, for the latest documentation.

That is true, but for the non-latest docs we still have a lot of broken links, which are also reported by this issue. 

I can open a new GH issue for those if you like, or we can reopen this.

gmarull, 2022-09-29:
> That is true, but for the non-latest docs we still have a lot of broken links, which are also reported by this issue.
>
> I can open a new GH issue for those if you like, or we can reopen this.

Old docs are never re-built, we should probably just keep fixes in LTS branches.

stephanosio, 2022-09-29:
> That is true, but for the non-latest docs we still have a lot of broken links, which are also reported by this issue.

I guess it makes sense to fix the broken links for the currently supported releases (2.7, 3.0, 3.1); though, we do not have a clear policy on them.

I will rename the issue to reflect that the scope of this is for the previous supported releases.

PR #50779 (2022-09-29): Fix broken links in the documentation
This series fixes various broken links in the documentation.

Partially fixes #50596
closed 2022-12-13 by github-actions[bot]