zephyrproject-rtos/zephyr #50598: UDP over IPSP not working on nRF52840

jmrua, 2022-09-23:
**Describe the bug**
UDP packets are not sent through the IPSP interface if the destination IP is a global IP. It works just fine if the destination IP is link-local. Both ICMPv6 and TCP traffic work as expected. 

Although the packets do not appear on the interface between the host and the device (analyzed using Wireshark from the host), the zephyr logs seem to indicate that they are sent, no errors occur and the packet send callback returns a value greater than zero. 
This was tested and replicated on the [IPSP sample](https://docs.zephyrproject.org/latest/samples/bluetooth/ipsp/README.html) and other samples such as the [echo server](https://docs.zephyrproject.org/latest/samples/net/sockets/echo_server/README.html).

**Environment**
Tested on:
- Host
    - Linux with Kernel 4.14 / 5.15 / 5.19.6
    - Several BLE USB dongles
- Zephyr Device
    - nRF52480DK (nrf52840dk_nrf52840)
         - Zephyr RTOS versions:
             - v.3.2.0-rc2
             - v.3.1.0
             - v.2.7.1
         - Zephyr SDK versions:
              - v.15.0  
              - v.14.2
              - v.14.1

The route and IP configuration were performed manually ( `CONFIG_NET_CONFIG_MY_IPV6_ADDR`  and `CONFIG_NET_CONFIG_PEER_IPV6_ADDR`) and also tested with a remote configuration via Router Advertisement with RADVD service in the host (configuration shown below).

radvd.conf
```
interface bt0
{
AdvSendAdvert on;
AdvDefaultPreference high;
prefix 2001:db8::/64
{
AdvOnLink on;
AdvAutonomous on;
AdvRouterAddr on;
};

	route 2001:db8::/64
	{
		AdvRoutePreference high;
	        AdvRouteLifetime 3600;
	};

};
```

**To Reproduce**
Steps to reproduce the behavior:
1. Compile and flash the `sample/bluetooth/ipsp` sample
2. Connect to the board from the host.
``` bash
 modprobe bluetooth_6lowpan
 echo 1 > /sys/kernel/debug/bluetooth/6lowpan_enable
 echo "connect  <nRF MAC> <type>" > /sys/kernel/debug/bluetooth/6lowpan_control
 ip address add 2001:db8::2/64 dev bt0 # only needed on manual IP configuration
```
  3.a. send a UDP packet from the host to the board IP with port 4242. (for example, using Netcat)
  ` nc -6 -u <nRF IP> 4242
  `

 3.b. Alternatively, the Zephyr shell command can be used to replicate the same behavior
`net udp send <host IP> <host port> <Payload>
`

When using a global IP as destination, the board never replies and starts to send a Neighbor Solicitation request to the host IP every time it should send a UDP packet, even if there is a valid neighbor entry on `net nbr` or a valid route installed on the board.

Additionally, the following settings have been added to the sample's prj_dbg.conf to try to identify the problem.

```
CONFIG_NET_BUF_LOG_LEVEL_DBG=y
[... 131 lines cut ...]
[00:14:00.579,437] <dbg> net_6lo: compress_tfl: (shell_uart): Traffic class and Flow label elided
[00:14:00.579,711] <dbg> net_if: net_if_tx: (shell_uart): Calling context send cb 0x20005d64 status 45
[00:14:01.530,700] <dbg> net_core: net_recv_data: (BT RX): prio 0 iface 0x200018f4 pkt 0x20010ae0 len 51
[00:14:01.530,883] <dbg> net_core: net_rx: (rx_q[0]): Received pkt 0x20010ae0 len 51
[00:14:01.530,914] <dbg> net_6lo: get_ihpc_inlined_size: (rx_q[0]): Size of inlined IP HDR data: 17
[00:14:01.530,944] <dbg> net_6lo: uncompress_IPHC_header: (rx_q[0]): Enough tailroom. Uncompress inplace
[00:14:01.530,975] <dbg> net_6lo: uncompress_tfl: (rx_q[0]): Tcl and Flow label elided
[00:14:01.531,005] <dbg> net_6lo: uncompress_sa: (rx_q[0]): SAC_0
[00:14:01.531,036] <dbg> net_6lo: uncompress_sa: (rx_q[0]): SAM_11 generate src addr from ll
[00:14:01.531,066] <dbg> net_6lo: uncompress_da: (rx_q[0]): DAC_0
[00:14:01.531,097] <dbg> net_6lo: uncompress_da: (rx_q[0]): DAM_00 full dst addr inlined
[00:14:01.531,188] <dbg> net_ipv6: net_ipv6_input: (rx_q[0]): IPv6 packet len 72 received from fe80::8c88:2bff:fe20:39d8 to 2001:db8::c4b2:5bff:fe9b:d45a
[00:14:01.531,341] <dbg> net_ipv6: handle_ns_input: (rx_q[0]): Received Neighbor Solicitation from fe80::8c88:2bff:fe20:39d8 to 2001:db8::c4b2:5bff:fe9b:d45a, target 2001:db8::c4b2:5bff:fe9b:d45a iface 0x200018f4/1
[00:14:01.531,463] <dbg> net_ipv6: net_ipv6_nbr_add: (rx_q[0]): [0] nbr 0x20000ab8 state 0 router 0 IPv6 fe80::8c88:2bff:fe20:39d8 ll 8C:88:2B:20:39:D8 iface 0x200018f4/1
[00:14:01.531,707] <dbg> net_ipv6: net_ipv6_send_na: (rx_q[0]): Sent Neighbor Advertisement from 2001:db8::c4b2:5bff:fe9b:d45a to fe80::8c88:2bff:fe20:39d8, target 2001:db8::c4b2:5bff:fe9b:d45a iface 0x200018f4/1
[00:14:01.531,799] <dbg> net_ipv6: check_route: (rx_q[0]): Router 0x20005b90 nexthop fe80::8c88:2bff:fe20:39d8
[00:14:01.531,860] <dbg> net_ipv6: net_ipv6_prepare_for_send: (rx_q[0]): Neighbor lookup 0x20000ab8 (0) iface 0x200018f4/1 addr fe80::8c88:2bff:fe20:39d8 state static
[00:14:01.531,921] <dbg> net_ipv6: net_ipv6_prepare_for_send: (rx_q[0]): Neighbor 0x20000ab8 addr 8C:88:2B:20:39:D8
[00:14:01.531,951] <dbg> net_if: net_if_tx: (rx_q[0]): Processing (pkt 0x20010810, prio 1) network packet iface 0x200018f4/1
[00:14:01.531,982] <dbg> net_6lo: compress_da: (rx_q[0]): DAM_11 dst addr fully elided
[00:14:01.531,982] <dbg> net_6lo: compress_hoplimit: (rx_q[0]): HLIM compressed (255)
[00:14:01.532,012] <dbg> net_6lo: compress_tfl: (rx_q[0]): Traffic class and Flow label elided
[00:14:01.532,135] <dbg> net_core: processing_data: (rx_q[0]): Consumed pkt 0x20010ae0
 
```

carlescufi, 2022-09-23:
@jmrua thanks for reporting this. Note that IPSP unfortunately lacks a maintainer right now in Zephyr, so we might not be able to help out in a timely manner.

carlescufi, 2022-09-28:
@jmrua feel free to open a Pull Request if you have identified a fix for this.

jmrua, 2022-09-28:
@carlescufi After some more debugging, I think that the problem may be a problem with the [Bluetooth layer 2 code,](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/bluetooth/bluetooth.c) specifically in the send function ( bt_l2cap_chan_send). For the same payload, (The same case as above, TCP/UDP packet with the string "test" ) the return value for UDP with a destination link-local IP it returns **13**, with TCP it returns a **17** and with UDP and global IP as destination IP it returns **45**. Which, It is a bit odd that fluctuates that differences in bytes between the global UDP send and the link local, when the difference in the packet should only be the IP fields. If I'm able to figure out a fix, I will Pull Request it.

carlescufi, 2022-09-29:
> @carlescufi After some more debugging, I think that the problem may be a problem with the [Bluetooth layer 2 [...]

Thanks for the analysis! In that case I'll ask @jori-nordic to take a look, since he recently looked into the L2CAP internals.

jori-nordic, 2022-09-30:
@jmrua could you log the exact L2CAP payload? I'm thinking it differs between the packet types you listed.
A simple 
```c
	LOG_HEXDUMP_ERR(buf->data, buf->len, "l2cap payload");
```
inside `bt_l2cap_chan_send` (before the `return` ofc) should suffice.

The return value of chan_send is a bit cryptic, because it will sometimes return a lower value, even though it has queued the rest of the packet and will transmit it without needing further calls. I'll make a documentation PR to highlight this later.

jmrua, 2022-10-03:
@jori-nordic Here are the logs with your suggested addition

UDP Global IP:
```
uart:~$ net udp send 2001:db8::8c88:2bff:fe20:39d8 2222 "test"
Message sent
[00:08:17.503,784] <dbg> net_ctx: net_context_bind: (shell_uart): Context 0x20005bf0 binding to UDP [::]:49737 iface 1 (0x20001898)
[00:08:17.503,906] <dbg> net_ctx: context_setup_udp_packet: (shell_uart): context_setup_udp_packet
[00:08:17.504,089] <dbg> net_ipv6: net_ipv6_prepare_for_send: (shell_uart): Neighbor lookup 0x20000b0c (0) iface 0x20001898/1 addr 2001:db8::8c88:2bff:fe20:39d8 state stale
[00:08:17.504,119] <dbg> net_ipv6: net_ipv6_prepare_for_send: (shell_uart): Neighbor 0x20000b0c addr 8C:88:2B:20:39:D8
[00:08:17.504,150] <dbg> net_ipv6: ipv6_nbr_set_state: (shell_uart): nbr 0x20000b0c stale -> delay
[00:08:17.504,180] <dbg> net_if: net_if_tx: (shell_uart): Processing (pkt 0x20010710, prio 1) network packet iface 0x20001898/1
[00:08:17.504,486] <err> net_bt: l2cap payload
                                 ad 81 05 00 5a                                   |....Z            

[00:08:17.504,547] <dbg> net_if: net_if_tx: (shell_uart): Calling context send cb 0x20005bf0 status 45
[00:08:17.504,730] <dbg> net_ctx: net_context_unref: (shell_uart): Context 0x20005bf0 released
[00:08:22.504,272] <dbg> net_ipv6: ipv6_nd_reachable_timeout: (sysworkq): nbr 0x20000b0c moving 2001:db8::8c88:2bff:fe20:39d8 state to PROBE (4)
[00:08:22.504,302] <dbg> net_ipv6: ipv6_nd_reachable_timeout: (sysworkq): nbr 0x20000b0c probe count 1
[00:08:22.504,547] <dbg> net_ipv6: net_ipv6_send_ns: (sysworkq): Sent Neighbor Solicitation from 2001:db8::c4b2:5bff:fe9b:d45a to ff02::1:ff20:39d8, target 2001:db8::8c88:2bff:fe20:39d8 iface 0x20001898/1
[00:08:22.504,608] <dbg> net_if: net_if_tx: (sysworkq): Processing (pkt 0x20010710, prio 1) network packet iface 0x20001898/1
[00:08:22.504,730] <err> net_bt: l2cap payload
                                 0d 00 40 00 ff fe 20 39  d8 01 01 c4 b2 5b 9b d4 |..@... 9 .....[..
                                 5a                                               |Z                

```

UDP Link-Local:

```
uart:~$ net udp send fe80::8c88:2bff:fe20:39d8 2222 "test"
Message sent
[00:05:25.071,929] <dbg> net_ctx: net_context_bind: (shell_uart): Context 0x20005bf0 binding to UDP [::]:58176 iface 1 (0x20001898)
[00:05:25.072,052] <dbg> net_ctx: context_setup_udp_packet: (shell_uart): context_setup_udp_packet
[00:05:25.072,235] <dbg> net_ipv6: check_route: (shell_uart): Router 0x20005a98 nexthop fe80::8c88:2bff:fe20:39d8
[00:05:25.072,296] <dbg> net_ipv6: net_ipv6_prepare_for_send: (shell_uart): Neighbor lookup 0x20000ab8 (0) iface 0x20001898/1 addr fe80::8c88:2bff:fe20:39d8 state static
[00:05:25.072,326] <dbg> net_ipv6: net_ipv6_prepare_for_send: (shell_uart): Neighbor 0x20000ab8 addr 8C:88:2B:20:39:D8
[00:05:25.072,357] <dbg> net_if: net_if_tx: (shell_uart): Processing (pkt 0x20010710, prio 1) network packet iface 0x20001898/1
[00:05:25.072,387] <dbg> net_if: net_if_tx: (shell_uart): NET_IF_UP
[00:05:25.072,509] <err> net_bt: l2cap payload
                                 66 3a 66 65 32 30 3a 33  39 64 38 00 06 66 66
![udp](https://user-images.githubusercontent.com/82411856/193522286-021499e8-6a4d-47ed-a0fc-29e1527195ef.png)
 30 |f:fe20:3 9d8..ff0
                                 32 3a 3a                                         |2:: 
![tcp](https://user-images.githubusercontent.com/82411856/193522305-19a9aaa6-6f7e-4800-b83c-43cbe3c17d41.png)
             
[00:05:25.072,570] <dbg> net_if: net_if_tx: (shell_uart): Calling context send cb 0x20005bf0 status 13
[00:05:25.073,059] <dbg> net_ctx: net_context_unref: (shell_uart): Context 0x20005bf0 released

```

TCP:

```
uart:~$ net tcp send "test"
Message sent
[00:11:53.549,865] <dbg> net_ipv6: net_ipv6_prepare_for_send: (shell_uart): Neighbor lookup 0x20000b0c (0) iface 0x20001898/1 addr 2001:db8::8c88:2bff:fe20:39d8 state reachable
[00:11:53.549,926] <dbg> net_ipv6: net_ipv6_prepare_for_send: (shell_uart): Neighbor 0x20000b0c addr 8C:88:2B:20:39:D8
[00:11:53.549,957] <dbg> net_if: net_if_tx: (shell_uart): Processing (pkt 0x200105f0, prio 1) network packet iface 0x20001898/1
[00:11:53.550,262] <err> net_bt: l2cap payload
                                 69 87 05 00 07 02 00 00  59 82 05 00 e0 83 05 00 |i....... Y.......
                                 dd 25 00                                         |.%.              

```
I also attach here some Wireshark screenshots for the UDP Link-Local and TCP packets
UDP LL:
[image]
TCP:
[image]

jmrua, 2022-11-28:
Hello @jori-nordic, are you able to extract meaningful information from the message below?

jori-nordic, 2022-11-30:
Not really. I don't think the issue is in l2cap though, but I will try to reproduce it locally just in case.
What strikes me as odd is that for the first two samples (udp ll / global), only the LL seems to have the IP in the payload.

jmrua, 2022-12-09:
@rlubos Thank you for answering, I have upgraded my branch to master branch and I had the problem @jori-nordic reported above and using your PR seems to fix this. (Before I was testing with the last GitHub release and I didn't have connection issues)
But I'm having the same issue with the UDP packets in #52934. UDP global packets are not received in the PC. I have done some pcapng captures on TCP, UDP and ICMP. TCP and ICMP work as expected, these are just for comparison.
In the case of UDP link-local and global, I also attach a L2CAP capture, as for UDP global, there isn't any meaningful data in bt0 IP interface capture.

[test-ipsp-udp.zip](https://github.com/zephyrproject-rtos/zephyr/files/10196198/test-ipsp-udp.zip)

rlubos, 2022-12-12:
I'm afraid that I'm not able to help beyond #52934, I'm not well acquainted with BLE IPSP internals, especially given that I could not reproduce the issue on my end. Therefore, stepping down from the assignee role.

laurenmurphyx64, 2022-12-13:
@carlescufi Just assigning you to help figure out who to pass this to

jmrua, 2022-12-28:
Hello everyone, after analyzing the packets in my previous message, I think I have located the problem in the connection. The main difference between both Wireshark captures is that in the case of UDP LL the header is compressed, so the packet is considerably smaller, but in the case that global IPs are used the packet is not compressed, the IP packet is sent uncompressed and unmodified through the L2CAP channel. 

If we analyze the syslog of the host computer, the following message appears each time a UDP packet from the board arrives:
`received unknown nhc id which was not found`

From [ Linux 6LoWPAN next header compression](https://github.com/torvalds/linux/blob/master/net/6lowpan/nhc.c). 

What seems to be happening is that the host is waiting for the compressed packet to arrive in the format defined in [RFC 6282](https://www.rfc-editor.org/rfc/rfc6282) and the nRF board is sending it as a flat IPv6 packet through the L2CAP channel, so the expected headers([LOWPAN_NHC format](https://www.rfc-editor.org/rfc/rfc6282#section-4.3.3)) are not there. So the host is reading data that does not make sense and is not able to retrieve the original IP packet.

In view of [RFC 7668](https://www.rfc-editor.org/rfc/rfc7668), which indicates that IPv6 header compression must be performed (“Per this specification, the IPv6 header compression format specified in [RFC 6282] MUST be used.”)

It appears that this compression and header addition is not being done for UDP traffic with global IPs in the current versions of Zephyr.

It appears that this compression and header appending is not being performed for UDP traffic with global IPs in the verions and yet it is being performed correctly for UDP Link Local traffic.

irgendwie, 2022-12-29:
We were able to reproduce this issue on **some** of our nRF**5340**DK boards. While most of them work correctly, some don't. They show the exact same behavior as described in this issue.

jori-nordic, 2023-02-23:
@rlubos doesn't the last message from @jmrua mean that it's actually the zephyr network layer that's buggy in this case?

rlubos, 2023-02-23:
> he host is waiting for the compressed packet to arrive in the format defined in [RFC 6282](https://www.rfc-editor.org/rfc/rfc6282) and the nRF board is sending it as a flat I

Yes, it could indicate some issue with 6lowpan IPHC, although I'm no really expert in this area to comment further.

bamkrs, 2023-02-23:
'm currently onto this, too.
The UDP packet is correctly routed through `6lo.c:compress_nh_udp` which can not compress ports and inlines them (`6lo.c:673`).
The `inline_ptr_udp` also points to seemingly good looking buffer.

closed 2023-08-13 by github-actions[bot]