zephyrproject-rtos/zephyr #50603: Upgrade to loramac-node 4.7.0 when it is released to fix async LoRa reception on SX1276

dberlin, 2022-09-23:
Right now, asynchronous LoRa reception on the SX1276 is broken because the payload buffer gets memset before the callback is run.
[This pull request](https://github.com/zephyrproject-rtos/zephyr/pull/48765) has more details.

A patch was submitted upstream to loramac-node to fix the issue there (rather than change zephyr's handling), and was accepted. It is targeted at the 4.7.0 release of loramac-node, so zephyr will need to upgrade once released in order to fix this bug.

I'm just opening an issue so it doesn't get lost.

Once 4.7.0 is released, i'm happy to prepare a pull request to do the update for zephyr.

JordanYates, 2022-09-24:
It is not necessarily required to wait for 4.7 to be released in order to fix the bug. In the past we have cherry-picked official commits that fix issues back to our current Zephyr branch. Unfortunately the hard freeze deadline for v3.2 has passed and I don't think this qualifies as a blocker-bug, but we would be able to fix it as soon as main opens up again.

mmahadevan108, 2022-09-27:
@Mani-Sadhasivam, if you decide to cherry-pick this fix into the Lora module, then we might be able to get this in for 3.2 as this would be a bug fix.

dberlin, 2022-09-27:
FWIW - it's not a regression in the strictest sense because it never worked.
It has been broken on the 1276 since async receive was added :)

But it's definitely a bug fix, and it will be surprising to anyone who tries to use LoRa with zephyr, since it affects 100% of async receives on the sx127x.  The sx127x chips (which silently fail) are *much* more popular than the sx126x series (which works).  

At a minimum, if it's not fixed, it should be noted as a known issue somewhere so people know to avoid it

Mani-Sadhasivam, 2022-09-28:
@mmahadevan108  Done! https://github.com/zephyrproject-rtos/zephyr/pull/50733