zephyrproject-rtos/zephyr #50614: Zephyr if got the ip is "10.xxx.xxx.xxx" when join in the switchboard, then the device may can not visit the outer net, also unable to Ping.

GasperChen, 2022-09-24:
**Describe the bug**
Zephyr if got the ip is "10.xxx.xxx.xxx" when join in the switchboard, then the device may can not visit the outer net, also unable to Ping.

Please also mention any information which could help others to understand
the problem you're facing:

 - NXP imxrt1061, PHY RTL8201F
 - ZEPHYR 2.6.99
 - （128.1.0.0-191.255.0.0）and （192.0.1.0-223.255.255.0） is ok.

There is the ip can make the issue:
10.70.100.133
10.4.239.xxx

******I have report this issue one months ago, but there is not any info to answered.  Please test this issue，TKS！*******

stephanosio, 2022-09-24:
I have many devices running Zephyr in our intranet that are assigned 10.0.0.0/8 addresses and they work just fine.

You will need to provide more information than what you have provided for us to identify the issue (e.g. device log, packet dump, your network configuration).

cc @rlubos

GasperChen, 2022-09-24:
> I have many devices running Zephyr in our intranet that are assigned 10.0.0.0/8 addresses and they work just fine.
> 
> You will need to provide more information than what you have provided for us to identify the issue (e.g. device log, packet dump, your network configuration).
> 
> cc @rlubos

Can you test the IP: 10.70.100.133 when you devices got IP from switchboard, not got the IP from router. and what is you os version. 
[Uploading log.TXT…]()

GasperChen, 2022-09-24:
> I have many devices running Zephyr in our intranet that are assigned 10.0.0.0/8 addresses and they work just fine.
> 
> You will need to provide more information than what you have provided for us to identify the issue (e.g. device log, packet dump, your network configuration).
> 
> cc @rlubos

[01-01 00:00:00]DriverUart_init,183: Success
[01-01 00:00:00]DriverGpio_init,866: Success
[01-01 00:00:00]DriverRtc_init,372: Success
*** Booting Zephyr OS version 2.6.99  ***
*** J01_Boot_0 Jun  2 2022 09:51:20 ***
[01-01 00:00:00]GwBoot_loadBootloader,1272: Success
[01-01 00:00:00]DriverUart_init,183: Success
[01-01 00:00:00]DriverGpio_init,866: Success
[01-01 00:00:00]DriverRtc_init,372: Success
[01-01 00:00:00]DriverNorFlash_init,1747: Success
*** Booting Zephyr OS version 2.6.99  ***
[01-01 00:00:00]main,33: Bootloader start...
[01-01 00:00:00]main,34: Bootloader start...
[01-01 00:00:00]main,35: Bootloader start...
*** J01_Boot_V1.1.0 Apr 20 2022 10:17:47 ***
[01-01 00:00:00]GwBoot_readBootParamFromFlash,101: Success
[01-01 00:00:00]GwBoot_show,454: eCurAppNum:                     1
[01-01 00:00:00]GwBoot_show,456: astImageInfo[0].ifNewOne:       0
[01-01 00:00:00]GwBoot_show,457: astImageInfo[0].ifInvalid:      0
[01-01 00:00:00]GwBoot_show,458: astImageInfo[0].uiTotalSize:    2773 KB
[01-01 00:00:00]GwBoot_show,469: astImageInfo[0].Record[0]:      1970-01-01 00:03:11
[01-01 00:00:00]GwBoot_show,469: astImageInfo[0].Record[1]:      1970-01-01 00:00:03
[01-01 00:00:00]GwBoot_show,469: astImageInfo[0].Record[2]:      1970-01-01 00:00:31
[01-01 00:00:00]GwBoot_show,469: astImageInfo[0].Record[3]:      1970-01-01 00:01:00
[01-01 00:00:00]GwBoot_show,474: astImageInfo[1].ifNewOne:       0
[01-01 00:00:00]GwBoot_show,475: astImageInfo[1].ifInvalid:      0
[01-01 00:00:00]GwBoot_show,476: astImageInfo[1].uiTotalSize:    2773 KB
[01-01 00:00:00]GwBoot_show,487: astImageInfo[1].Record[0]:      1970-01-01 00:03:12
[01-01 00:00:00]GwBoot_init,523: eCurAppNum = 1
!?盏?1??F琮剄瓖!刟?!菎!?4睌?5%1@攧1%牨%?5%1佸?%@肀!縈!?醯 5瓖[01-01 00:00:03]DriverUart_configBaudrate,500: Success
[01-01 00:00:03]GwBoot_writeBootParamToFlash,53: Success
[01-01 00:00:03]GwBoot_init,636: Success
[01-01 00:00:03]GwBoot_loadImage,717: Success
[01-01 00:00:00]DriverUart_init,183,[]: Success
[01-01 00:00:00]DriverGpio_init,939,[main]: Success
[01-01 00:00:04]DriverRtc_init,372,[main]: Success
[01-01 00:00:04]DriverNorFlash_init,1747,[main]: Success

[1;32muart:~$ [m[01-01 00:00:04]DriverNandFlash_checkBlocksMgtInFlash,1526,[main]: Success
[01-01 00:00:05]DriverNandFlash_checkWholeSpareArea,1021,[main]: Success
[01-01 00:00:05]DriverNandFlash_init,2065,[main]: Success
[01-01 00:00:05]GwTest_readTestParamFromFlash,191,[main]: Success
[8D[J[00:00:01.394,007] [0m<dbg> net_core: net_init: (main): Priority 90[0m
[1;32muart:~$ [m[8D[J[00:00:01.402,589] [0m<dbg> net_pkt: net_pkt_init: (main): Allocating 36 RX (2304 bytes), 36 TX (2304 bytes), 128 RX data (0 bytes) and 128 TX data (0 bytes) buffers[0m
[1;32muart:~$ [m[8D[J[00:00:01.419,820] [0m<dbg> net_tcp: net_tcp_init: (main): Workq started. Thread ID: 0x81006470[0m
[1;32muart:~$ [m[8D[J[00:00:01.430,762] [0m<dbg> net_core: l3_init: (main): Network L3 init done[0m
[1;32muart:~$ [m[8D[J[00:00:01.440,093] [0m<dbg> net_mgmt: net_mgmt_event_init: (main): Net MGMT initialized: queue of 2 entries, stack size of 768[0m
[1;32muart:~$ [m[8D[J[00:00:01.453,785] [0m<dbg> net_if: net_if_init: (main): [0m
[1;32muart:~$ [m[8D[J[00:00:01.461,517] [0m<dbg> net_tc: net_tc_tx_init: (main): No TX thread created[0m
[1;32muart:~$ [m[8D[J[00:00:01.471,243] [0m<dbg> net_if: init_iface: (main): On iface 0x81001d00[0m
[1;32muart:~$ [m[8D[J[00:00:01.480,550] [0m<dbg> net_ethernet: ethernet_init: (main): Initializing Ethernet L2 0x81037778 for iface 0x81001d00[0m
[1;32muart:~$ [m[8D[J[00:00:01.493,835] [0m<dbg> net_ethernet: ethernet_init: (main): [0] alloc ctx 0x81037778 iface 0x81001d00[0m
[1;32muart:~$ [m[8D[J[00:00:01.520,758] [0m<dbg> net_if: init_iface: (main): On iface 0x81001d48[0m
[1;32muart:~$ [m[8D[J[00:00:01.544,961] [0m<dbg> net_ethernet: ethernet_init: (main): Initializing Ethernet L2 0x81037778 for iface 0x81001d48[0m
[1;32muart:~$ [m[8D[J[00:00:01.573,168] [0m<dbg> net_ethernet: ethernet_init: (main): [1] alloc ctx 0x81037778 iface 0x81001d48[0m
[1;32muart:~$ [m[8D[J[00:00:01.600,062] [0m<dbg> net_if: init_iface: (main): On iface 0x81001d90[0m
[1;32muart:~$ [m[8D[J[00:00:01.624,261] [0m<dbg> net_ethernet: ethernet_init: (main): Initializing Ethernet L2 0x81037778 for iface 0x81001d90[0m
[1;32muart:~$ [m[8D[J[00:00:01.652,472] [0m<dbg> net_ethernet: ethernet_init: (main): [2] alloc ctx 0x81037778 iface 0x81001d90[0m
[1;32muart:~$ [m[8D[J[00:00:01.679,363] [0m<dbg> net_if: init_iface: (main): On iface 0x81001dd8[0m
[1;32muart:~$ [m[8D[J[00:00:01.703,563] [0m<dbg> net_ethernet: ethernet_init: (main): Initializing Ethernet L2 0x81037778 for iface 0x81001dd8[0m
[1;32muart:~$ [m[8D[J[00:00:01.731,774] [0m<dbg> net_ethernet: ethernet_init: (main): [3] alloc ctx 0x81037778 iface 0x81001dd8[0m
[1;32muart:~$ [m[8D[J[00:00:01.758,667] [0m<dbg> net_if: init_iface: (main): On iface 0x81001e20[0m
[1;32muart:~$ [m[8D[J[00:00:01.782,865] [0m<dbg> net_ethernet: ethernet_init: (main): Initializing Ethernet L2 0x81037778 for iface 0x81001e20[0m
[... cut ...]
[1;32muart:~$ [m[8D[J[00:00:07.253,612] [0m<dbg> net_pkt: net_pkt_unref_debug: (sysworkq): TDATA (tx_bufs) [127] frag 0x817cfa54 ref 0 frags (nil) (arp_entry_cleanup():43)[0m
[1;32muart:~$ [m[8D[J[00:00:07.284,359] [0m<dbg> net_pkt: net_pkt_frag_unref_debug: (sysworkq): TDATA (tx_bufs) [127] frag 0x817cfa54 ref 0 (net_pkt_unref_debug():586)[0m
[1;32muart:~$ [m[8D[J[00:00:07.314,746] [0m<dbg> net_buf: net_buf_unref_debug: (0x8101f880) buf 0x817cfa54 ref 1 pool_id 4 frags (nil)[0m
[1;32muart:~$ [m

GasperChen, 2022-09-24:
[image]

maweicoltsmart, 2022-09-26:
@GasperChen Author
I encountered a similar error, my solution is to make CONFIG_NET_UDP_CHECKSUM=n, it seems that the check code is wrong。

The issue I created is #49811

rlubos, 2022-09-27:
> Can you test the IP: 10.70.100.133 when you devices got IP from switchboard, not got the IP from router. and what is you os version.

I don't understand what do you mean by "when you devices got IP from switchboard". Anyway, I've tried with `qemu_x68`, reconfigured my local network so that qemu lies within `10.0.0.0/8` subnet, and statically configured `10.70.100.133` on my Zephyr device - the communication with internet works just fine. I really doubt there's some Zephyr issue that prevents network communication within a specific address range, right now it more looks like a network configuration on your side.

Please, either be more specific about the problem (from the logs you've posted I can only tell that DNS query timed out) or provide a minimal sample to reproduce the issue (ideally in emulated environment like `qemu_x86` or `native_posix`, I don't have access to every board supported by Zephyr (and you seems to use a custom one anyway).

carlescufi, 2022-10-05:
@GasperChen please provide additional information, including your exact network setup and, if possible, a reproducible sample so we can debug it.

closed 2022-10-19 by carlescufi