zephyrproject-rtos/zephyr #50615: ESP32 GPIO driver

sam131208, 2022-09-24:
**Describe the bug**
The input configuration of GPIO34~GPIO39 cannot be performed. When the following code block is executed, it ends abnormally.

	if (flags & GPIO_PULL_UP) {
		if (!rtc_gpio_is_valid_gpio(io_pin) || SOC_GPIO_SUPPORT_RTC_INDEPENDENT) {
			gpio_ll_pullup_en(&GPIO, io_pin);
		} else {
#if SOC_RTCIO_INPUT_OUTPUT_SUPPORTED
			int rtcio_num = rtc_io_num_map[io_pin];

			if (rtc_io_desc[rtcio_num].pullup) {
				rtcio_hal_pullup_enable(rtcio_num);
			} else {
				ret = -ENOTSUP;
				goto end;
			}
#endif
		}
	} else {
		if (!rtc_gpio_is_valid_gpio(io_pin) || SOC_GPIO_SUPPORT_RTC_INDEPENDENT) {
			gpio_ll_pullup_dis(&GPIO, io_pin);
		} else {
#if SOC_RTCIO_INPUT_OUTPUT_SUPPORTED
			int rtcio_num = rtc_io_num_map[io_pin];

			if (rtc_io_desc[rtcio_num].pulldown) {
				rtcio_hal_pullup_disable(rtcio_num);
			} else {
				ret = -ENOTSUP;
				goto end;
			}
#else
			ret = -ENOTSUP;
			goto end;
#endif
		}
	}

henrikbrixandersen, 2022-09-26:
@sam131208 Please revisit the description of this issue and fix the formatting to make it readable.

sam131208, 2022-09-26:
@henrikbrixandersen format fixed。

sylvioalves, 2022-09-27:
@sam131208  thanks for reporting. Indeed there is an issue in there.

PR #50725 (2022-09-28): drivers: gpio: esp32: fix error when using input only pins
Input pins in ESP32 (34 to 39) have no pull-up / pull-down internally. Current behavior is that gpio driver fails when "disabling it", which should not happen. It need to fail only when enabling this feature.

Fixes #50615
commit 82bd3410 (2022-09-28): drivers: gpio: esp32: fix error when using input only pins

Input pins in ESP32 (34 to 39) have no pull-up / pull-down internally.
Current behavior is that gpio driver fails when "disabling it", which
should not happen. It need to fail only when enabling this feature.

Fixes #50615
closed 2022-09-28 by fabiobaltieri