zephyrproject-rtos/zephyr #50619: tests/kernel/timer/starve fails to run on devices

nashif, 2022-09-24:
**Describe the bug**
This test is marked build_only and is never run by any device. Trying to run it on a device results in an MPU fault.

**To Reproduce**
west build -b nrf52840dk_nrf52811
west flash 

**Expected behavior**
Test runs

**Impact**
Bitrot, duplication? Maybe this can be folded into or removed into some other tests? timer_behaviour maybe?

**Logs and console output**
```
E: ***** MPU FAULT *****
E:   Instruction Access Violation
E: r0/a1:  0xE: ***** HARD FAULT *****
E:   Fault escalation (see below)
E: ***** MPU FAULT *****
E:   Instruction Access Violation
E: r0/a1:  0x000003e7  r1/a2:  0x00000000  r2/a3:  0x00000010
E: r3/a4:  0x00000000 r12/ip:  0x00000048 r14/lr:  0x0000678d
E:  xpsr:  0x61000004
E: Faulting instruction address (r15/pc): 0x000000b8
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Fault during interrupt handling

E: Current thread: 0x20000cd0 (�)
E: Halting system

```

anangl, 2022-09-27:
For which commit was this issue observed? I cannot reproduce it with the recent `main`. For c4cc1621a5a2fc6d99287fe22aed324ba4e4e803 on nRF52840 DK (target `nrf52840dk_nrf52811`) I got the following output:
```
*** Booting Zephyr OS build v3.2.0-rc3-57-gc4cc1621a5a2  ***
Running TESTSUITE starve_fn
===================================================================
START - test_starve
Cycle clock runs at 32768 Hz
There are 1 cycles per tick (32768 Hz ticks)
[     0.399] 1: still running, would pass at 3600 s
[    60.070] 600: still running, would pass at 3600 s
[   120.058] 1202: still running, would pass at 3600 s
[   180.042] 1804: still running, would pass at 3600 s
[   240.024] 2406: still running, would pass at 3600 s
[   300.002] 3008: still running, would pass at 3600 s
[   360.088] 3611: still running, would pass at 3600 s
[   420.068] 4213: still running, would pass at 3600 s
[   480.065] 4815: still running, would pass at 3600 s
[   540.039] 5417: still running, would pass at 3600 s
[   600.010] 6019: still running, would pass at 3600 s
[   660.062] 6622: still running, would pass at 3600 s
[   720.011] 7224: still running, would pass at 3600 s
[   780.052] 7827: still running, would pass at 3600 s
[   840.096] 8430: still running, would pass at 3600 s
[   900.046] 9032: still running, would pass at 3600 s
[   960.080] 9635: still running, would pass at 3600 s
[  1020.027] 10237: still running, would pass at 3600 s
[  1080.063] 10840: still running, would pass at 3600 s
[  1140.027] 11442: still running, would pass at 3600 s
[  1200.087] 12045: still running, would pass at 3600 s
[  1260.035] 12647: still running, would pass at 3600 s
[  1320.097] 13250: still running, would pass at 3600 s
[  1380.060] 13852: still running, would pass at 3600 s
[  1440.024] 14454: still running, would pass at 3600 s
[  1500.072] 15057: still running, would pass at 3600 s
[  1560.020] 15659: still running, would pass at 3600 s
[  1620.064] 16262: still running, would pass at 3600 s
[  1680.003] 16864: still running, would pass at 3600 s
[  1740.040] 17467: still running, would pass at 3600 s
[  1800.090] 18070: still running, would pass at 3600 s
[  1860.007] 18672: still running, would pass at 3600 s
[  1920.031] 19275: still running, would pass at 3600 s
[  1980.054] 19878: still running, would pass at 3600 s
[  2040.095] 20481: still running, would pass at 3600 s
[  2100.030] 21083: still running, would pass at 3600 s
[  2160.071] 21686: still running, would pass at 3600 s
[  2220.006] 22288: still running, would pass at 3600 s
[  2280.053] 22891: still running, would pass at 3600 s
[  2340.097] 23494: still running, would pass at 3600 s
[  2400.048] 24096: still running, would pass at 3600 s
[  2460.090] 24699: still running, would pass at 3600 s
[  2520.021] 25301: still running, would pass at 3600 s
[  2580.043] 25904: still running, would pass at 3600 s
[  2640.082] 26507: still running, would pass at 3600 s
[  2700.033] 27109: still running, would pass at 3600 s
[  2760.075] 27712: still running, would pass at 3600 s
[  2820.020] 28314: still running, would pass at 3600 s
[  2880.058] 28917: still running, would pass at 3600 s
[  2940.016] 29519: still running, would pass at 3600 s
[  3000.083] 30122: still running, would pass at 3600 s
[  3060.035] 30724: still running, would pass at 3600 s
[  3120.085] 31327: still running, would pass at 3600 s
[  3180.017] 31929: still running, would pass at 3600 s
[  3240.053] 32532: still running, would pass at 3600 s
[  3300.004] 33134: still running, would pass at 3600 s
[  3360.052] 33737: still running, would pass at 3600 s
[  3420.096] 34340: still running, would pass at 3600 s
[  3480.049] 34942: still running, would pass at 3600 s
[  3540.096] 35545: still running, would pass at 3600 s
[  3600.059] 36147: still running, would pass at 3600 s
[  3601.060] 36156: Completed 36156 iters without failure
 PASS - test_starve in 3600.673 seconds
===================================================================
TESTSUITE starve_fn succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [starve_fn]: pass = 1, fail = 0, skip = 0, total = 1 duration = 3600.673 secons
 - PASS - [starve_fn.test_starve] duration = 3600.673 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

nashif, 2022-09-28:
tried again with latest main commit and I do not have the issue, so something went wrong or I had a dirty tree.

closed 2022-09-28 by nashif