zephyrproject-rtos/zephyr #50621: The history of the multi API / MFD discussions 2022 July - Sep

bjarki-trackunit, 2022-09-24:
## Intruduction
This issue will go through the entire MFD / multi API device model discussions starting in with the modem driver design issue. Please note that this is summarizing multiple PRs and issues with about 100 comments across them, some details will be left out.

## The issue with modem driver design July 2022
Modems often contain multiple functions, which are not common between all modems. The BG96 contains a cellular modem, GNSS modem and GPIO controller. Each of these modules support multiple features. A simple graph of features is shown below:

```
BG96 {
    DFOTA (download firmware over the air)

    Cellular modem {
        network interface
        dtm (direct test mode)
        cellular info (RSSI, IMEI, towers)
        SMS
    }

    GNSS {
        location (lat, long, speed, altitude etc.)
        gnss (tracking mode, satelite count, power modes, NMEA frames)
    }

    QuecLocator (cellular location) {
        location (lat, long, speed, altitude etc.)
    }
}
```

This is the basis for the need for multiple APIs since the **BG96** contains multiple functions, which may contain multiple APIs each.
This issue was brought up at the architecture meeting, and no solution to the problem was suggested at that time.

## The MFD devicetree parent/child solution July 2022 https://github.com/zephyrproject-rtos/zephyr/pull/47655

This solution to the problem is simple, just add every function or feature which requires its own API, be it an actual device or just a device struct being used as a handle, to the devicetree. Then instanciate a single driver for the top parent node which instanciates device instances for all features (child devices).

The issue proposed design rules and some macros which would help to keep the design and implementations of MFDs consistent.

```
BG96 {
    compatible = "quectel,bg96"

    cellular info {
        compatible = "quectel,bg96-cellinfo";
    };

    sms {
        compatible = "quectel,bg96-sms";
    };

    dtm {
        compatible = "quectel,bg96-dtm";
    };

    gnss {
        compatible = "quectel,bg96-gnss";
    };
};
```

The summed up opposing opinions  to this solution at the time was:
* This solution is already in use in-tree
* We should consider if the current device model is sufficient for MFDs moving forwards.
* This solution deviates from the linux driver design as they support multiple APIs pr node.
* features like sms and dtm are not actual devices, they are purely software, so they should not be in devicetree as nodes

The last opposing opinion sparked the research into multiple APIs pr device node.

## The runtime API lookup solution Aug 9 2022 https://github.com/zephyrproject-rtos/zephyr/pull/48817
This solution pulled out the API pointer from the struct device, and placed it in an iterable section based on API type along
with a pointer to the device struct it belongs to.
[... 134 lines cut ...]
This solution solves
* Multi API devices

This solution requires
* Optional addition to docs
* Optional macros to help with device driver instantiation

Issues with this solution
* Places software constructs in devicetree
* Uses the most ROM
* Only works for devices in devicetree

## Properties files PR Sep 2022 https://github.com/zephyrproject-rtos/zephyr/pull/50441

The supporting responses to the properties files on their own:
* May contain metadata regarding drivers that should not be in devicetree bindings files
* Are easy to integrate into CI
* May be used for declaring APIs

The opposing responses to the properties files on their own are:
* To complex to only be used to declare APIs and metadata
* Some properties might be declared in Kconfig instead
* Some properties might be declared in bindings files instead
* Some properties might be declared in source files and extracted by scrubbing elf files instead
* There are not enough features that need the functionality brought by the properties files at this time to justify adding them

bjarki-trackunit, 2022-09-25:
## Solution Bjarki

### Proposed solution
Using the api_device structs model, placing the api / device type information in the bindings files, generating macros using gen_defines.py / edtlib.py.

### Supporting remarks
* Only adds 1 pointer pr API to ROM if struct device pointer is removed.
* No runtime lookup of devices
* Type safety through specifying device type in `DEVICE_API_GET`

### Requirements
* Requires rework of PM device subsystem
* Requires updating DEVICE_DEFINEs
* Requires updating edtlib

### Further arguments
The PM device subsystem is already remarked as being flawed and will be updated in the future in any case. For single API devices, the PM subsystem works as expected, it also works with multi API devices, but it is awkward.

The update of DEVICE_DEFINEs can be done gradually, and only require changing the api_ptr argument, which allows for an easy transition to this model.

The **api:** key in the bindings files is nearly identical to the Linux **device_type:** property, we could even call it device_type, but it is not as clear as **api:**. So there is precedent for having that information in the devicetree bindings files.

gmarull, 2022-09-26:
> ## The MFD devicetree parent/child solution July 2022 #47655
> This solution to the problem is simple, just add every function or feature which requires its own API, be it an actual device or just a [...]

It is used e.g. by STM32 timers (specific functions of the timer are represented as child nodes: `pwm`, `qdec`, etc. Same approach is used in Linux, so we're not deviating here. Also used now by GD32 RCU (Reset and Clock Unit).

> * We should consider if the current device model is sufficient for MFDs moving forwards.
> * This solution deviates from the linux driver design as they support multiple APIs pr node.

The main problem I see in Zephyr really is that we use DT node identifiers (node labels, paths, aliases, etc.) to obtain `struct device` instances. This forces us to have a 1:1 relationship between DT nodes and `struct device` objects. This is the main difference with Linux, I guess. It's not really a problem, but a design choice that has important consequences on how we use Devicetree. For example, node labels are used in DT to reference to a node, e.g. in `pwms = <&pwm1 0 ...>;` but that's it. In my view, the fact we use them to obtain _software_ devices means we're in practice using DT as a compile time source of software configuration.

> * features like sms and dtm are not actual devices, they are purely software, so they should not be in devicetree as nodes

They could also be seen as hardware blocks of the modem (some definitely are, like GPIO), but it's true that some cases may be at the edge. But as I said above, we don't use Devicetree as Linux does, and I think we need to be pragmatic in the end. Is it worth breaking the whole device model? Or is it better to allow for certain exceptions in the case of MFDs (allow for _Zephyrisms_)?

I solved the GD32 RCU case this way:

https://github.com/zephyrproject-rtos/zephyr/blob/e348fe494a21aad2bfd5744e0e2ce57262a24651/dts/arm/gigadevice/gd32f4xx/gd32f4xx.dtsi#L39-L55

RCU is a HW block that has both reset and clock control registers. I could just describe the RCU in DT (as ST does on Linux with RCC, same/similar IP). But to my understanding, it's not inaccurate to add a couple of sub-nodes to represent the reset and clock control blocks.

> 
> The last opposing opinion sparked the research into multiple APIs pr device node.
> 

I get the point, but I think we need to be careful on breaking the device model for what it seems to be an edge case that has, in my view, a viable solution with the existing model.

bjarki-trackunit, 2022-09-26:
@gmarull Sorry, but could you copy your comment to this discussion https://github.com/zephyrproject-rtos/zephyr/discussions/50641 ? I have created a copy in the discussions section instead of the issues section. I tried to close this issue but it did not work it seems :)

closed 2022-09-26 by bjarki-trackunit

pdgendt, 2022-09-26:
FYI; there is a "Convert to discussion" action for issues.

bjarki-trackunit, 2022-09-26:
> FYI; there is a "Convert to discussion" action for issues.

Ha, of course there is :)