zephyrproject-rtos/zephyr #50652: RAM Loading on i.MXRT1160_evk

Sathishkumar-rk, 2022-09-26:
**Describe the bug**
   
   --> Want to run zephyr application from SDRAM location using RAM Loading feature.

**Information**

- Build the code for SDRAM using CONFIG_CODE_SEMC=y. However, when I test with this, I see when z_bss_zero() is called in the startup code, it trashes the code memory in SDRAM. 

- So it seems the linker settings for this RT1160 are not configured for CONFIG_CODE_SEMC=y, and allow the BSS section to overlap code in SDRAM.

**Expected behavior**
  
   --> Copy the content from FLASH to given SDRAM location and execute from SDRAM.

Thanks,
Sathishkumar K

DerekSnell, 2022-09-26:
Adding that @Sathishkumar-rk has additional details on this use-case using MCUboot at https://github.com/zephyrproject-rtos/zephyr/discussions/48242

SanthoshRag, 2023-01-20:
@mmahadevan108 @DerekSnell 
Is there an update on this Application loading from SDRAM location? We are critically in need of this RAM loading functionality as we are planning to add & verify few other features (eg: Encryption). Kindly support at the earliest.

DerekSnell, 2023-01-20:
Hi @SanthoshRag ,
The NXP i.MX RT devices have high performance XIP capability directly from flash.  It is not that common for applications to execute from external SDRAM.  That is why this issue has been labeled as low priority.

NXP appreciates all the improvements and issues posted here from the community.  And we do our best to address them in a timely manner.  But with finite resources, I do not have a schedule right now when this feature will be supported.

If this is critical for you, we would gladly support your contribution to the Zephyr project to add this capability.  

Best regards

SanthoshRag, 2023-01-21:
Hi @DerekSnell 
Thanks for your immediate reply and appreciate the support you all have been providing. 
Sure, we would like to support & contribute to Zephyr projects as you had mentioned. We need some insights from NXP members to approach this SDRAM issue? Any inputs here would be appreciated. 

Initially, we were trying to use the XIP flash feature of the RT platform and even started including the OTFAD Encryption capabilities outside the zephyr binary builds. We were able to apply the OTFAD Encryption on the MCUBoot, but when trying for the Application level, we encountered the firmware upgrade methods (SWAP/SCRATCH) which required Swapping of Images between slots which was a challenge for applying OTFAD. 
Still, we can make the OTFAD module work by modifying the MCUBoot code to support multiple SW reset during TEST & REVERT commands, but we are trying to see the possibility of controlling the OTFAD module enabling/disabling at runtime so as to swap encrypted images whenever required.

Moreover, our intention was to have minimum or no modifications to the MCUBoot code to support the OTFAD Encryption/Decryption capabilities.

That is why we intend to use the RAM loading capabilities such that the images prepared and stored in the flash slots can be applied with OTFAD encryption and on execution the OTFAD Decryption will apply on the image available on the flash and later upon the MCUBoot execution, it will check the higher version application image and transfer it to the RAM(SDRAM) location for execution.

DerekSnell, 2023-01-24:
Hi @SanthoshRag ,

Thanks for the explanation.  Here is the [linker script file](https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/linker_script/arm/linker.cmake) that places the code in "flash" and the data in "ram", using the Kconfig symbols.

When I use the command below to place the code in SEMC...
```
west build -b mimxrt1160_evk_cm7 samples\hello_world --pristine -- -DCONFIG_CODE_SEMC=y
```

and then look at the generated `.config` file in <build_dir>/zephyr, I see the following:
```
CONFIG_SRAM_BASE_ADDRESS=0x80000000
CONFIG_FLASH_BASE_ADDRESS=0x80000000
```

That is causing the overlap you described, as both code and data is written to the SDRAM base address.  To remedy, you will want to move one of these based off `CONFIG_CODE_SEMC`, `CONFIG_FLASH_SIZE` or `CONFIG_SRAM_SIZE`, and avoid any overlap.

Best regards

Sathishkumar-rk, 2023-02-02:
Hi @danieldegrasse,

Yes, as you mentioned if we do the with flag `-DCONFIG_CODE_SEMC=y` base address is same for `SRAM & FLASH`.

```
CONFIG_FLASH_BASE_ADDRESS=0x80020000
CONFIG_SRAM_BASE_ADDRESS=0x80020000
```

we are changing base address using menu configuration for flash and sram, after changes base address changes as follows
```
CONFIG_FLASH_BASE_ADDRESS=0x30000000
CONFIG_SRAM_BASE_ADDRESS=0x80020000
```

while running code getting the below halt error
```
*** Booting Zephyr OS build zephyr-v3.2.0-3843-g3b7e7575a926 ***
I: Starting bootloader
I: Primary   slot: version=9.8.7+6
I: Image 0 Secondary slot: Image not found
I: RAM loading to 0x80020000 is succeeded.
I: Image 0 loaded from the primary slot
I: Bootloader chainload address offset: 0x20000
I: Jumping to the first image slot
E: ***** MPU FAULT *****
E:   Instruction Access Violation
E: r0/a1:  0x80004f5c  r1/a2:  0x0000000a  r2/a3:  0x00000010
E: r3/a4:  0x80022255 r12/ip:  0x00000000 r14/lr:  0x300088f3
E:  xpsr:  0x81000000
E: Faulting instruction address (r15/pc): 0x80022254
E: >>> ZEPHYR FATAL ERROR 20: Unknown error on CPU 0
E: Current thread: 0x80000130 (unknown)
E: Halting system
````

we didn't modified the linker file as of now, if we get some insights about linker file functionality will do the modifications as required.

Thanks,
Sathishkumar K

Sathishkumar-rk, 2023-02-07:
Hi @DerekSnell ,

Thanks for the update.

Meanwhile we carried our experiment as below and adding information here for your reference.

`
As a trial we are tried to use the ITCM location instead of SDRAM, our observation is content is copied in the given region but execution is not happening. In this case MPU fault is not happening and added config files which used for building.
`
[blinky.zip](https://github.com/zephyrproject-rtos/zephyr/files/10675245/blinky.zip)

```
*** Booting Zephyr OS build zephyr-v3.2.0-3843-g3b7e7575a926 ***
I: Starting bootloader
I: Primary   slot: version=2.3.4+5
I: Image 0 Secondary slot: Image not found
I: RAM loading to 0x10000 is succeeded.
I: Image 0 loaded from the primary slot
I: Bootloader chainload address offset: 0x20000
I: Jumping to the first image slot
I: Address foot print after image copy into ITCM memory
I: vector msp address 0x18dc0 : reset address 0x12281 : load address 0x10000
```

Regards,
Sathishkumar K

Sathishkumar-rk, 2023-04-11:
I would like to have the label removed

jwtaylor24, 2023-04-13:
I am also trying to RAM load an application using MCUBoot and would like to see some support added.

Side question - I always assumed executing in-place from flash would result in a huge performance hit. Unfortunately, I think my application is too large to fit into ITCM/DTCM memories... so, I assumed SDRAM was the "next best thing". Will I see a noticable difference executing from SDRAM vs. QSPI flash? Or should I just continue to execute from flash?

Thanks!

DerekSnell, 2023-04-17:
Hi @jwtaylor24 ,

NXP's Appnote [AN12437 ](https://www.nxp.com/docs/en/application-note/AN12437.pdf) explains the performance from different memories.

Using the internal TCMs is always the fastest performance.  And then the internal OCRAM.  With external memories, executing from SDRAM or an octal DDR flash are typically about the same performance.  But a QSPI SDR flash is definitely slower.  

Best regards

jwtaylor24, 2023-04-18:
@DerekSnell Thank you for the reply! I completely missed the AppNote on the NXP site, that helped a lot. 

Now, regarding building Zephyr for RAM execution... is there any current workaround for the issue described above? I tried building / running the application from OCRAM1 on the M7 core (0x2024_0000), but for some reason the application dies during the `z_sys_init_run_level(INIT_LEVEL_PRE_KERNEL_1)` function call in `z_cstart()`. Stepping through the code shows that the devices have null / garbage pointers and I can't figure out why this would be the case. **Note:** this bug/failure is during the application initialization, not during MCU boot execution. MCU boot 'seems' to properly copy the image to OCRAM and jump to it, but then the application fails during init.

I would have thought building the application for OCRAM would have been a "workaround" for the issue described above, since during MCU boot the stack lives in 0x8000_0000 so copying the image to OCRAM wouldn't nuke any data.

To build for OCRAM1, I am using the following Kconfigs:
```
CONFIG_USE_DT_CODE_PARTITION=n    // For this one, I had to change the Kconfig.defconfig file to prevent MCU boot from setting 'y'.
CONFIG_FLASH_SIZE=512
CONFIG_FLASH_BASE_ADDRESS=0x20240000
```

Any help is much appreciated - I'm really struggling to get Zephyr built for a completely RAM based execution.

Thanks!

closed 2023-08-29 by carlescufi