zephyrproject-rtos/zephyr #50653: Contributor nomination of Stine Åkredalen

Andrewpini, 2022-09-26:
I would like to nominate the following user to the contributer role in Zephyr:

Full Name: Stine Åkredalen
GitHub username: akredalen
Organization: Nordic Semiconductor ASA, Bluetooth Mesh team

She needs this role to be able to contribute to the Bluetooth mesh implementation in Zephyr.

stephanosio, 2022-09-27:
Anyone will become a "Contributor" once they submit a pull request that is merged. You do not need any nominations for this.

The "Contributor Nomination" template is for requesting "an elevated permission level in
the Contributor role" (i.e. Triage permissions), which one can apply for once they "demonstrate the user's dedication to the
Zephyr project" (add links to 3-5 pull requests) as requested in the template.

Andrewpini, 2022-09-27:
Ok, thanks for the clarification 👍

stephanosio, 2022-09-27:
Added to the repository with "read" permission so that the user can be added as a reviewer.

closed 2022-09-27 by stephanosio