zephyrproject-rtos/zephyr #50654: Some files are being ALWAYS built, without them being used

nashif, 2022-09-26:
**Describe the bug**
Many files are being by default although they are not being used pulling files from upper layers and other subsystems.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b qemu_x86 samples/synchronization/
2. cat  build/compile_commands.json  | jq .[].file   |less
3. Review list of files being built

**Expected behavior**
We should only build files that are relevant to the application being built

**Impact**
 We should not be pulling upper layer code into the core kernel and basic needed set creating unnecessary dependenies.

**Logs and console output**
example of files being built for EACH application, whether needed or not, maybe this more, but those I can spot immediately. 

```
"/home/nashif/zephyrproject/zephyr/lib/posix/perror.c"
"/home/nashif/zephyrproject/zephyr/lib/posix/pthread_common.c"
"/home/nashif/zephyrproject/zephyr/lib/posix/nanosleep.c"
"/home/nashif/zephyrproject/zephyr/lib/os/crc32c_sw.c"
"/home/nashif/zephyrproject/zephyr/lib/os/crc32_sw.c"
"/home/nashif/zephyrproject/zephyr/lib/os/crc16_sw.c"
"/home/nashif/zephyrproject/zephyr/lib/os/crc8_sw.c"
"/home/nashif/zephyrproject/zephyr/lib/os/crc7_sw.c"
"/home/nashif/zephyrproject/zephyr/lib/os/dec.c"
"/home/nashif/zephyrproject/zephyr/lib/os/fdtable.c"
"/home/nashif/zephyrproject/zephyr/lib/os/hex.c"
"/home/nashif/zephyrproject/zephyr/lib/os/notify.c"
"/home/nashif/zephyrproject/zephyr/lib/os/onoff.c"
```

nashif, 2022-09-26:
cc @cfriedt  FYI

"/home/nashif/zephyrproject/zephyr/lib/posix/perror.c"
"/home/nashif/zephyrproject/zephyr/lib/posix/pthread_common.c"
"/home/nashif/zephyrproject/zephyr/lib/posix/nanosleep.c"

also

"/home/nashif/zephyrproject/zephyr/lib/os/fdtable.c"

why isn't that part of posix?

cfriedt, 2022-09-26:
> "/home/nashif/zephyrproject/zephyr/lib/os/fdtable.c"
> 
> why isn't that part of posix?

That's a great question. From what I can tell, it has to do with the divide that exists between posix and the sockets API. There is a fair amount of work to do to clean that up. Ultimately, I would like to prove that `CONFIG_NET_SOCKETS_POSIX_NAMES` is more or less synonymous with `CONFIG_POSIX_API` (or something similar), so that we can have a simple API wrapper with properly namespaced constants, functions, and types.

PR #50659 (2022-09-28): posix: conditionally compile perror pthread_common and nanosleep
Previously, these files were compiled unconditionally.

Partially fixes #50654
PR #50670 (2022-11-23): crc: Make the build of crc function dependent on a Kconfig
Add CONFIG_CRC for building CRC related routines.
CRC routines are now being built for each application, whether used or
not and are add in the build system unconditionally.

Partially fixes #50654
closed 2022-09-28 by fabiobaltieri