zephyrproject-rtos/zephyr #50656: Wrong definition of bank size for intel memory management driver.

dabekjakub, 2022-09-26:
I believe that this code fragment 
`#define SRAM_BANK_SIZE				(128 * 1024)`
from mm_drv_intel_adsp_mtl_tlb.c 
should be set to 
`#define SRAM_BANK_SIZE				(64 * 1024)`
So currently it returns much greater value of ebb banks for TLB translation.
I am not 100% sure on this that's why I create the issue.

fabiobaltieri, 2022-09-27:
Hi, Zephyr is an open-source project, and as such we welcome and encourage the community to submit patches directly. Would you be able to submit a pull request with the fix? This is the [contribution guideline documentation](https://docs.zephyrproject.org/latest/contribute/guidelines.html#contribution-guidelines).

dabekjakub, 2022-09-27:
I will whenever I find time to check the documentation for the platform - left it out here just so We remember.  Do we use TLB feature right now ? If we do this should be high priority. I will consult it with @mwasko

dabekjakub, 2022-09-30:
Closing - confirmed 128 was correct.

closed 2022-09-30 by dabekjakub