zephyrproject-rtos/zephyr #50684: After enabling  CONFIG_SPI_STM32_DMA in project config file for STM32MP157-dk2 Zephyr throwing error

swatikumari1-eaton, 2022-09-27:
I am trying to enable CONFIG_SPI_STM32_DMA macro in my config file but I am getting below error.

Building C object zephyr/drivers/dma/CMakeFiles/drivers__dma.dir/dma_stm32.c.obj
FAILED: zephyr/drivers/dma/CMakeFiles/drivers__dma.dir/dma_stm32.c.obj
D:\arm-embedded\gcc-arm-none-eabi-10.3-2021.07-win32\gcc-arm-none-eabi-10.3-2021.07\bin\arm-none-eabi-gcc.exe -DCORE_CM4 -DHSE_VALUE=8000000 -DKERNEL -DSTM32MP157Cxx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/include -ID:/BLERTOS/catalina/Firmware/MMPlus_Gen2/MMPlus_BL/build/zephyr/include/generated -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/soc/arm/st_stm32/stm32mp1 -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/lib/open-amp/. -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/drivers -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/soc/arm/st_stm32/common -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/modules/hal/cmsis/CMSIS/Core/Include -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/modules/hal/stm32/stm32cube/stm32mp1xx/soc -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/modules/hal/stm32/stm32cube/stm32mp1xx/drivers/include -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/modules/hal/stm32/stm32cube/stm32mp1xx/drivers/include/Legacy -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/modules/hal/stm32/stm32cube/common_ll/include -ID:/BLERTOS/catalina/Firmware/MMPlus_Gen2/MMPlus_BL/build/modules/libmetal/libmetal/lib/include -ID:/BLERTOS/zephyrproject/zephyr-v3.0.0/modules/lib/open-amp/open-amp/lib/include -isystem D:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/lib/libc/minimal/include -isystem D:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/subsys/cpp/include -isystem d:/arm-embedded/gcc-arm-none-eabi-10.3-2021.07-win32/gcc-arm-none-eabi-10.3-2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/include -isystem d:/arm-embedded/gcc-arm-none-eabi-10.3-2021.07-win32/gcc-arm-none-eabi-10.3-2021.07/bin/../lib/gcc/arm-none-eabi/10.3.1/include-fixed -Os -imacros D:/BLERTOS/catalina/Firmware/MMPlus_Gen2/MMPlus_BL/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee -imacros D:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=D:/BLERTOS/catalina/Firmware/MMPlus_Gen2/MMPlus_BL=CMAKE_SOURCE_DIR -fmacro-prefix-map=D:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr=ZEPHYR_BASE -fmacro-prefix-map=D:/BLERTOS/zephyrproject/zephyr-v3.0.0=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -MD -MT zephyr/drivers/dma/CMakeFiles/drivers__dma.dir/dma_stm32.c.obj -MF zephyr\drivers\dma\CMakeFiles\drivers__dma.dir\dma_stm32.c.obj.d -o zephyr/drivers/dma/CMakeFiles/drivers__dma.dir/dma_stm32.c.obj -c D:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/drivers/dma/dma_stm32.c
D:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/drivers/dma/dma_stm32.c: In function 'dma_stm32_configure':
D:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/drivers/dma/dma_stm32.c:450:16: error: 'LL_DMA_InitTypeDef' has no member named 'Channel'
  450 |  DMA_InitStruct.Channel = dma_stm32_slot_to_channel(config->dma_slot);
      |                ^
D:/BLERTOS/zephyrproject/zephyr-v3.0.0/zephyr/drivers/dma/dma_stm32.c:450:27: warning: implicit declaration of function 'dma_stm32_slot_to_channel' [-Wimplicit-function-declaration]
  450 |  DMA_InitStruct.Channel = dma_stm32_slot_to_channel(config->dma_slot);
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~
[181/272] Linking C static library zephyr\drivers\console\libdrivers__console.a
ninja: build stopped: subcommand failed.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\STM32MP157-dk2
3. make
4. See error

**Impact**
Build failed

**Environment (please complete the following information):**
 - OS: - Windows
 - Toolchain -  Zephyr SDK
 - Commit SHA or Version -zephyr-v3.0.0
[image]

erwango, 2022-09-27:
Moving to "enhancement" as DMA has not yet been ported on STM32MP1.

FRASTM, 2022-09-27:
as a first WA I could suggest you  to try to adapt the dma_stm32.c driver like for the stm32H7 family: 
```
diff --git a/drivers/dma/dma_stm32.c b/drivers/dma/dma_stm32.c
index 6f716c8c14..ecb52bdebf 100644
--- a/drivers/dma/dma_stm32.c
+++ b/drivers/dma/dma_stm32.c
@@ -449,7 +449,7 @@ DMA_STM32_EXPORT_API int dma_stm32_configure(const struct device *dev,
 	DMA_InitStruct.PeriphBurst = stm32_dma_get_pburst(config,
 							stream->source_periph);
 
-#if !defined(CONFIG_SOC_SERIES_STM32H7X)
+#if !defined(CONFIG_SOC_SERIES_STM32H7X) && !defined(CONFIG_SOC_SERIES_STM32MP1X)
 	if (config->channel_direction != MEMORY_TO_MEMORY) {
 		if (config->dma_slot >= 8) {
 			LOG_ERR("dma slot error.");
@@ -595,7 +595,8 @@ DMA_STM32_EXPORT_API int dma_stm32_stop(const struct device *dev, uint32_t id)
 		return -EINVAL;
 	}
 
-#if !defined(CONFIG_DMAMUX_STM32) || defined(CONFIG_SOC_SERIES_STM32H7X)
+#if !defined(CONFIG_DMAMUX_STM32) \
+	|| defined(CONFIG_SOC_SERIES_STM32H7X) || defined(CONFIG_SOC_SERIES_STM32MP1X)
 	LL_DMA_DisableIT_TC(dma, dma_stm32_id_to_stream(id));
 #endif /* CONFIG_DMAMUX_STM32 */
```

FRASTM, 2022-09-27:
see https://github.com/FRASTM/zephyr/tree/dma_mp1

erwango, 2022-10-06:
> see branch https://github.com/FRASTM/zephyr/tree/dma_mp1

 @swatikumari1-eaton Feedback would be appreciated

closed 2022-11-04 by carlescufi