zephyrproject-rtos/zephyr #50686: Bluetooth: GATT: Verify of handle during reads with handle ranges

Thalley, 2022-09-27:
**Is your enhancement proposal related to a problem? Please describe.**
The issue was found due to a remote server with a bug when handling read by type request with a UUID. 

The basic issue is that Zephyr would perform a read by type request with a specific range, and a UUID, e.g. `start_handle = 0x10`, `end_handle=0xFFFF` and some UUID. 

In this case, the remote GATT server would return with a `handle = 0x20` for the specified GATT attribute. 

In `parse_read_by_uuid`, Zephyr then updates the start handle by 
```c
/* Update start_handle */
params->by_uuid.start_handle = handle;
```

and continues the read with a new starting handle. This is all working as expected, and is fine when reading from proper servers. The issue occurs because the server would incorrect send the same handle again next time, when Zephyr did the read by type request with `start_handle = 0x20` and `end_handle=0xFFFF`, where the server would once again return `handle = 0x20`, and Zephyr would then keep reading from handle `0x20`. 

This is without a doubt an issue on the remote GATT server, but we should verify the returned handle to be within the expected return range (in this case, between 0x21 and 0xFFFF inclusive). 

I have not checked if this same issue can occur for other GATT operations. 

**Describe the solution you'd like**
Instead of blindly updating the handle with 
```c
/* Update start_handle */
params->by_uuid.start_handle = handle;
```

We should verify that `handle > params->by_uuid.start_handle`, or better yet, `IN_RANGE(handle, params->by_uuid.start_handle, params->by_uuid.end_handle)`. 

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

Thalley, 2022-09-27:
@alwa-nordic I wasn't sure if I would categorize this as a bug or enhancement, so I'll let you decide on that :)

alwa-nordic, 2022-09-28:
> In this case, the remote GATT server would return with a handle = 0x20 for the specified GATT attribute. [... Then] Zephyr did the read by type request with start_handle = 0x20 and end_handle=0xFFFF, where the server would once again return handle = 0x20, and Zephyr would then keep reading from handle 0x20. This results in an infinite loop.

Did you see this in a sniffer log? Then is clearly a bug in the Zephyr ATT client. Obviously the start handle of the next read-by-type operation must be larger than the handle of all of the returned attributes.

Thalley, 2022-09-28:
> > In this case, the remote GATT server would return with a handle = 0x20 for the specified GATT attribute. [... Then] Zephyr did the read by type request with start_handle = 0x20 and end_handle=0xFFFF, where the server would once again return handle = 0x20, and Zephyr would then keep reading from handle 0x20. This results in an infinite loop.
> 
> Did you see this in a sniffer log? Then it's clearly a bug in the Zephyr ATT client. Obviously the start handle of the next read-by-type operation must be larger than the handles of all of the returned attributes.

Not from a sniffer log, but from using the BT shell to do `audio discover`:

```c
	[2022-09-15 18:02:37] [00:01:21.484,802] <dbg> bt_unicast_client: unicast_client_ase_read_func: conn 0x20029990 err 0x00 len 2
	[2022-09-15 18:02:38] [00:01:21.484,832] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
	[2022-09-15 18:02:38] [00:01:21.484,893] <dbg> bt_unicast_client: unicast_client_ep_set_status: ep 0x20040d90 handle 0x002f id 0x01 dir 1 state idle -> idle
	[2022-09-15 18:02:38] [00:01:21.484,924] <dbg> bt_unicast_client: unicast_client_ep_subscribe: ep 0x20040d90 handle 0x2f
	[2022-09-15 18:02:38] [00:01:21.634,796] <dbg> bt_unicast_client: unicast_client_ase_read_func: conn 0x20029990 err 0x00 len 2
	[2022-09-15 18:02:38] [00:01:21.634,826] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
	[2022-09-15 18:02:38] [00:01:21.634,857] <dbg> bt_unicast_client: unicast_client_ep_set_status: ep 0x20040d90 handle 0x002f id 0x01 dir 1 state idle -> idle
	[2022-09-15 18:02:38] [00:01:21.634,887] <dbg> bt_unicast_client: unicast_client_ep_subscribe: ep 0x20040d90 handle 0x2f
	[2022-09-15 18:02:38] [00:01:21.694,824] <dbg> bt_unicast_client: unicast_client_ase_read_func: conn 0x20029990 err 0x00 len 2
	[2022-09-15 18:02:38] [00:01:21.694,824] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
	[2022-09-15 18:02:38] [00:01:21.694,854] <dbg> bt_unicast_client: unicast_client_ep_set_status: ep 0x20040d90 handle 0x002f id 0x01 dir 1 state idle -> idle
	[2022-09-15 18:02:38] [00:01:21.694,915] <dbg> bt_unicast_client: unicast_client_ep_subscribe: ep 0x20040d90 handle 0x2f
	[2022-09-15 18:02:38] Conn: 0x20029990, Sink #129: ep 0x20040d90
	[2022-09-15 18:02:38] Conn: 0x20029990, Sink #130: ep 0x20040d90
	[2022-09-15 18:02:38] Conn: 0x20029990, Sink #131: ep 0x20040d90
	[2022-09-15 18:02:38] [00:01:21.754,852] <dbg> bt_unicast_client: unicast_client_ase_read_func: conn 0x20029990 err 0x00 len 2
	[2022-09-15 18:02:38] [00:01:21.754,882] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
	[2022-09-15 18:02:38] [00:01:21.754,913] <dbg> bt_unicast_client: unicast_client_ep_set_status: ep 0x20040d90 handle 0x002f id 0x01 dir 1 state idle -> idle
	[2022-09-15 18:02:38] [00:01:21.754,943] <dbg> bt_unicast_client: unicast_client_ep_subscribe: ep 0x20040d90 handle 0x2f
	[2022-09-15 18:02:38] [00:01:21.934,814] <dbg> bt_unicast_client: unicast_client_ase_read_func: conn 0x20029990 err 0x00 len 2
	[2022-09-15 18:02:38] [00:01:21.934,844] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
	[2022-09-15 18:02:38] [00:01:21.934,875] <dbg> bt_unicast_client: unicast_client_ep_set_status: ep 0x20040d90 handle 0x002f id 0x01 dir 1 state idle -> idle
	[2022-09-15 18:02:38] [00:01:21.934,906] <dbg> bt_unicast_client: unicast_client_ep_subscribe: ep 0x20040d90 handle 0x2f
	[2022-09-15 18:02:38] [00:01:21.994,812] <dbg> bt_unicast_client: unicast_client_ase_read_func: conn 0x20029990 err 0x00 len 2
	[2022-09-15 18:02:38] [00:01:21.994,842] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
	[2022-09-15 18:02:38] [00:01:21.994,873] <dbg> bt_unicast_client: unicast_client_ep_set_status: ep 0x20040d90 handle 0x002f id 0x01 dir 1 state idle -> idle
	[2022-09-15 18:02:38] [00:01:21.994,903] <dbg> bt_unicast_client: unicast_client_ep_subscribe: ep 0x20040d90 handle 0x2f
	[2022-09-15 18:02:38] Conn: 0x20029990, Sink #132: ep 0x20040d90
	[2022-09-15 18:02:38] Conn: 0x20029990, Sink #133: ep 0x20040d90
	[2022-09-15 18:02:38] Conn: 0x20029990, Sink #134: ep 0x20040d90
	[2022-09-15 18:02:38] [00:01:22.054,809] <dbg> bt_unicast_client: unicast_client_ase_read_func: conn 0x20029990 err 0x00 len 2
	[2022-09-15 18:02:38] [00:01:22.054,870] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
	[2022-09-15 18:02:38] [00:01:22.054,901] <dbg> bt_unicast_client: unicast_client_ep_set_status: ep 0x20040d90 handle 0x002f id 0x01 dir 1 state idle -> idle
	[2022-09-15 18:02:38] [00:01:22.054,901] <dbg> bt_unicast_client: unicast_client_ep_subscribe: ep 0x20040d90 handle 0x2f
	[2022-09-15 18:02:38] [00:01:22.204,833] <dbg> bt_unicast_client: unicast_client_ase_read_func: conn 0x20029990 err 0x00 len 2
	[2022-09-15 18:02:38] [00:01:22.204,833] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
	[2022-09-15 18:02:38] [00:01:22.204,864] <dbg> bt_unicast_client: unicast_client_ep_set_status: ep 0x20040d90 handle 0x002f id 0x01 dir 1 state idle -> idle
	[2022-09-15 18:02:38] [00:01:22.204,895] <dbg> bt_unicast_client: unicast_client_ep_subscribe: ep 0x20040d90 handle 0x2f
	[2022-09-15 18:02:38] [00:01:22.264,831] <dbg> bt_unicast_client: unicast_client_ase_read_func: conn 0x20029990 err 0x00 len 2
	[2022-09-15 18:02:38] [00:01:22.264,831] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
[... cut ...]
	[2022-09-15 18:02:40] [00:01:23.944,885] <dbg> bt_unicast_client: unicast_client_ase_read_func: handle 0x002f
	[2022-09-15 18:02:40] [00:01:23.944,915] <dbg> bt_unicast_client: unicast_client_ep_set_status: ep 0x20040d90 handle 0x002f id 0x01 dir 1 state idle -> idle
	[2022-09-15 18:02:40] [00:01:23.944,946] <dbg> bt_unicast_client: unicast_client_ep_subscribe: ep 0x20040d90 handle 0x2f
	[2022-09-15 18:02:40] Conn: 0x20029990, Sink #153: ep 0x20040d90
	[2022-09-15 18:02:40] Conn: 0x20029990, Sink #154: ep 0x20040d90
```

This was from testing at UPF, and the remote server were aware of this issue.

alwa-nordic, 2022-09-28:
The log output lacks evidence suggesting that we repeat a request-by-type with the same start handle. I looked at the code, in `parse_read_by_uuid` and it looks like it should work.

I suspect that the server is returning an attribute outside the requested range. Would you like to try the following just above the line with `/* Update start_handle */`?

```c
if (handle < params->by_uuid.start_handle) {
	BT_WARN("Remote returned attribute with handle outside requested range.");
	params->func(conn, BT_ATT_ERR_UNLIKELY, params, NULL, 0);
	return;
}
```

alwa-nordic, 2022-09-28:
Do you think it's useful to gracefully handle this quirk by just skipping any attributes outside the requested range?

Thalley, 2022-09-28:
> The log output lacks evidence suggesting that we repeat a request-by-type with the same start handle. I looked at the code in `parse_read_by_uuid` and it looks fine.
> 
> I suspect that the server is returning an attribute outside the requested range. 

I think you are correct in your suspicion (it it also what I wrote is the problem in the issue). 

> Would you like to try the following just above the line with `/* Update start_handle */`?
> 
> ```c
> if (handle < params->by_uuid.start_handle) {
> 	BT_WARN("Remote returned attribute with handle outside requested range. Stopping discovery.");
> 	params->func(conn, BT_ATT_ERR_UNLIKELY, params, NULL, 0);
> 	return;
> }
> ```

I don't have access to the faulty server, so I cannot just apply this to test if it works. It would require modifying the Zephyr GATT server to be faulty and then try with this patch on the GATT client. Unfortunately I do not have the time for doing this at the moment. 

> EDIT: I understand you already tried something like this and it worked from reading the rest of your post.

I have no not tried modifying anything yet.

Thalley, 2022-09-28:
> Do you think it's useful to gracefully handle this quirk by just skipping any attributes outside the requested range?

Yes and no - Yes as in we can just skip it, but no as in that we will still need to update our `start_handle` correctly. 

I am unsure of whether we should try to simply mitigate this by skipping the response, or if we should disconnect from the device since it is sending an invalid response. Since this issue caused an infinite loop, the former solution may not be enough.

closed 2023-12-21 by carlescufi