zephyrproject-rtos/zephyr #50691: Bluetooth: Host: CONFIG_BT_LOG_SNIFFER_INFO doesn't work as intended without bonding

Thalley, 2022-09-27:
**Describe the bug**
The help text of `CONFIG_BT_LOG_SNIFFER_INFO` says that it will enable logging of e.g. the LTK and private keys. While this is mostly true, the LTK is not logged if the link isn't bonding (triggered by e.g. setting `CONFIG_BT_BONDABLE=n`). 

**To Reproduce**
Steps to reproduce the behavior:
1. Build the BT shell (`zephyr/tests/bluetooth/shell`) with `CONFIG_BT_BONDABLE=n`
2. Connect to a device and do `bt security 2` to encrypt the link
3. See missing logging of the LTK

**Expected behavior**
Expected to see the LTK regardless of bonding, as debugging a non-bonded connection is equally valid. 

**Impact**
Annoyance

**Logs and console output**

Central with bonding:
```
uart:~$ bt security 2
LE conn param updated: int 0x0018 lat 0 to 42
Security changed: 5E:C8:43:72:21:1E (random) level 2
Identity resolved 5E:C8:43:72:21:1E (random) -> C2:35:E5:CA:5E:A1 (random)
Bonded with C2:35:E5:CA:5E:A1 (random)
[00:01:19.844,970] <inf> bt_keys: SC LTK: 0xca1505d416abfc96ab0db3cba088c6b8
uart:~$ 
```

Peripheral with bonding:
```
Security changed: 59:48:1E:1B:D4:30 (random) level 2
Identity resolved 59:48:1E:1B:D4:30 (random) -> D7:8C:A2:75:B6:07 (random)
Bonded with D7:8C:A2:75:B6:07 (random)
[00:02:24.922,943] <inf> bt_keys: SC LTK: 0xca1505d416abfc96ab0db3cba088c6b8
uart:~$ 
```

Central without bonding:
```
uart:~$ bt security 2
Security changed: 4E:22:41:2D:FE:6F (random) level 2
Paired with 4E:22:41:2D:FE:6F (random)
LE conn  param req: int (0x0018, 0x0028) lat 0 to 42
LE conn param updated: int 0x0018 lat 0 to 42
uart:~$ 
```

Peripheral without bonding:
```
Security changed: 71:46:D3:9B:A6:7A (random) level 2
Paired with 71:46:D3:9B:A6:7A (random)
LE conn param updated: int 0x0018 lat 0 to 42
uart:~$ 
```

**Environment (please complete the following information):**
 - Commit SHA or Version used: 489e8eb02c2a7bd46a9e73ee1a07a9bd20a61e98

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

mmahadevan108, 2022-09-27:
@carlescufi can you please help assign this Issue

Thalley, 2022-09-28:
> Is this to support a different sniffer that could decrypt the connection afterwards, e.g Ellisys?

Exactly. Generally if you have a sniffer trace of an encrypted link, you should be able to apply the encryption key (LTK) at any point to do the decryption. Ellisif may be the only sniffer software at this point that supports that (not sure).

PR #50758 (2022-09-29): Bluetooth: host: Log SC LTK in pairing procedure without bonding
Log the LE Secure Connection (SC) LTK in a pairing procedure without bonding when CONFIG_BT_LOG_SNIFFER_INFO is enabled.

Fixes: #50691
commit cc74b3b9 (2022-09-29): Bluetooth: host: Log SC LTK in pairing procedure without bonding

Log the LE Secure Connection (SC) LTK in a pairing procedure without
bonding when CONFIG_BT_LOG_SNIFFER_INFO is enabled.

Fixes: #50691
closed 2022-09-29 by fabiobaltieri