zephyrproject-rtos/zephyr #50709: tests: arch: arm: arm_thread_swap fails on stm32g0 or stm32l0

FRASTM, 2022-09-27:
**Describe the bug**
running the tests/arch/arm/arm_thread_swap  on the   nucleo_g071rb or nucleo_l073rz will fail

Please also mention any information which could help others to understand
the problem you're facing:
 In the tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c  the hard fault occurs after  
line 326  `z_move_thread_to_end_of_prio_q(_current);`
in  
`#if defined(CONFIG_ARMV6_M_ARMV8_M_BASELINE)`

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_g071rb tests/arch/arm/arm_thread_swap
2. west flash
3. See error

**Logs and console output**
```
*** Booting Zephyr OS build v3.2.0-rc2-233-g88705be89b8d  ***
Running TESTSUITE arm_thread_swap
===================================================================
START - test_arm_syscalls
 PASS - test_arm_syscalls in 0.001 seconds
===================================================================
START - test_arm_thread_swap
E: ***** HARD FAULT *****
E: r0/a1:  0x00000000  r1/a2:  0x20000504  r2/a3:  0x200001b8
E: r3/a4:  0x20000718 r12/ip:  0x89abcdef r14/lr:  0x08007203
E:  xpsr:  0x41000000
E: Faulting instruction address (r15/pc): 0x080009d8
E: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
E: Current thread: 0x200001b8 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
 - OS: ubuntu
 - Toolchain : Zephyr SDK, 0.15.0
 - Commit SHA 88705be89b8d03de3df256bf5ade7a3b24a22866

**Additional context**
Add any other context that could be relevant to your issue, such as pin setting,
target configuration, ...

mmahadevan108, 2022-09-27:
@FRASTM can you help narrow down to the commit that introduced the failure.

FRASTM, 2022-09-28:
I can identify sha1 e7591615724894a2319ea2c09d7eb21b4263fbf0 where test PASSED (with stm32g071rb and stml32073rz nucleo boards)  
and from sha1 6edcee886cf1f4231cc0c826fe0cf9fedf324ba7 the  test FAILED (on both boards) 
(In between it seems that python version changed  as on that sha1 we had a build error)

FRASTM, 2022-10-13:
It seems that one instruction is resetting R0 ( "mov r0, %0;\n\t") followed by (2 identical) multiple load of 4 registers at address R0 : Mem. address 0x0 (R0) cannot store R4, R5, R6, R7 

Suggested patch: 
```
diff --git a/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c b/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c
index c13926e104..022bbbad8f 100644
--- a/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c
+++ b/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c
@@ -344,8 +344,7 @@ static void alt_thread_entry(void)
 		"push {r0, r1};\n\t"
 		"mov r1, r7;\n\t"
 		"mov r0, %0;\n\t"
-		"ldmia r0!, {r4-r7};\n\t"
-		"ldmia r0!, {r4-r7};\n\t"
+		"ldmia r1!, {r4-r7};\n\t"
 		"mov r8, r4;\n\t"
 		"mov r9, r5;\n\t"
 		"mov r10, r6;\n\t"
```

FRASTM, 2022-10-13:
@microbuilder Could you please have a look and give any advice

microbuilder, 2022-10-14:
> @microbuilder Could you please have a look and give any advice

I don't have that specific board, but I'll try to test this on another M0+.

Looks like that code was added in a4049773ac445cd963faffdc46cd0772aa255d10:

```bash
$ git log -L347,+2:'tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c'
commit a4049773ac445cd963faffdc46cd0772aa255d10
Author: Ioannis Glaropoulos <Ioannis.Glaropoulos@nordicsemi.no>
Date:   Thu Oct 3 11:17:32 2019 +0200

    tests: arch: arm_thread_swap: extend the test for Cortex-M Baseline
    
    We rework the arm_thread_swap test, so it can build
    and run for Cortex-M Basline architecture (Cortex-M0,
    Cortex-M0+, and Cortex-M23). In most cases, this rework
    involved re-implementing the code blocks for storing
    and loading the callee-saved registers to and from
    memory. In addition, we skip the verification of
    BASEPRI, and replace it with verifying PRIMASK.
    
    Signed-off-by: Ioannis Glaropoulos <Ioannis.Glaropoulos@nordicsemi.no>

diff --git a/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c b/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c
--- a/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c
+++ b/tests/arch/arm/arm_thread_swap/src/arm_thread_arch.c
@@ -262,0 +288,2 @@
+               "ldmia r0!, {r4-r7};\n\t"
+               "ldmia r0!, {r4-r7};\n\t"
```

microbuilder, 2022-10-14:
Haven't been able to reproduce in qemu on the M0, so ordered a board to be able to reliably reproduce.

microbuilder, 2022-10-17:
@FRASTM @erwango Just a FYI as something to be aware of on ST's part, but it seems the latest release of openocd is broken with STLink due to a changes in libusb: https://github.com/libusb/libusb/issues/928

With openocd 0.11, you will get a segfault trying to run `west debugserver`:

> The command below is what `west debugserver` runs beneath the hood 

```bash
$ /opt/homebrew/bin/openocd -s /Users/kevintownsend/Linaro/zephyr/zephyr/boards/arm/nucleo_g071rb/support -s /Users/kevintownsend/zephyr-sdk-0.15.0/sysroots/arm64-pokysdk-linux/usr/share/openocd/scripts -f /Users/kevintownsend/Linaro/zephyr/zephyr/boards/arm/nucleo_g071rb/support/openocd.cfg -c 'tcl_port 6333' -c 'telnet_port 4444' -c 'gdb_port 3333' '-c init' '-c targets' -c 'reset init'
Open On-Chip Debugger 0.11.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Info : clock speed 2000 kHz
zsh: segmentation fault  /opt/homebrew/bin/openocd -s  -s  -f  -c 'tcl_port 6333' -c 'telnet_port 4444
```

You need to build the latest openocd, such as the following on MacOS:

```bash
$ brew unlink open-ocd
$ brew install openocd --HEAD
```

microbuilder, 2022-10-17:
@FRASTM The issue here is that the `ldmia` instruction is operating on `%0` (`"mov r0, %0;\n\t"`), which is resolving to `r5` in your case, which got clobbered at some point and has a value of `0` assigned to it.  This needs to be a valid address in memory that we can modify/increment. With the code as is, `%0/r5` is set to `0`, which causes a hard fault when `ldmia` tries to execute, writing to the address at `%0/r5`. The inline code itself is sort of broken, depending on getting lucky that `r5` hasn't been clobbered by the time we get here.

The M3/M4/M7, etc., version has the same issue, since it uses `%0` as well and there is a lot of code before this inline assembly fires.

Trying to figure out how to rework this, but the test feels broken as is.

microbuilder, 2022-10-18:
@FRASTM Please let me know if this works for you: https://github.com/zephyrproject-rtos/zephyr/pull/51366

gromero, 2022-10-19:
@FRASTM Kevin pushed a freshest code to #51366 in case you would like to re-test it.

FRASTM, 2022-10-20:
same result : that fixes the pb for stm32g0 and stm32l0 (arm M0+ ) : successfully tested on nucleo_g071rb and nucleo_l073rz

closed 2022-10-20 by stephanosio