zephyrproject-rtos/zephyr #50735: Intel CAVS18: tests/boards/intel_adsp/hda_log/boards.intel_adsp.hda_log.printk failed

shuai1x, 2022-09-28:
**Describe the bug**
On intel_adsp_cavs18 platform, tests/boards/intel_adsp/hda_log/boards.intel_adsp.hda_log.printk failed due to No Console Output(Timeout).
But it only appeared on daily report, and I can't reproduce it by manual operation.

Please also mention any information which could help others to understand
the problem you're facing:
 - intel_adsp_cavs18
 - tests/boards/intel_adsp/hda_log/boards.intel_adsp.hda_log.printk No Console Output(Timeout)

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W --hardware-map /home/ztest/cavs/cavs.map --device-testing -x=CONFIG_BOOT_DELAY=500 -s tests/boards/intel_adsp/hda_log/boards.intel_adsp.hda_log.printk -e can -vv --no-skipped-report
2. See error

**Logs and console output**
```
TEST:031372^M
TEST:031373^M
TEST:031374^M
TEST:031375^M
TEST:031376^M
TEST:031377^M
TEST:031378^M
TEST:031379^M
TEST:031380^M
TEST:031381^M
TEST:031382^M
TEST:031383^M
TEST:031384^M
TEST:031385^M
TEST:031386^M
TEST:031387^M
TEST:031388^M
TEST:031389^M
TEST:031390^M
TEST:031391^M
TEST:031392^M
TEST:031393^M
TEST:031394^M
TEST:031395^M
TEST:031396^M
TEST:031397^M
TEST:031398^M
TEST:031399^M
[00:00:17.895,000] <inf> hda_test: Timeout flush working if shown^M
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain - SDK 15.0

teburd, 2022-09-28:
Is this with or without #50450

shuai1x, 2022-09-29:
> Is this with or without #50450

Without https://github.com/zephyrproject-rtos/zephyr/pull/50450. Because I can't reproduce it, it only appeared on daily. So I haven't tried https://github.com/zephyrproject-rtos/zephyr/pull/50450.

shuai1x, 2022-10-12:
This failure is no longer present. So will close it.

closed 2022-10-12 by shuai1x