zephyrproject-rtos/zephyr #50744: Allow on creating incomplete neighbor entries and routes in case of receiving Router Advertisement.

kkasperczyk-no, 2022-09-28:
**Is your enhancement proposal related to a problem? Please describe.**
Currently it looks like Zephyr implementation assumes that after receiving Router Advertisement, neighbor entry for the source router and route to it should be added only if RA includes SLLAO (Source Link Layer Address Option) field and in result also Link Layer address.
Handling SLLAO:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/ipv6_nbr.c#L2400

Assert checking LL address before creating route:
https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/ip/route.c#L372

The problem is that according to RFC 9131, chapter 2, point 1, the router doesn't have to include LL address and SLLAO in RA:

> 1.  A host joins the network and receives a Router Advertisement (RA)
>        packet from the first-hop router (either a periodic unsolicited
>        RA or a response to a Router Solicitation sent by the host).  The
>        RA contains information the host needs to perform SLAAC and to
>        configure its network stack.  The RA is sent from the router's
>        link-local address to a link-local destination address and **may**
>        **contain the link-layer address** of the router.  As a result, the
>        host can populate its Neighbor Cache with the router's link-local
>        and link-layer addresses.

In my particular case I can see the router I use (specifically OpenThread Border Router) doesn't add SLLAO field to RA, so Zephyr host after receiving them adds only IPv6 prefix, but not neighbor entry and route to the source address. In result if I try to send something on the address belonging to assigned IPv6 prefix, it fails due to no route available.

RFC 4861, 6.3.4 Processing Received Router Advertisements states what to do in case SLLAO is included and in case SLLAO is not included, but neighbor entry exists, but it doesn't state what if neighbor should be created if SLLAO is not included:

>    After extracting information from the fixed part of the Router
>    Advertisement message, the advertisement is scanned for valid
>    options.  If the advertisement contains a Source Link-Layer Address
>    option, the link-layer address SHOULD be recorded in the Neighbor
>    Cache entry for the router (creating an entry if necessary) and the
>    IsRouter flag in the Neighbor Cache entry MUST be set to TRUE.  If no
>    Source Link-Layer Address is included, but a corresponding Neighbor
>    Cache entry exists, its IsRouter flag MUST be set to TRUE.  The
>    IsRouter flag is used by Neighbor Unreachability Detection to
>    determine when a router changes to being a host (i.e., no longer
>    capable of forwarding packets).  If a Neighbor Cache entry is created
>    for the router, its reachability state MUST be set to STALE as
>    specified in Section 7.3.3.  If a cache entry already exists and is
>    updated with a different link-layer address, the reachability state
>    MUST also be set to STALE.

**Describe the solution you'd like**
I think that Zephyr should handle this situation and add route even if Link Layer address is not available (in fact it's not needed to reach the destination). I would imagine that implementation could add incomplete neighbor (without Link Layer address) every time it receives RA and add route for the received IPv6 prefix to the router link local source address. If after some time Neighbor Discovery would be performed and Link Layer address known, neighbor entry should be updated.

I created some local prove of concept that adds neighbor entry in described way and removes few asserts preventing from adding route to node that doesn't have LL address and it worked for me:
https://github.com/kkasperczyk-no/zephyr/commit/05beef972d0583293919ee0539297f9552aad834

I'm not able to predict all consequences to routing module after such change, so I wanted to get initial opinions on the problem and proposed solution

**Describe alternatives you've considered**
Unfortunately I didn't find alternative solution.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

jukkar, 2022-09-30:
@kkasperczyk-no Please send a PR for your proposal, it is then public and people can find and comment on it.

kkasperczyk-no, 2022-10-03:
@jukkar ah ok, I thought my fork changes should be visible to anyone. I opened PR and linked to this issue.

jukkar, 2022-10-03:
Well, your forked changes are visible but usually we do not review code that is not in the PR.

closed 2022-10-05 by fabiobaltieri