zephyrproject-rtos/zephyr #50746: Stale kernel memory pool API references

stephanosio, 2022-09-28:
**Describe the bug**

There are various stale references to the kernel memory pool API (`k_mem_pool`) in the current Zephyr codebase.

The `k_mem_pool` support was (supposed to be) removed in the Zephyr v2.5.0 release as part of #28611, but it seems to have missed a few references.

```
$ git grep k_mem_pool
cmake/linker_script/common/common-ram.cmake:zephyr_iterable_section(NAME k_mem_pool GROUP DATA_REGION ${XIP_ALIGN_WITH_INPUT} SUBALIGN 4)
doc/kernel/usermode/memory_domain.rst:must assign themselves to a resource pool, which is a k_mem_pool object.
... (release notes stuff)
include/zephyr/kernel.h:struct k_mem_pool;
include/zephyr/kernel.h: * k_mem_pool_malloc().
include/zephyr/kernel/mempool_heap.h:/* Compatibility implementation of a k_mem_pool backend in terms of a
include/zephyr/linker/common-ram.ld:    ITERABLE_SECTION_RAM_GC_ALLOWED(k_mem_pool, 4)
scripts/pylib/twister/twisterlib/size_calc.py:        "k_mem_pool_area",
soc/xtensa/esp32_net/linker.ld:    Z_LINK_ITERABLE_GC_ALLOWED(k_mem_pool);
tests/benchmarks/app_kernel/src/master.h:extern struct k_mem_pool DEMOPOOL;
```

stephanosio, 2022-09-28:
cc @andyross @peter-mitsis @dcpleung

stephanosio, 2022-09-28:
Keeping this open since there are still leftovers that require more intrusive cleaning as noted in https://github.com/zephyrproject-rtos/zephyr/pull/50749#issue-1389197533.

> [1] There are also k_mem_block and all the mailbox "memory block" support bits, which were also supposed to be removed in https://github.com/zephyrproject-rtos/zephyr/pull/28611 (see https://github.com/zephyrproject-rtos/zephyr/commit/c844bd87b3fdb059d7a81dcb02777fad558f77db), but it failed to do so completely. This is going to be done in the next release.

sylvioalves, 2022-10-11:
@stephanosio `soc/xtensa/esp32_net/linker.ld` is outdated and will be updated soon.

PR #50749 (2022-09-28): Remove stale kernel memory pool API references (for 3.2)
This series removes most<sup>[1]</sup> references to the kernel memory pool API, which was [removed in the Zephyr v2.5.0 release](https://github.com/zephyrproject-rtos/zephyr/pull/28611).

<sup>[1]</sup> There are also `k_mem_block` and all the mailbox "memory block" support bits, which were also supposed to be removed in #28611 (see c844bd87b3fdb059d7a81dcb02777fad558f77db), but it failed to do so completely. This is going to be done in the next release.

Partially fixes #50746
closed 2023-08-13 by github-actions[bot]