zephyrproject-rtos/zephyr #50747: cache API rework

carlocaione, 2022-09-28:
The current cache API needs a deep rework as came out in #50136.

### The problem
The cache operations must be quick, optimized and possibly inlined. The current API is clunky, functions are not inlined and passing parameters around that are basically always known at compile time.

See [include/zephyr/cache.h](https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/cache.h)

### The proposal
We extend the set of functions so that we have:
- `cache_{data,instr}_enable()`
- `cache_{data,instr}_disable()`
- `cache_{data,instr}_line_size_get()`
- `cache_{data,instr}_flush_all()`
- `cache_{data,instr}_flush_all(addr, size)`
- `cache_{data,instr}_invd_all()`
- `cache_{data,instr}_invd_range(addr, size)`
- `cache_{data,instr}_flush_and_invd_all(addr, size)`
- `cache_{data,instr}_cache_data_flush_and_invd_range(addr, size)`

This should allow us to get rid of useless parameters and make inlining easier.

### The requirements

We still have to keep supporting external cache controllers besides arch operations (see for example [drivers/cache/cache_aspeed.c](https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/cache/cache_aspeed.c) ). This is currently done acting on `CONFIG_HAS_EXTERNAL_CACHE / CONFIG_HAS_ARCH_CACHE` and this is going to be kept also in the new API.

### Open questions
1. Do we want to keep supporting `syscall` wrappings for this API?
2. Do we need other operations besides `flush`, `invalidate` and `flush and invalidate`? We have of course a scaling problem now that we got rid of parameters.

carlocaione, 2022-09-28:
@andyross @lgirdwood @lyakh @nashif @carlescufi @stephanosio

teburd, 2022-09-29:
Would creating a userspace friendly wrapper around DMA buffers (that does the work of flushing/invalidating at the right times perhaps) allow for these to *not* be syscalls?

Is there another use case that requires them to be syscalls?

carlocaione, 2022-09-30:
> Would creating a userspace friendly wrapper around DMA buffers similar to but not exactly like Linux's dma_buf (that does the work of flushing/invalidating at the right times perhaps) allow for these to _not_ be syscalls?

Possibly but we cannot exclude other possible uses. Quoting @andyross from discord: `Whether cache control instructions are priviledged or not is going to be arch-specific.  It's probably safe to assume that usermode code will never care, but at the same time we should probably support the API wherever we can.`

> Is there another use case that requires them to be syscalls?

I'm thinking for example user-space apps using OpenAMP. In that case OpenAMP is using cache operations so everything comes down to whether the arch supports operations in low-privilege threads or not.

lgirdwood, 2022-10-05:
@carlocaione can we also have 
```c
cache_{data,instr}_lock()
cache_{data,instr}_unlock()
```
This will allow us some more fine grained control over cache when needed for performance or debug reasons. Thanks

andyross, 2022-10-05:
@lgirdwood  What does the lock/unlock entail, semantically?  Pinning cache lines (which I'm guessing is what you mean?) is a thing, but it tends to be highly hardware-specific, and may even have things like different interfaces for different levels of cache.  You need to worry about aliasing between ways (i.e. you can only have N lines pinned with an address modulo the way stride), etc...  This just seems like a hard API to have at the generic layer, though I'm not an expert.

andyross, 2022-10-05:
And regarding the syscall discussion from last week: I guess my feeling is that we should have some kind of support for userspace here[1].  I don't necessarily know that it needs to be exactly the same API.  Something like the old API could be left as the "high level" API, since it has fewer entry points to make into syscalls, and obviously the concern about inlinability needed by the kernel and drivers isn't going to be a factor.

[1] Actually something like future SOF is a good use case.  Imagine two tightly-coupled userspace modules talking over shared memory in the incoherent cache; we don't necessarily want to have to modify the SOF framework or the Zephyr kernel to handle the communication, it would be nice for them to just map the shared region and do their thing.  But on Xtensa that requires cache management.

carlocaione, 2022-10-21:
> in this case there is an NXP platform with an ARM Cortex-M4 and a cache. According to the M4 spec there should not be a cache but NXP has added a custom one onto the chip

Wow, that's quite unusual. Are both functional at the same time? If they do, this is currently not supported (I mean, that would be very hard to support in a general way: you have to flush and invalidate twice every time?)

decsny, 2022-10-23:
> Wow, that's quite unusual. Are both functional at the same time? If they do, this is currently not supported (I mean, that would be very hard to support in a general way: you have to flush and [...]

I think maybe we have a misunderstanding here, what I mean is there is only a non-standard cache. ARM Cortex M4 has no cache supported by it's specification, NXP basically added their own custom one despite it not being in the M4 specification. So this has caused problems for me because a lot of the ARM code that is conditionally compiled due to Kconfigs when an ARM core has a cache assumes that the core is an M7, because that is basically the only ARM v7-m core with a cache, when this is not the case. So I tried to get around the problem using the external cache API. You can see what I am describing in my PR here: #50871

carlocaione, 2022-10-24:
> ARM Cortex M4 has no cache supported by it's specification, NXP basically added their own custom one despite it not being in the M4 specification.

Ok, got it. In that case it's correct that you introduced an external cache controller driver

closed 2023-05-04 by carlocaione