zephyrproject-rtos/zephyr #50766: Zephyr toolchain doesn't setup CMake host environment correctly

SkryptX, 2022-09-28:
The toolchain of Zephyr doesn't correctly setup the CMake environment when using Zephyr on a Linux/Posix host.  

Examples of this are: 
- `CMAKE_SIZEOF_VOID_P` is not set
- The whole `find_package` mechanism for libraries on a Linux host does not work nicely
- `GnuInstallDirs` also doesn't work

I sadly do not yet understand enough from the setup Zephyr tries to achieve, but when using a Posix build it is basically impossible to get all the moving parts functioning correctly if you try to use shared libraries from the host for example.

Maybe it needs a configuration setting to allow using host dependencies or something. I would just like to hear your opinion based on your own experience. Benefits of an improvement are a much easier time developing simulation tools for Zephyr based on the Posix board or similar.

Environment (please complete the following information):
 - OS: Linux
 - Toolchain: Zephyr SDK host (gcc/ld)
 - Version: 3.1.99

stephanosio, 2022-09-28:
cc @tejlmand

stephanosio, 2022-09-28:
> The toolchain of Zephyr doesn't correctly setup the CMake environment when using Zephyr on a Linux/Posix host.

I suppose you mean "build system" and not toolchain?

Also, Zephyr SDK does not include any host toolchains.

> using a Posix build it is basically impossible to get all the moving parts functioning correctly if you try to use shared libraries from the host for example.

I do not think this is something that is supported. @tejlmand can confirm.

SkryptX, 2022-09-28:
> Also, Zephyr SDK does not include any host toolchains.

Yes, my bad. Has nothing to do with Zephyr SDK.

> I do not think this is something that is supported. @tejlmand can confirm.

I understand that it isn't supported currently. 

> ...please state clearly if your objective is to build for the `native_posix` board exclusively.

I am not interested in the `native_posix` board _exclusively_. I am currently trying to build a much more useful and less hacky posix board (obviously based on the `native_posix` board). My experiments so far did show that it is possible to build a Zephyr application on Linux that is configurable with a device tree just like on embedded boards, but with instrumentation for simulation purposes and all of this within existing Zephyr semantics. 

If you now want to write a custom driver specifically for the posix board, then you might need host libraries and that is my motivation for this ticket; to be able to load these libraries pretty much as easy as in a "normal" build system and interface with the host system just like a HAL library would with embedded hardware.

SkryptX, 2022-11-29:
I have learned about the `ZEPHYR_TOOLCHAIN_VARIANT` variable and setting it to `host` (or `llvm`?) should have the effect of disabling cross compiling and keep the host detection mechanisms like `UNIX` etc. That would be an easy and especially explicit fix for this issue in my opinion.

tejlmand, 2023-03-02:
> The build system of Zephyr doesn't correctly setup the CMake environment when using Zephyr on a Linux/Posix host.

Not sure exactly what you are requesting, so if you providing a very basic example of what is not working and why you expected it work / how you would like it to work would be very helpful.

You do point to a couple of specific areas, so let me reply to those:
> `CMAKE_SIZEOF_VOID_P` is not set

Correct, cause Zephyr is not using the CMake toolchain infrastructure for various reasons, some of those being historic.
But we do address issues of missing variables being set when they are raised, such as the BYTE_ORDER case: https://github.com/zephyrproject-rtos/zephyr/issues/45270, and `SIZEOF` could be of similar nature.
But I would prefer this to be raised as an independent issue if you like it addressed.

>  The whole find_package mechanism for libraries on a Linux host does not work nicely

Not sure I understand ?
`find_package()` is used a lot and is working as expected, but there could of course be specific packages which are not compatible with Zephyr build system.
But without more info on this, then it's too vaguely described to address.

> `GnuInstallDirs` also doesn't work

Not understood ?
Did you do `include(GNUInstallDirs)` in your CMakeLists.txt file ?

That provides the expected settings, for example:
```
$ grep CMAKE_INSTALL CMakeCache.txt
...
CMAKE_INSTALL_BINDIR:PATH=bin
CMAKE_INSTALL_DATADIR:PATH=
CMAKE_INSTALL_DATAROOTDIR:PATH=share
CMAKE_INSTALL_DOCDIR:PATH=
CMAKE_INSTALL_INCLUDEDIR:PATH=include
CMAKE_INSTALL_INFODIR:PATH=
CMAKE_INSTALL_LIBDIR:PATH=lib
CMAKE_INSTALL_LIBEXECDIR:PATH=libexec
...
```

> I have also learned about the ZEPHYR_TOOLCHAIN_VARIANT variable and setting it to host (or llvm?) should have the effect of disabling cross compiling and keep the host detection mechanisms like UNIX etc.

Correct, when setting toolchain to `host` or building for `native_posix` then no cross-compilation is done.

> That would be an easy and especially explicit fix for this issue in my opinion.

So does that mean that things are now working as you expect ?

Changing to `enhancement`.
So far this is not a bug, but still unsure what @SkryptX expects, so keeping it open but `stale` as it is too vaguely described.

SkryptX, 2023-03-03:
@tejlmand From what I can remember is that the `find_package` mechanism failed on my system, because `GnuInstallDirs` uses `CMAKE_SIZEOF_VOID_P` in its detection and instead of the `lib64` it always set `lib` as library path which is wrong on our system. So the detection mechanism fails and simple stuff like finding a library like `libpng` becomes impossible. 

Root cause is that Zephyr, no matter what toolchain is selected, always sets up cross-compiling. That's also what I tried to say in my comment from Nov 29. For example, I build my custom POSIX board, through [FindHostTools.cmake](https://github.com/zephyrproject-rtos/zephyr/blob/main/cmake/modules/FindHostTools.cmake) the `ZEPHYR_TOOLCHAIN_VARIANT` gets set to `host` and _still_, when I print out the variables like this: 

    message(STATUS "===${ZEPHYR_TOOLCHAIN_VARIANT}===${CMAKE_SYSTEM_NAME}===${CMAKE_CROSSCOMPILING}===${UNIX}===")

The output is this: `===host===Generic===TRUE======`. A large portion of the CMake platform detection mechanism is not working, even while using the `host` toolchain and this just makes it annoying. Especially knowing that CMake would usually set up everything correctly automatically.

My wish would be to be able to build Zephyr in a POSIX/host variant pretty much out of the box with library and host tool support that I normally get in a normal CMake project.

I hope this is helpful to understand what this issue is about :)