zephyrproject-rtos/zephyr #50774: ESP32 GPIO34 IRQ not working

Machine-Hum, 2022-09-28:
I can't find anything in the docs about GPIO34 not being able to produce an interrupt. I know it's an "input only" pin, but this should still be able to produce an interrupts.

``` 
        // rdy-gpios = <&gpio0 27 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
        rdy-gpios = <&gpio0 34 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;
```
When the top line is uncommented, pin27 will produce an interrupt fine. When I use the bottom line it doesn't work. I have jumped the two pins together with a wire.

``` C
int iqs5xx_init_interrupt(const struct device *dev)
{
    struct iqs5xx_data *data = dev->data;
    const struct iqs5xx_dev_config *cfg = dev->config;
    int rc;

    if (!device_is_ready(cfg->rdy_gpio.port)) {
        LOG_ERR("%s: device %s is not ready", dev->name,
            cfg->rdy_gpio.port->name);
        return -ENODEV;
    }

    rc = gpio_pin_configure_dt(&cfg->rdy_gpio, GPIO_INPUT | cfg->rdy_gpio.dt_flags);
    if (rc < 0) {
        return rc;
    }

    gpio_init_callback(&data->gpio_cb,
               iqs5xx_gpio_callback,
               BIT(cfg->rdy_gpio.pin));

    rc = gpio_add_callback(cfg->rdy_gpio.port, &data->gpio_cb);
    if (rc < 0) {
        return rc;
    }

    data->dev = dev;

    data->work.handler = iqs5xx_work_cb;
    return 0;
}
```

``` C
static void setup_int(const struct device *dev,
              bool enable)
{
    const struct iqs5xx_dev_config *cfg = dev->config;
    gpio_flags_t flags = enable
        ? GPIO_INT_EDGE_RISING
        : GPIO_INT_DISABLE;

    gpio_pin_interrupt_configure_dt(&cfg->rdy_gpio, flags);
}

static void iqs5xx_gpio_callback(const struct device *dev,
                  struct gpio_callback *cb, uint32_t pins)
{
    struct iqs5xx_data *data =
        CONTAINER_OF(cb, struct iqs5xx_data, gpio_cb);

    setup_int(data->dev, false);
    k_work_submit(&data->work);
}
```
Output...
```
�*** Booting Zephyr OS build v3.2.0-rc2-158-gb3ead37efaa5  ***
[00:00:00.432,000] <err> net_if: There is no network interface to work with!
[00:00:00.432,000] <err> iqs5xx: Failed to initialize interrupt!
```

Any ideas would be appreciated. I *might* have missed something, but looking all over the place, it seems this should work fine.

sylvioalves, 2022-09-28:
@Machine-Hum There was an issue regarding input pins and this PR fixed that: #50725  
Can you update your repo and test again to confirm whether your issue was also fixed?

Machine-Hum, 2022-09-28:
> @Machine-Hum There was an issue regarding input pins and this PR fixed that: #50725 Can you update your repo and test again to confirm whether your issue was also fixed?

I've updated it and am seeing the same issues. Thanks for your reply.

sylvioalves, 2022-09-28:
oh, I think I know the reason: GPIO34 needs to be addressed by `gpio1 2`.  
` rdy-gpios = <&gpio1 2 (GPIO_PULL_DOWN | GPIO_ACTIVE_HIGH)>;`  
Please, try that out.

Machine-Hum, 2022-09-28:
Exact same issue trying this :(

sylvioalves, 2022-09-28:
Should be without `GPIO_PULL_DOWN` as GPIOS34~39 have no PU nor PD resistors.  

`rdy-gpios = <&gpio1 2 GPIO_ACTIVE_HIGH>;`

Machine-Hum, 2022-09-28:
> Should be without `GPIO_PULL_DOWN` as GPIO34~39 have no PU nor PD resistors.
> 
> `rdy-gpios = <&gpio1 2 GPIO_ACTIVE_HIGH>;`

Awesome, that does it!

closed 2022-09-28 by Machine-Hum