zephyrproject-rtos/zephyr #50788: Use YAML files to describe device driver capabilities to validate kconfig/devicetree coherency

bjarki-trackunit, 2022-09-28:
## Introduction

We need a place to describe the capabilities of a driver. This information is used to validate coherence between the included device drivers and the devicetree before compilation, and to provide promises to the application regarding how a device driver may be interacted with.

The specific information we need:
1. Does a driver exists
2. Which devices does the driver support
3. How does the application interact with said driver

In Zephyr, we know at compile time exactly which driver will be included, what device it supports, and what API it implements. However, the build system doesn't know this information. With this information, the build system can validate coherency between devicetree and included drivers during build before compilation, generate relevant code/macros etc.

### Problem description
An example devicetree fragment shows a modem which needs a driver.
```
&uart0 {
    bg95 {
        compatible = "quectel,bg95";
    }; 
};
```

It requires some driver which supports the "quectel,bg95" compatable. In this example,
we have two drivers, gsm_ppp and quectel_bg9x.

Currently, we can run into the following obscure build errors
1. Build without any driver, we will get the classic linker error "undefined reference to __device_ ..."
2. Build with both drivers, we will get the rare linker  error "__device_ ... defined multiple times"

### Proposed change
Add YAML file alongside driver sourcecode
```
modem/quectel,bg9x.h
modem/quectel,bg9x.c
modem/quectel,bg9x.yaml <- added file
```

Fill in device driver capabilities in YAML file
```
# The drivers kconfig symbol
driver: MODEM_GSM_PPP

# The devicetree compatibles this driver supports
compatible:
  - "quectel,bg95"
  - "quectel,bg96"
  - "simcom,sim-7080"
  - "ubloc,sara-r4"

# Which API(s) does the driver implement
api:
  - net_if
```

## Detailed RFC

The relevant steps of the build system would look like this if this solution is implemented
1. Run devicetree cmake module
2. Run kconfig cmake module
3. Run devicedriver cmake module (name of module which this solution adds)

### Proposed change (Detailed)

To implement this solution, we must create a cmake module which runs a Python script, after kconfig and edtlib have executed. 

The new script searches through the zephyr/drivers folder collecting all YAML files. 
The script then validates the YAML files. 
Lastly, .config and edt.pickle are used to look for nodes and enabled drivers, validating coherency between them. If any error is found, it is reported in a helpful message, and the build is stopped.

The script should combine the deduced information from kconfig/devicetree into a dictionary or similar which can then be used by downstream scripts to generate macros etc.

### Dependencies

The feature is a new addition to the build system which depends on kconfig and devicetree to function.

The feature will only cause a validation errors if one or more compatible drivers is found for an enabled device node, and none of them are enabled. This allows for adding the YAML files over time without causing erroneous build errors.

### Concerns and Unresolved Questions

## Alternatives

One promising alternative is described here https://github.com/zephyrproject-rtos/zephyr/pull/50724
It suggests standardizing the kconfig files and updating the kconfig.py script to deduce device driver capabilities by linking kconfig symbols by name to properties. For example:
```
config UART_SAM
	bool "Atmel SAM MCU family UART driver"
	default y
	depends on DT_HAS_ATMEL_SAM_UART_ENABLED
	select SERIAL_HAS_DRIVER
	select SERIAL_SUPPORT_INTERRUPT
	help
	  This option enables the UARTx driver for Atmel SAM MCUs.
```

would deduce: 
* SERIAL_HAS_DRIVER -> supports uart API
* SERIAL_SUPPORT_INTERRUPT -> support uart_interrupt feature
* UART_SAM -> "atmel,sam-uart"

Issues with this approach:
* This approach is limited to a 1-1 link between compatible and device driver, which is not the case for drivers which support multiple devices.
* Requires strict standardization of symbol names, which are hard to validate as many kconfig files are not used to describe a single device driver, they also source other files, describe subsystems etc.

Notes regarding this approach
* Making this solution scalable and usable out-of-tree adds complexity similar to what adding the YAML files and script
* The solution is convoluted.

galak, 2022-09-29:
> * This approach is limited to a 1-1 link between compatible and device driver, which is not compatible with drivers which support multiple devices.

This is not true, we can easily describe 1 driver to multiple dts compatibles in Kconfig, we do it today (see `drivers/adc/Kconfig.ads1x1x`):

```
config ADC_ADS1X1X
        bool "ADS1X1X driver"
        default y
        depends on DT_HAS_TI_ADS1013_ENABLED || DT_HAS_TI_ADS1014_ENABLED || \
                   DT_HAS_TI_ADS1015_ENABLED || DT_HAS_TI_ADS1113_ENABLED || \
                   DT_HAS_TI_ADS1114_ENABLED || DT_HAS_TI_ADS1115_ENABLED || \
                   DT_HAS_TI_ADS1119_ENABLED
```

galak, 2022-09-29:
One downside to the YAML file approach is we have to keep information in-sync information between the Kconfig and capabilities YAML file.  If we shift this information to just the capabilities YAML than we introduce yet another "configuration" system.

bjarki-trackunit, 2022-09-30:
> This is not true, we can easily describe 1 driver to multiple dts compatibles in Kconfig, we do it today (see `drivers/adc/Kconfig.ads1x1x`):
>
> ```
> config ADC_ADS1X1X
>         bool "ADS1X1X [...]

Corrected issues

bjarki-trackunit, 2022-09-30:
> One downside to the YAML file approach is we have to keep information in-sync information between the Kconfig and capabilities YAML file. If we shift this information to just the capabilities YAML than we introduce yet another "configuration" system.

The YAML files inform the builds system of which device drivers exist and what they are capable of. The capabilities of a device driver is determined by the source code, the Kconfig configures which sources to include for a given device driver.

I would not say the YAML files configure anything, they inform the build system of the capabilities implemented in the source code for a device driver. The only link to Kconfig is the symbol which selects the driver. The only relevant information we get from the Kconfig output is if the device driver is included or not.

Just like with the alternate Kconfig solution, synchronization must happen only if the source code changes to a degree where capabilities are altered.

The YAML file asserts which Kconfig symbol selects the device driver, if this symbol changes, the YAML file must reflect this to. 

Note that all of these capabilities will rarely change.

galak, 2022-10-04:
We already have a solution and pattern to the problem of having a driver enabled w/o devicetree node being enabled.  That is what the `depends on DT_HAS_TI_ADS1013_ENABLED` in Kconfig accomplishes.  I'm sure there are some cases that still need to be added in Kconfig.

In general we should not be having multiple drivers for the same device.  If this is really needed, Kconfig can describe the mutual exclusion between the drivers if needed.

bjarki-trackunit, 2022-10-04:
> We already have a solution and pattern to the problem of having a driver enabled w/o devicetree node being enabled. That is what the `depends on DT_HAS_TI_ADS1013_ENABLED` in Kconfig accomplishes. I'm sure there are some cases that still need to be added in Kconfig.

As mentioned in the issue, under **Concerns and Unresolved Questions**, this is expanded upon. The solution proposed with this issue also does the reverse check, is there a node which is enabled, for which any driver exists, which has not been enabled. And if so, notify the user of the existence of these drivers.

> In general we should not be having multiple drivers for the same device. If this is really needed, Kconfig can describe the mutual exclusion between the drivers if needed.

Having multiple drivers is beneficial for out-of-tree drivers, where you want to overwrite an existing driver with your own implementation. It is also useful for devices which can be interacted with in different ways, like a generic cover all driver, like MODEM_GSM_PPP, and a tailored driver, like MODEM_QUECTEL_BG9X.
Turning it around, we should not disallow or prevent having multiple drivers which support the same devices.

gregshue, 2022-10-21:
> In general we should not be having multiple drivers for the same device. If this is really needed, Kconfig can describe the mutual exclusion between the drivers if needed.

Hmm.  I would not be surprised to see a chip vendor provide a licensed module with an enhanced device driver. Providing a configuration and build system that supports this seems to be consistent with the Project mission and vision.

gregshue, 2022-10-21:
> select DEVICE_DRIVER  -> Declares that this symbol enables a device driver

AFAICT, selecting select DEVICE_DRIVER only tells the config system that at least one device driver exists in the system. Won't that always be true outside of test cases? How is the DEVICE_DRIVER setting bound to this symbol?

bjarki-trackunit, 2022-10-22:
> AFAICT, selecting select DEVICE_DRIVER only tells the config system that at least one device driver exists in the system. Won't that always be true outside of test cases? How is the DEVICE_DRIVER [...]

In the current Kconfig output `.config`, that is correct. This proposal adds a step to Kconfig.py (which has access to all individual symbols and their contexts) which creates a new output: `selects.pickle` which is a dictionary containing all symbols which select any other symbol. This output contains all individual symbols, and what they select

This Kconfig symbol
```
config MODEM_QUECTEL_BG9X
    bool "Enable quectel modem device driver"
    select DEVICE_DRIVER
    select DT_COMPAT_QUECTEL_BG95
    select DT_COMPAT_QUECTEL_BG96
    select DEVICE_DRIVER_API_NET_IF

```
becomes
```
selects = {
    "MODEM_QUECTEL_BG9X": ["DEVICE_DRIVER", "DT_COMPAT_QUECTEL_BG95", "DT_COMPAT_QUECTEL_BG96", DEVICE_DRIVER_API_NET_IF]
}
```
in the new selects.pickle file (pickle is just a object serialization format like JSON, but binary).

gregshue, 2022-10-24:
@bjarki-trackunit It seems to me that this proposal is changing the *semantics* of the 'select' keyword.  If true, then @carlescufi's response in [this #build-system message](https://discord.com/channels/720317445772017664/906521547672522752/1033376166377168896) is appropriate here:

> Ulf, the original author of Kconfiglib, was always very careful to match 1:1 the Kconfig language from upstream Linux. If we start deviating from it, we need a very good reason, because then we won't be using "Kconfig" anymore, but our own dialect.

I strongly prefer any Kconfig language deviations to be with new keywords (introducing easily recognizable extensions) rather than changing semantics (introducing inconsistencies).

bjarki-trackunit, 2022-10-24:
> @bjarki-trackunit It seems to me that this proposal is changing the _semantics_ of the 'select' keyword. If true, then @carlescufi's response in [this #build-system [...]

The semantics are not changed, select still selects a reverse dependency, we simply deduce meaning from the names of the selected symbols. Kconfig does not care if a selected symbol is named DEVICE_DRIVER, or ABCD, we can however reserve some symbol names and decide that they indicate some property.

We already do this with symbols like SERIAL_HAS_DRIVER and SOC_STM32U585XX. This solution essentially goes all the way with this approach by agreeing on a symbol naming convention that can be recognized by the build system.

mkschreder, 2022-10-26:
@bjarki-trackunit do you have some code that implements at least a portion of this proposal?

bjarki-trackunit, 2022-10-27:
> @bjarki-trackunit do you have some code that implements at least a portion of this proposal?

So far I have simply made a quick and dirty addition to kconfig.py to export the symbols and their selections, if we agree on the concept, we can go ahead with actual implementation in a PR.

> P.S. The cryptic error messages are easily solved if you know that you can just search for the ord in generated headers (build/zephyr/include/generated/device_extern.h) and from there find the offending dts node.

I would not say the error message is solved, that is a way of diagnosing the obscure error. The obscurity of the error is rooted in the "undefined reference to", and the fact that it is a compilation error. The build system should ensure that the user has included all required drivers before compilation starts, and should be able to help the user find an appropriate driver if it detects a device missing one.

mkschreder, 2022-10-27:
I think a good check to implement would be to find a way to guarantee that all device tree nodes that are 'status = "okay"' are instantiated (have a corresponding driver included). I don't have an off the cuff suggestion yet on how to do this elegantly but one possible way is to simply have a pre-build check that goes over the generated zephyr.dts (final device tree) and over the full list of files included in the build and cross references device tree compatible strings to all DT_DRV_COMPAT defines found in the list of built files (Although I have a hunch that there could be a better way by using variables generated by device instantiation macros). I would like to hear more input from others on an elegant way to implement this cross reference check.

henrikbrixandersen, 2022-10-27:
> I think a good check to implement would be to find a way to guarantee that all device tree nodes that are 'status = "okay"' are instantiated (have a corresponding driver included).

What problem are you trying to solve here? Having devictree nodes with `status = "okay"` but no Zephyr driver instantiated is a perfectly valid configuration.

bjarki-trackunit, 2022-10-27:
> I think a good check to implement would be to find a way to guarantee that all device tree nodes that are 'status = "okay"' are instantiated (have a corresponding driver included). I don't have an [...]

Crawling through code is not exactly an elegant solution IMO. DT_DRV_COMPAT is only used for drivers which only support one device compatible, some drivers support multiple compatibles. We also have many different macros which instantiate devices, NET_IF_DEFINE, DEVICE_DEFINE, DEVICE_DT_DEFINE etc. It is rather complex and will be hard to maintain.

The core of this solution is to tell the build system about the drivers, in a way which keeps that information in one place without duplication. We can then use this information to implement the behavior we want.

Your example with status="okay" we can handle more elegantly if the build systems knows which drivers exists for that compatible, if any. A device can have the status okay without any driver, but if a driver exists, we can inform the user that he may want to enable one of the compatible drivers.

The solution should be helpful, without blocking the user, and the meta information should be provided by the driver developers, to help the end users (developer and end user are often the same people, but still :))

mkschreder, 2022-10-28:
> What problem are you trying to solve here? Having devictree nodes with `status = "okay"` but no Zephyr driver instantiated is a perfectly valid configuration.

What is the logical sense of that? If one sets something to "status = 'okay'" then one almost expects the driver to automatically be enabled and included so that the node is in fact "okay". The fact that we need to enable the driver as well is almost a bit of a nuisance.

mkschreder, 2022-10-28:
@gregshue at this time I'm just brainstorming ideas and it's good to map out possible solutions and issues with each. 

What about making the mappings from device tree bindings to device drivers? One could for instance link from "dts/bindings/watchdog/nxp,lpc-wwdt.yaml" to "drivers/watchdog/wdt_mcux_wwdt.c" and always in the same source tree. So for example, if one wants to add a custom driver, one would extend the existing binding in the directory tree of the custom driver and then the linking will search directories first in the extended yaml location, then in the location of any base bindings. 

What are issues of linking dts bindings to one or more device driver implementations?

bjarki-trackunit, 2022-10-28:
> > > I think a good check to implement would be to find a way to guarantee that all device tree nodes that are 'status = "okay"' are instantiated (have a corresponding driver included).
> > 
> > 
> > What problem are you trying to solve here? Having devictree nodes with `status = "okay"` but no Zephyr driver instantiated is a perfectly valid configuration.
> 
> What is the logical sense of that? If one sets something to "status = 'okay'" then one almost expects the driver to automatically be enabled and included so that the node is in fact "okay". The fact that we need to enable the driver as well is almost a bit of a nuisance. But of course we can have multiple drivers that implement a compatible so I understand the necessity of the extra step of picking a driver - however, I would expect that having status = okay and no driver included in build should in fact generate an error.

Not all nodes have drivers, even when the node is enabled, and some nodes are supported by multiple drivers, so we need the user to enable the one they want to use.

gregshue, 2022-10-28:
> for instance link from "dts/bindings/watchdog/nxp,lpc-wwdt.yaml" to "drivers/watchdog/wdt_mcux_wwdt.c" ...

Consider that the link needs to still be meaningful when the driver implementation and/or filename changes. (E.g., when a file is refactored into separate files, when a driver supports some of the versions of a HW block.)

> ... and always in the same source tree.

For end users that are limited to _reuse_ an upstream repository (no additional commits) fixing any mapping errors must be done through configuration settings and mappings in a different repository.  If both repositories have parallel directory structures, is it the same source tree? 🤔 

> What are issues of linking dts bindings to one or more device driver implementations?

In addition to those above, consider:
- Drivers provide an API/behavior and are written to use one or more API/register sets/behaviors.
- As driver versions evolve, APIs get implicitly or explicitly versioned when deprecations, enhancements, and extensions get realized.
- When a FW image supports multiple boards/SOC/block versions the adjustment needs to happen during start time. In this case multiple drivers for the same driver API may need to be built into the image.
- As mentioned above, sometimes an image will never use a specific HW block.  The associated driver may need to NOT be in the image.  Similarly, there are points in time where is valid to have a hardware block declared without having a driver (e.g., prior to driver development).

gregshue, 2022-10-28:
@nashif @MaureenHelm The considerations I mentioned above are other examples of User Needs that need to be part of the conversation.  Should I create issues for these as well?

mkschreder, 2022-10-28:
@gregshue do you see any reason to ever have a block in device tree enabled (status = "okay") and not have a driver for it? I'm just curious what your take on this is. Here I'm talking only about nodes that have a compatible string (ie expect a driver to handle them).

gregshue, 2022-10-28:
🤔 ... Different users, different needs ... Is this good enough?

As a contracted designer of bitstreams for FPGA with embedded CPU cores,
I need to generate and validate device tree files describing the memory-mapped blocks in the bitstream,
so that my client's own firmware engineers have DT files ready for whenever they choose to implement drivers.