zephyrproject-rtos/zephyr #50789: west: runners: blackmagicprobe: Doesn't work on windows due to wrong path separator

aeremin, 2022-09-28:
It seems that `blackmagicprobe` runner is broken on windows. Running
```
west flash --runner blackmagicprobe COM1
```
results in error like this

```
D:Devhardwarezephyrfirefly_zephyrbuildzephyrzephyr.elf: No such file or directory.
```
(actual file location is `D:\Dev\hardware\zephyr\firefly_zephyr\build\zephyr\zephyr.elf` in that case).

This happens because:
1) `self.elf_file` is set [here](https://github.com/zephyrproject-rtos/zephyr/blob/b56310d4d4ff9e7725388246ab2fbb3ae1905272/scripts/west_commands/runners/blackmagicprobe.py#L18) to the path containing backslashes.
2)  gdb (even on windows) doesn't recognise backslashes for `load` command, it seem to just discard them.

Potential, but hacky fix is to do something like `self.elf_file = cfg.elf_file.replace("\\", "/")` which fixes the issue.

mbolivar-nordic, 2022-10-07:
Thanks for the report. The Python standard library has a non-hacky facility for this that we use in other places (see e.g. the openocd runner) which I've used in the linked PR fixing this issue.

aeremin, 2022-10-07:
Thanks for the quick fix, appreciate it!

PR #51089 (2022-10-10): west: runners: blackmagicprobe: fix on windows
We need to give gdb slash-separated paths to ELF files on Windows. Just make sure we always have slash-separated paths. We use this technique in various places around the tree.

Fixes: #50789
commit 98bcc2d8 (2022-10-10): west: runners: blackmagicprobe: fix on windows

We need to give gdb slash-separated paths to ELF files on Windows.
Just make sure we always have slash-separated paths. We use this
technique in various places around the tree.

Fixes: #50789
closed 2022-10-10 by carlescufi