zephyrproject-rtos/zephyr #50795: Add NAND Flash driver and NFTL driver to enable LittlefS to run on NAND Flash

rogeryou, 2022-09-29:
**Is your enhancement proposal related to a problem? Please describe.**
A clear and concise description of what the problem is.
This issue is an extension of #46399.
At present, LittleFS only supports running on NOR Flash. However, some users want LittleFS to run on NAND Flash, so we are currently developing NFTL and NAND Flash drivers to achieve this goal.
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
The architecture of the entire driver is shown in the figure below. SPI NAND Flash and ONFI NAND Flash will be supported. We have submitted a PR on SPI NAND Flash.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered. 

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

deyro, 2023-01-19:
Hi, any plans on when this will be available to users?
Thanks

clockis, 2023-06-30:
Greetings, are there any updates on this? I think there is much demand for this as many applications utilize NAND flash.

fwittenfeld, 2023-08-02:
Any chance to get early access to the project?

rogeryou, 2023-08-02:
We  will submit a PR about NFTL 2 weeks later.@fwittenfeld@clockis

rogeryou, 2023-08-08:
Hi ,
We have previously developed NFTL and littleFS based on mbed for using NAND Flash. You can refer to this [link ](https://github.com/ARMmbed/mbed-os/pull/14931)for more information.

kyrreaa, 2023-09-18:
+1 here too.
Currently on NRF5340 and have our own NAND version of the NRF QSPI NOR driver but performance is very poor. (Hardcoded instructions in the NRF for QSPI requires use of custom commands and slow manual sequence.)

Yaxit, 2024-02-16:
+1 a solid interface could help to bring so many projects up to speed

kyrreaa, 2024-02-16:
I actually ended up rewriting my driver to use SPIM instead on the NRF5340 as this allowed fully async operations in large read/writes instead of having to bypass the NOR instruction flow with custom transfers to do NAND. Took a huge load of the cpu.
(QSPI hardware on NRF5340 has hardcoded instructions for NAND that cannot be bypassed.)

Yaxit, 2024-02-16:
@kyrreaa I'd be interested in having a look at what you did, if possible :)

D-Buckingham, 2025-01-15:
I created a Nand flash multistack implementation with the flash translation layer dhara. However, I would love it, if someone could help me to integrate it and contribute to the community, since it is the first time I would be doing something like this.  Below a rough diagram.  I won't have time to maintain the library, but would love to contribute. Though I won't even have a lot of time to do the integration, so please be patient. The implementation is thouroughly tested etc. One can even have multiple flashes in parallel and the wear leveling, bad block management and ECC will be performed over all of them. The file system is fat, which makes sense, since we can not do atomic accesses to the NAND, since the page size is usually the smallest package that can be saved or retrieved from the flash.

[image]

de-nordic, 2025-01-15:
> I created a Nand flash multistack implementation with the flash translation layer dhara. However, I would love it, if someone could help me to integrate it and contribute to the community, since it [...]

You need to elaborate a little more on how it is done, how it uses and/or modifies our subsystems, etc.
Best would be to just create new issue for that.
It is hard to tell us more. The rough diagram does not help that much to identify required work.

de-nordic, 2025-01-21:
> I just made my project public. See here: https://github.com/D-Buckingham/NAND_flash.git If you stumble upon some errors/ bugs, please let me know. It comes with many advantages, such as less power consumption, lower cost, higher storage density than NOR flashes etc (and, or even SD cards). I did some power profiling etc to check its performance and did many other tests. If you are interested in them let me know. Actually, I wrote it a few months ago when I still had some time on my hands, but never came to the point to share it. Please don't judge too much, since this was still part of my first project in Zephyr and I am no pro as well. ;) Currently, I have my hands full with personal matters that have priority, but I will at some point be back and push it to Zephyr. Though this might take some time. Until then, enjoy the code. ;)

OK, so you have ported dhara, which would be fine if that would not define NAND specific API. That will not fly,
We have Flash API and if that needs, and it does, extensions to support NAND devices this is fine, but adding additional API, aside from existing Flash and EEPROM, to support devices which area the same type of device with different characteristics, is just fragmenting API for accessing storage. It is like you are trying to add separate API for supporting SSD disks even though they have same interface as spinning rust, just additional features.

The NAND should go under Flash API (https://github.com/zephyrproject-rtos/zephyr/pull/50690/files), the dhara could be implemented as virtual flash device as it, as far as I remember, does not need the ECC/USER fields at all and could work on any Flash device.
Note that implementing additional API also means that other subsystems have to serve all these API to be able to use the future.

daniel-0723, 2025-02-07:
Can dhara NFTL be integrated into LittleFS instead of FatFS?

de-nordic, 2025-02-07:
> Can dhara NFTL be integrated into LittleFS instead of FatFS?

Littlefs itself has wear leveling, so it does not really make that much sense.

Yaxit, 2025-02-07:
AFAIK LittleFS would also not perform well on NAND memory. I though it's more NOR-oriented, although I don't remember the exact reason. Am I wrong?

de-nordic, 2025-02-07:
> AFAIK LittleFS would also not perform well on NAND memory. I though it's more NOR-oriented, although I don't remember the exact reason. Am I wrong?

You are correct. Huge erase size. In hundred of kb, which means you take a lot of time to move things around when compacting data.

So here actually ntlf helps.
But that probably should be included into Disk Access subsystem, which also does caching. The Flash API is base level API exposing features of device as they are. Although device may do wear leveling and block management itself, or driver may.

D-Buckingham, 2025-02-10:
> OK, so you have ported dhara, which would be fine if that would not define NAND specific API. That will not fly, We have Flash API and if that needs, and it does, extensions to support NAND devices [...]

Could you elaborate this a bit more?
Maybe I misunderstood you, but your concern is, that I introduce a new API? (Which I did not ??? )
As far as I remember, all the NOR flash, storage integrations are connected to the disk "API". See my picture, where the disk layer is called. The disk layer (I will call it that, it is provided by Zephyr, built in, which manages all the different storage systems) actually puts constraints on what kind of handles/functions the underlying storage integrations need to provide. I put my focus so that I do extend the existing and bind to that API, not creating a new one and "fragmenting" Zephyr.
Meaning it is under the "flash API" as you called it. 
ALL THE FLASH API FUNCTIONALITIES CAN BE USED PROPERLY SINCE IT IS BOUND TO THE SYSTEM THROUGH THE DISK LAYER/API.
About the littlefs you are absolutely right. It doesn't make sense to discuss this more. For the people that are still interested, here a short explanation from the top of my head: 
Littflefs doesn't make sense at all as @de-nordic stated correctly. We have more challenges with NAND flashes to overcome than "just" wear leveling. Plus we can not make atomic accesses, meaning working with extra small data packages as in NOR flashes. The smallest data package in NAND flashes is a page. Either way bad block management, error correction code and many more challenges need to be overcome, to ensure data integrity etc.

After some research DHARA seemed to be a good / leightweight solution for embedded systems. I tried to make the implementation as modular as possible. Tried to improve dhara a bit etc. so it might even work more efficiently especially for energy constraint systems. 

If you need further information about the NAND flash multistack implementation, let me know. Althgough it has been some time since I implemented it. Maybe from my repo it is not that clear. So sorry for that ;)
Plus as mentioned, this was my first bigger Zephyr project. So if you have any recommendations and tips, let me knwo :)

de-nordic, 2025-02-11:
> Could you elaborate this a bit more? Maybe I misunderstood you, but your concern is, that I introduce a new API? (Which I did not ??? ) 

This is literally API interface https://github.com/D-Buckingham/NAND_flash/blob/c77f9cb78aacc6f3b29c7a3d7766f991124d6f8e/src/NAND_FLASH_DHARA/inc/nand_driver.h

> As far as I remember, all the NOR flash, storage integrations are connected to the disk "API".

Not true.

LittleFS has been originally using direct Flash API access, support of Disk Access (aka block access) has been added later ron.
NVS, ZMS, FCB all use direct Flash API access.
Stream Flash, for writing stream of data uses Flash API directly.
Disk API is used by FAT, ext2 natively, and LittleFS as addition. From all of the three the last one is closes to the embedded FS and it is not originally written for block access - aka Disk Access.

Disk Access is provided for stuff that has been built for block access, like FAT file system, and provides caching so that memories where erase block size is larger than block of a software requirement. 
Disk Access can be backed by various subsystems, including Flash API and SD/MMC card access.

> See my picture, where the disk layer is called. The disk layer (I will call it that, it is provided by Zephyr, built in, which manages all the different storage systems) actually puts constraints on [...]

Not as I called it, that is how it is defined: https://docs.zephyrproject.org/latest/doxygen/html/group__flash__interface.html

> ALL THE FLASH API FUNCTIONALITIES CAN BE USED PROPERLY SINCE IT IS BOUND TO THE SYSTEM THROUGH THE DISK LAYER/API

Try shouting more.

> (...) The smallest data package in NAND flashes is a page. 

Same is for NOR. Actually no, wait, smallest one is sector for NAND. No... wait. Yeah, because this naming does not matter: the definition for purpose in a context of a device does.
The problem is how do you manage write block and erase block, and naming here pages/sectors/blocks does nothing for discussion, because this naming is completely mixed up between devices and software solutions above them (and sometimes vendors).

The problem with managing device, for data is how to manage moving around the largest unit of data, it does not matter whether this is NOR or NAND, although some characteristic is more applicable to one type of device than another, i.e. NOR will usually have erase block of <=4k and NAND has ~132k (including obb/ecc), and eeprom can have erase block too!

This means that if you have FS with 512 write sector (like designed for a floppy), FAT for example, to be able to write it again, in case of NOR device you have to move around up to 8 blocks of data, when you have to do erase before write, for 4k erase page device, in case of a NAND up to 256 sectors have to be moved, for 132k erase block device (yes page vs block here, because that is how naming is mixed up).

Here is layout of w25n, as visible from the write perspective, where you are able to write data by 512byte sectors, that you can write with, another confusing name, "Partial Page Program" command.

[image]

If you only want to implement NFTL for Disk Access, I am happy about  it, but you still need to access Flash Device via Flash API, to avoid fragmentation when we already have other NAND flash driver implemented under that API.

If you want to implement the NFTL via Flash API, so that user can have special device that does the NFTL while user does `flash_read`/`flash_write` on it  I am also happy to see that, because I see usage like MCUboot using NAND to swap images without dealing with 128k erase blocks, or pages, or whatever.