zephyrproject-rtos/zephyr #50812: MCUmgr udp sample fails with shell - BUS FAULT

maxstreitberger, 2022-09-29:
**Describe the bug**
I am trying to run the `mcumgr` sample in `zephyr/samples/subsys/mgmt/mcumgr/smp_svr/` with an udp connection and a shell. However, I can only send one command when I compile with `overlay-shell-mgmt.conf` and `overlay-udp.conf`:
```
west build -b nucleo_f429zi zephyr/samples/subsys/mgmt/mcumgr/smp_svr -- -DOVERLAY_CONFIG="overlay-shell-mgmt.conf;overlay-udp.conf" -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-2048.pem\" && west flash
```

After flashing, the code boots as usual, and I can execute this command, which works fine:
`./mcumgr -c stm32udp echo hello`

But if I rerun the same command, I get a bus fault:
```
[00:00:19.573,000] <err> os: ***** BUS FAULT *****
[00:00:19.573,000] <err> os:   Precise data bus error
[00:00:19.573,000] <err> os:   BFAR Address: 0x1ffaef92
[00:00:19.573,000] <err> os: r0/a1:  0x00001000  r1/a2:  0x03ff5df2  r2/a3:  0x80000000
[00:00:19.573,000] <err> os: r3/a4:  0x1ffaef90 r12/ip:  0x00000000 r14/lr:  0x08030429
[00:00:19.573,000] <err> os:  xpsr:  0x21000000
[00:00:19.573,000] <err> os: Faulting instruction address (r15/pc): 0x08030456
[00:00:19.573,000] <err> os: >>> ZEPHYR FATAL ERROR 0: CPU exception on CPU 0
[00:00:19.573,000] <err> os: Current thread: 0x200022f0 (mcumgr smp)
[00:00:19.635,000] <err> os: Halting system
```

However, if I only use the `overlay-udp.conf`:
```
west build -b nucleo_f429zi zephyr/samples/subsys/mgmt/mcumgr/smp_svr -- -DOVERLAY_CONFIG="overlay-udp.conf" -DCONFIG_MCUBOOT_SIGNATURE_KEY_FILE=\"bootloader/mcuboot/root-rsa-2048.pem\" && west flash
```
I can execute the `mcumgr` command as often as I want.

I tracked the issue down to the `free_chunk` function in `zephyr/lib/os/heap.c`, where it fails on the left chunk. So I tried to set the following configurations, but none of them solved the issue.
```
CONFIG_HEAP_MEM_POOL_SIZE
CONFIG_MCUMGR_BUF_SIZE
CONFIG_NET_BUF_DATA_POOL_SIZE
CONFIG_MCUMGR_SMP_UDP_STACK_SIZE
```

**Expected behavior**
The board should send back `hello`

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK 15.0
 - `main` branch
 - `nucleo_f429zi` board

nordicjm, 2022-09-30:
Can't replicate this here on an nrf5340 with the same overlays:
```
｢182｣$ mcumgr --conntype udp --connstring=[192.168.3.220]:1337 image list
Images:
 image=0 slot=0
    version: 0.0.0
    bootable: true
    flags: active confirmed
    hash: 5d9298e961a6e3270731a7fd732874cd49733d6cde987424e90cef734c7f7a41
Split status: N/A (0)
｢183｣$ mcumgr --conntype udp --connstring=[192.168.3.220]:1337 shell exec kernel version
status=0

Zephyr version 3.2.0
｢184｣$ mcumgr --conntype udp --connstring=[192.168.3.220]:1337 shell exec kernel version
status=0

Zephyr version 3.2.0
｢185｣$ mcumgr --conntype udp --connstring=[192.168.3.220]:1337 echo hello
hello
｢186｣$ mcumgr --conntype udp --connstring=[192.168.3.220]:1337 echo hello
hello
```

nordicjm, 2022-09-30:
Can't reproduce any issue with `native_posix` either, sending very large kernel shell commands

nordicjm, 2022-10-04:
Have tested this with `nucleo_f746zg` with the same overlay and cannot replicate any crash.

nordicjm, 2022-10-12:
@maxstreitberger Can you provide any more information on this issue or close it if it's not an issue?

maxstreitberger, 2022-10-12:
@nordicjm I tried your suggestion to increase the `CONFIG_MCUMGR_SMP_WORKQUEUE_STACK_SIZE`, but that also didn't solve the issue. I also have the same issue when using the `stm32h735g_disco` board.
Further, I tried the sample with a serial connection, and everything worked as expected

nordicjm, 2022-10-12:
@maxstreitberger I've retried on the nucleo_f746zg again with the current main and found no problems, I've also wired up an enc28j60 SPI ethernet chip on a nucleo_f401re and tried it with the same settings above and also found no problems. The only other thing I can think is you're using IPv6 since all my testing has been with IPv4. If that's not it then I don't think there's anything more I can do to investigate this issue.

nordicjm, 2022-10-12:
And not IPv6, just tested both nucleos with IPv6 and it works fine on those.

maxstreitberger, 2022-10-13:
@nordicjm I also don't use IPv6. Did you make any changes to the configuration files?

nordicjm, 2022-10-13:
@maxstreitberger Configuration files are left as-is, the only change made is updating the static IP

maxstreitberger, 2022-10-13:
Ok, so resetting everything seems to have solved the issue! @nordicjm thank you for your help.

closed 2022-10-13 by maxstreitberger