zephyrproject-rtos/zephyr #50841: tests: picolibc SRAM overflow on nrf52dk_nrf52832

nordic-piks, 2022-09-30:
**Describe the bug**
The tests:
tests/kernel/common/kernel.common.picolibc
tests/lib/c_lib/libraries.picolibc
tests/lib/mem_alloc/libraries.libc.picolibc.mem_alloc
tests/lib/sprintf/libraries.picolibc.sprintf_new

Does not fit into SRAM - SRAM overflow.

Observed at:
- nrf52dk_nrf52832

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf52dk_nrf52832 connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/kernel/common -p nrf52dk_nrf52832 --device-testing --device-serial /dev/ttyACM0 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid tests output

**Impact**
Not clear

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.14.1
 - Commit SHA or Version used: v3.2.0-rc3-98-g1c953834d671

stephanosio, 2022-09-30:
SRAM overflow is not a bug.

stephanosio, 2022-09-30:
While SRAM overflow in itself is not a bug, it is strange that the picolibc version of `tests/lib/c_lib` is using much more SRAM than the minimal libc version on `nrf52dk_nrf52832`.

For `qemu_cortex_m3`, there is only a negligible difference:

```
# qemu_cortex_m3 minimal libc

Memory region         Used Size  Region Size  %age Used
           FLASH:       43436 B       256 KB     16.57%
            SRAM:        7272 B        64 KB     11.10%
        IDT_LIST:          0 GB         2 KB      0.00%

# qemu_cortex_m3 picolibc

Memory region         Used Size  Region Size  %age Used
           FLASH:       45380 B       256 KB     17.31%
            SRAM:        7624 B        64 KB     11.63%
        IDT_LIST:          0 GB         2 KB      0.00%
```

stephanosio, 2022-09-30:
# nrf52840dk_nrf52840, minimal libc

Memory region         Used Size  Region Size  %age Used
           FLASH:       96420 B         1 MB      9.20%
            SRAM:       15112 B       256 KB      5.76%
        IDT_LIST:          0 GB         2 KB      0.00%

```
# nrf52840dk_nrf52840, picolibc

[992/992] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      163628 B         1 MB     15.60%
            SRAM:       80736 B       256 KB     30.80%
        IDT_LIST:          0 GB         2 KB      0.00%
```

stephanosio, 2022-09-30:
The direct cause seems to be that the nRF52 platforms build with `CONFIG_USERSPACE=y` by default and it has `CONFIG_MPU_REQUIRES_POWER_OF_TWO_ALIGNMENT=y`:

https://github.com/zephyrproject-rtos/zephyr/blob/3110aa3a93171ce0e5bc62a1e37cf3235fac1e86/lib/libc/picolibc/Kconfig#L11-L15

https://github.com/zephyrproject-rtos/zephyr/blob/3110aa3a93171ce0e5bc62a1e37cf3235fac1e86/lib/libc/picolibc/libc-hooks.c#L94-L95

This leads to `CONFIG_PICOLIBC_HEAP_SIZE=65536` and static allocation of 64KB heap in the malloc partition resulting in an increased memory requirement.

With `CONFIG_PICOLIBC_HEAP_SIZE=-1`, the actual memory usage is very similar to that of the minimal libc, which defaults to `CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=0` (equivalent of `CONFIG_PICOLIBC_HEAP_SIZE` for the minimal libc).

```
#  nrf52840dk_nrf52840, picolibc, CONFIG_PICOLIBC_HEAP_SIZE=-1

Memory region         Used Size  Region Size  %age Used
           FLASH:       98076 B         1 MB      9.35%
            SRAM:       15200 B       256 KB      5.80%
        IDT_LIST:          0 GB         2 KB      0.00%
```

Similarly, building the minimal libc with `CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=65536` results in a significant increase memory requirement:

```
#  nrf52840dk_nrf52840, minimal libc, CONFIG_MINIMAL_LIBC_MALLOC_ARENA_SIZE=65536

Memory region         Used Size  Region Size  %age Used
           FLASH:      227616 B         1 MB     21.71%
            SRAM:      146260 B       256 KB     55.79%
        IDT_LIST:          0 GB         2 KB      0.00%
```

In conclusion, this is due to the different **default** malloc heap size configuration for the minimal libc (0 bytes) and the picolibc (65536 bytes).

stephanosio, 2022-09-30:
@nordic-piks [can you please check if https://github.com/zephyrproject-rtos/zephyr/pull/50854 reso](https://github.com/zephyrproject-rtos/zephyr/pull/50854)

nordic-piks, 2022-09-30:
@stephanosio PR works fine, thanks.

PR #50854 (2022-09-30): lib: picolibc: Use smaller default heap size
The picolibc heap size configuration (`CONFIG_PICOLIBC_HEAP_SIZE`) is used to set the statically allocated malloc heap size when userspace is enabled.

The current default heap size of 1048576 bytes (MMU) and 65536 bytes (MPU) is too large for most platforms that Zephyr supports and may result in the picolibc tests being filtered out due to the increased memory footprint of the compiled image (i.e. SRAM overflow).

This commit updates the default picolibc heap size to a more reasonable 16384 bytes for MMU platforms and 1024 bytes for MPU platforms.
 [... cut]
closed 2022-09-30 by fabiobaltieri