zephyrproject-rtos/zephyr #50844: zcbor module apis which are used for mcu boot functionality are not building in cpp file against v3.1.0

KarthikeyanM-eaton, 2022-09-30:
**Describe the bug**
We have created mcuboot test code, added following function in cpp file and we are able to build against zephyr v3.0.0. But if we try to build same cpp file against zephyr v3.1.0, It is giving build error, or else if we convert same file into c it is building success. 

[image]

We observed, the zcbor module is not present in zephyr v3.0.0 and it is newly added in zephyr v3.1.0. It is looks like zcbor_encode.h apis are not supported in cpp and It will only support c file. we need c and cpp compatability for our project. Kindly provide the solution.

zephyrproject_3_1_0\modules\lib\zcbor\include\zcbor_encode.h  

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
    We are using zephyr v3.1.0, SDK 0.14.2 and nucleo_f429zi board. The project created in cpp format.
 - What have you tried to diagnose or workaround this issue? 
     same file if we convert c , it is building.

 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. create above shared screenshot function in cpp 
2. west build -b nucleo_f429zi ./ --pristine
3. See error

**Expected behavior**
It should build in cpp environment

**Impact**
 Without this we cant release to our application team.

**Logs and console output**

[image]

**Environment (please complete the following information):**
 - OS: (Windows)
 - zephyr SDK 0.14.2 and zephyr v3.1.0

stephanosio, 2022-09-30:
@de-nordic looks like there is a somewhat serious bug in zcbor ...

To quote [GCC manual](https://gcc.gnu.org/onlinedocs/gcc/Compound-Literals.html):

> In C, a compound literal designates an unnamed object with static or automatic storage duration. In C++, a compound literal designates a temporary object that only lives until the end of its full-expression. As a result, well-defined C code that takes the address of a subobject of a compound literal can be undefined in C++, so G++ rejects the conversion of a temporary array to a pointer. For instance, if the array compound literal example above appeared inside a function, any subsequent use of foo in C++ would have undefined behavior because the lifetime of the array ends after the declaration of foo.

even in C, this can (if not already) result in a dangling pointer depending on the compiler implementation ...

laurenmurphyx64, 2022-10-04:
@oyvindronningstad Hi, could you update the zcbor revision is Zephyr to include the fix?

oyvindronningstad, 2022-10-05:
Yes, the plan is to release zcbor and update Zephyr within a week or two. Inspired by this, I also made the following C++ related changes: https://github.com/NordicSemiconductor/zcbor/pull/258

closed 2022-12-19 by github-actions[bot]