zephyrproject-rtos/zephyr #50853: STM32F7 series can't run at frequencies higher than 180MHz

vgeneves, 2022-09-30:
**Describe the bug**

When configuring clock with a frequency higher than 180MHz the board is not able to start.
It hangs just after configuring the clock `clock_stm32_ll_common.c:stm32_clock_control_init`).

By looking at the code :
`LL_SetFlashLatency` checks `LL_PWR_IsEnabledOverDriveMode` for frequenies higher than 180HMz.

Unfortunately overdrive mode is enable later in `set_up_plls`.

By calling `set_up_plls` before configuring the flash wait states I am able to boot the board correctly.

This as been tested on a custom board with stm32f723 (originally based on ST STM32F723E Discovery) running Zephyr 3.1
Any sample app built with stm32f7 board should have the bug.

**Expected behavior**
App should be able to run completely.

**Impact**
The board does not start.

**Logs and console output**
If applicable, add console logs or other types of debug information
e.g Wireshark capture or Logic analyzer capture (upload in zip archive).
copy-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue. (if unable to obtain text log, add a screenshot)

**Environment (please complete the following information):**
 - OS: Unintu 1804
 - Toolchain Zephyr 0.14.2
 - Zephyr 3.1 (2ddd73feafd3316af2c547c34d6969bea637d5c6)

**Additional context**

Clock config in device tree:
`&clk_hse {
	clock-frequency = <DT_FREQ_M(25)>;
	status = "okay";
};

&pll {
	div-m = <25>;
	mul-n = <432>;
	div-p = <2>;
	div-q = <9>;
	clocks = <&clk_hse>;
	status = "okay";
};

&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(216)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <4>;
	apb2-prescaler = <2>;
};`

FRASTM, 2022-10-11:
With stm32F746 configured for sysclock = 216MHz, 

```
&pll {
	div-m = <4>;
	mul-n = <216>;
	div-p = <2>;
	div-q = <9>;
	clocks = <&clk_hse>;
	status = "okay";
};

&rcc {
	clocks = <&pll>;
	clock-frequency = <DT_FREQ_M(216)>;
	ahb-prescaler = <1>;
	apb1-prescaler = <2>;
	apb2-prescaler = <1>;
};
```

Event if the samples/hello world is functionnale, the voltage scaling is on scale1 and  PWR CR1 shows that ODEN is not set ; FlashLatency is reported at 5.

FRASTM, 2022-10-12:
@vgeneves could you please look at the PR and check if it helps

closed 2022-11-22 by fabiobaltieri