zephyrproject-rtos/zephyr #50872: Error while installing python dependecies

WajahatRiaz, 2022-10-01:
I'm setting up zephyr for my eclipse IDE using the guide from the following link:
[https://www.nxp.com/docs/en/training-reference-material/BUILD-YOUR-FIRST-ZEPHYR-APPLICATION-ON-I.MX-RT.pdf](url)

I use the following command to install dependecies:
`pip3 install -r zephyr/scripts/requirements.txt
`

But i'm getting the following error message:
[image]

WajahatRiaz, 2022-10-01:
I believe I found a solution. It's trivial but solves the issue. Open the requirements.tx file on your windows and simply change **hub==2.0** to **hub==3.0.0**. Then save the file and run the following command:
`pip3 install --user -r zephyrproject/zephyr/scripts/requirements.txt`

carlescufi, 2022-10-05:
This report is incorrect:

- `hub` is not a requirement, direct or indirect, in Zephyr
- `hub` is now called `deeplake`, and again, is not a requirement

closed 2022-10-05 by carlescufi