zephyrproject-rtos/zephyr #50895: ADC Voltage Reference issue with STM32U5 MCU

vibhor-meshram, 2022-10-03:
Bug Description:-

I am using nucleo_u575zi_q board and wanted to implement ADC on it. my overlay file looks as follows:-

/ {
    zephyr,user {
        /* adjust channel number according to pinmux in board.dts */
        io-channels = <&adc1 1>;
    };
};

&adc1 {
    #address-cells = <1>;
    #size-cells = <0>;

    channel@1 {
        reg = <1>;
        zephyr,gain = "ADC_GAIN_1";
        zephyr,reference = "ADC_REF_INTERNAL";
        zephyr,acquisition-time = <ADC_ACQ_TIME_DEFAULT>;
        zephyr,resolution = <14>;
    };

};

I am trying to test the ADC code with different ADC reference voltages (like  VDD, VDD/2, External etc), when I change the reference voltage, it doesn't have any effect on output ADC value. Also while debugging, I found that in adc_stm32.c file the following piece of code that allows only ADC_INTERNAL reference to set.

 if (channel_cfg->reference != ADC_REF_INTERNAL) {
        LOG_ERR("Invalid channel reference");
        return -EINVAL;
    }

Expected Behaviour:-
The ADC raw values should saturate when input voltage provided to ADC channel is equal to Vref set in the overlay file. For example if Vref is set to VDD/2, the ADC values should saturate if 1.65 volts is provided to ADC input channel.

Platform:-
Windows 10
Zephyr Version 3.1.0

erwango, 2022-10-03:
@vibhor-meshram There were somesignificant changes on ADC during last release. Did you check status on V3.2 ?

vibhor-meshram, 2022-10-04:
@erwango I tried to compile adc example using zephyr 3.2.0 version for nucleo_u575zi_q board. I set the reference to "ADC_REF_VDD_1_2". On compilation I got following error:-
`C:/Users/zephyrproject/zephyr/drivers/gpio/gpio_stm32.c:714: undefined reference to `LL_PWR_EnableVddIO2'`

I fixed this error by making changes to gpio_stm32.c code. After successful compilation, the serial terminal output was as follows:-

`*** Booting Zephyr OS build v3.2.0-rc3-219-g2946a1c0786c  ***
Could not setup channel #0 (-22)
*** Booting Zephyr OS build v3.2.0-rc3-219-g2946a1c0786c  ***
Could not setup channel #0 (-22)`

I tried the above using reference as "ADC_REF_EXTERNAL0" as well and got the same results.

FRASTM, 2022-10-04:
With the nucleo_u575zi_q, the samples/drivers/adc gives:

- adc@42028000, channel 1: 0 = 0 mV   when pin A5 on GND
- adc@42028000, channel 1: 16383 = 3299 mV   when pin A5 on 3V3 or 5V 
- adc@42028000, channel 1: 9087 = 1830 mV  when pin A5 on 1V8

erwango, 2022-10-04:
> @erwango I tried to compile adc example using zephyr 3.2.0 version for nucleo_u575zi_q board. I set the reference to "ADC_REF_VDD_1_2". On compilation I got following error:-

Please run "west update".

vibhor-meshram, 2022-10-04:
> With the nucleo_u575zi_q, the samples/drivers/adc gives:
> 
> * adc@42028000, channel 1: 0 = 0 mV   when pin A5 on GND
> * adc@42028000, channel 1: 16383 = 3299 mV   when pin A5 on 3V3 or 5V
> * adc@42028000, channel 1: 9087 = 1830 mV  when pin A5 on 1V8

@FRASTM I get similar results when using reference as "ADC_REF_INTERNAL". I want to use External reference or may be VDD.

FRASTM, 2022-10-04:
The error "Could not setup channel #0 (-22)" comes from the adc_stm32.c driver that only accepts the ADC_REF_INTERNAL as reference. 
This `channel_cfg->reference` is an enum value that should be converted to a nb of mv if this has to be compared :  the  channel_cfg->reference against DT_INST_PROP(0, vref_mv) is  expressed in mV.

vibhor-meshram, 2022-10-04:
> 

@FRASTM Thanks for your resolution. Upon trying the the above solution my observation are as follows:-
1. After setting reference voltage as "ADC_REF_VDD_1" and adding vref-mv, I still get the adc error - `Could not setup channel #0 (-22) `. I had to comment the following line of code from adc_stm32.c

`	if (channel_cfg->reference != ADC_REF_INTERNAL) {
		LOG_ERR("Invalid channel reference");
		return -EINVAL;
	}
`

2. I set reference to "ADC_REF_VDD_1_2" and vref-mv to "1650". It had no effect on the raw values of ADC. The output is as follows when supplied 1.8 V to adc input channel ( 5 ).
`ADC reading:
- adc@42028000, channel 1: 9004 = 906 mV
ADC reading:
- adc@42028000, channel 1: 8942 = 900 mV
ADC reading:
- adc@42028000, channel 1: 8901 = 896 mV
ADC reading:
- adc@42028000, channel 1: 8844 = 890 mV`
Here I see no effect of reference voltage on ADC raw values. Ideally it should have shown 16383 as I am supplying more than reference voltage to the ADC input channel.

FRASTM, 2022-10-05:
Yes, the stm32 adc driver is limited to the ADC_REF_INTERNAL mainly, I guess, due to the hw design where the input pin VREF+ is VDDA  = 3.3V
When input pin channel 1 is 1.8 V this is half the physical Vref voltage (on the hw).

vibhor-meshram, 2022-10-06:
> 

I removed solder points that connects VDDA to Vref+ and now I am able to use external vref IC to provide reference voltage to my board. This worked without making any changes in the code ( Keeping the reference as ADC_REF_INTERNAL ). I am not sure about the enum provided for voltage reference. Is it only used for calculation purpose (convert raw data to mv) or it does some register configuration to use internal reference?

FRASTM, 2022-10-06:
The ADC_REF_INTERNAL is for calculation (giving the voltage for the Full range of ADC)

vibhor-meshram, 2022-10-06:
Is there any way I could configure internal VREFBUF register to use as reference for ADC?

FRASTM, 2022-10-06:
There is an internal channel (0) in the ADC1  to measure the V Ref.
It works more or less like Vbat (drivers/sensor/stm32_vbat/stm32_vbat.c) or temp sensor  (drivers/sensor/stm32_temp/stm32_temp.c) but is not implemented
Then I guess the measured value  should be injected to be variable insted of ADC_REF_INTERNAL constant

vibhor-meshram, 2022-10-07:
Thanks @FRASTM for your help. I am closing the issue now.

closed 2022-10-07 by vibhor-meshram

GilDev, 2024-07-18:
Any news on this? The driver still does not support using VDD as reference…