zephyrproject-rtos/zephyr #50897: Bluetooth: Support for generating HCI/btmon log while using the shell

Thalley, 2022-10-03:
**Is your enhancement proposal related to a problem? Please describe.**
Currently it is impossible to enable `CONFIG_BT_DEBUG_MONITOR_UART` and use the shell, as `CONFIG_UART_CONSOLE` needs to be disabled to output the `btmon` log. 

Since both the HCI/`btmon` log and the shell application are both very useful, it would be great if both could be used at the same time in some way. 

**Describe the solution you'd like**
A way where the BT shell can be used while also getting a HCI log. The HCI/`btmon` log does not need to be in the same window and could just be a file logged somewhere. 

**Describe alternatives you've considered**
Alternatively find a way to use the logging system to log all the HCI events, so that we get a non-`btmon` HCI log from Zephyr instead. This approach might actually be better, as we remove the dependency on `btmon` from `BlueZ` and can be used on all platforms. 

**Additional context**
N/A

jori-nordic, 2023-02-16:
can't this be addressed by either:
- using the supported RTT transport
- using a second UART peripheral (+ FTDI cable) for btmon

Thalley, 2024-02-12:
> can't this be addressed by either:
>     using the supported RTT transport

Technically yes, but I never got that to work. Will have to retry soon, as I'm heading to UPF with Zephyr and BT Mon logs are quite useful to have there :)

jrintaha, 2025-03-21:
I just recently tried this, and enabling BT_DEBUG_MONITOR_RTT in the shell app worked fine with Zephyr 4.1. Besides just the HCI traffic, the shell app does output its boot banner text etc. as btsnoop user logging events, but that's not really a problem.

Thalley, 2025-03-21:
Closing as it is possible

closed 2025-03-21 by Thalley