zephyrproject-rtos/zephyr #50922: Improve management and documentation of  API states

nashif, 2022-10-03:
## Introduction

Current API [overview of API](https://docs.zephyrproject.org/latest/develop/api/overview.html) is maintained manually and does not cover all public APIs provided by Zephyr. We need  to introduce a better solution for maintaining such data and increase the coverage and have all APIs listed and their state updated whenever they change.

There are other issues with such table and a limitation for using the API life-cycles states, especially when introducing breaking and non-breaking changes to a stable API and how such changes can be conveyed to the users of those stable APIs.

### Problem description

- State of API is maintained in a table and updated manually
- Table is not complete and does not cover all public APIs in Zephyr
- There is no way to capture the state of a stable API when it introduced breaking changes.

### Proposed change

Maintain The state of the API, when it is what introduced and other information related to the stability shall be maintained at the source level and in the API headers using doxygen. Where possible, existing doxygen commands shall be used, otherwise, we shall introduce aliases. 
Using the data from doxygen, an overview table of the APIs similar to what we have now shall be auto-generated to replace the existing table. Depending on the solution chosen, we shall consider using auto-generated table by doxygen itself.

## Detailed RFC

The information we have right now is as follows:
1. When API was introduced
2. State of API (experimental, unstable, stable)
3. When the API was changed

For 1., the existing doxygen command [@since](https://doxygen.nl/manual/commands.html#cmdsince) shall be used. Each API shall be extended with this information, either using the existing table or by looking into when the API was introduced. This information does provide any information about stability of an API.

For 2., we have multiple options, depending on how much information we want to provide and how useful we want to make the information available for each API, especially when it comes to API breaking changes:

### Using ``@version ``

Use available [@version](https://doxygen.nl/manual/commands.html#cmdversion) command which can be populated with the version of an API. For the version we shall use [semantic version](https://semver.org/) to maintain the level of stability of each API and to manage breaking and non-breaking changes to users. From the semver spec:

* Once a versioned package has been released, the contents of that version MUST NOT be modified. Any modifications MUST be released as a new version.
*  Major version zero (0.y.z) is for initial development. Anything MAY change at any time. The public API SHOULD NOT be considered stable.
*  Version 1.0.0 defines the public API. The way in which the version number is incremented after this release is dependent on this public API and how it changes.
* Patch version Z (x.y.Z | x > 0) MUST be incremented if only backwards compatible bug fixes are introduced. A bug fix is defined as an internal change that fixes incorrect behavior.
* Minor version Y (x.Y.z | x > 0) MUST be incremented if new, backwards compatible functionality is introduced to the public API. It MUST be incremented if any public API functionality is marked as deprecated. It MAY be incremented if substantial new functionality or improvements are introduced within the private code. It MAY include patch level changes. Patch version MUST be reset to 0 when minor version is incremented.

In doxgen, this will look like this:

```
 /**
  * @brief I3C Interface
  * @defgroup i3c_interface I3C Interface
  * @since 3.2.0
  * @version 0.1.0
  * @ingroup io_interfaces
  * @{
  */
```

for a stable API:

```
 /**
  * @brief I2C Interface
  * @defgroup i2c_interface I2C Interface
  * @since 1.0.0
  * @version 1.0.0
  * @ingroup io_interfaces
  * @{
  */
```

### Use Custom Command
Use a custom alias for API state (experimental, unstable, stable) in the folloing form ``@state{stable}``  or ``@api_state{stable}`` which will be part of the group definition of each API.

In doxgen, the alias can be defined as follows:

```status{1}=\xrefitem status \"API Status\" \"API Status\"\1"```

In an API, this would be used the following way:

```
 /**
  * @brief I3C Interface
  * @defgroup i3c_interface I3C Interface
  * @since 3.2.0
  * @status{experimental}
  * @ingroup io_interfaces
  * @{
  */
```

Doxygen in this case will generate a summary page with all the APIs and their status that can be used as is or can be converted to RST and be included as part of the documentation.

### Concerns and Unresolved Questions

This will require and overdue review and audit of all APIs and their states, so it will require some work.

## Alternatives

Now aware of any, the only alternative is to keep maintaining the API overview manually and extend with the missing APIs, but this had its own issues listed above.

henrikbrixandersen, 2022-10-04:
We need to establish the level of granularity for where to use `@since`. API functions? Definitions? Enumeration values? ...?

carlescufi, 2022-10-04:
Architecture WG meeting:

- @henrikbrixandersen mentions that other projects that use `@since` tend to do so in a per-API call basis
- The current table in the API overview page covers only a very small subset
- The objective is to help users of Zephyr that have no direct connection with upstream
- Proposal is to introduce `@since` in new APIs only
- We would tag all stable APIs are `@version 1.0.0` upon introduction of this proposal
- The alternative is to have `@status{experimental}` instead of a full version in the API

galak, 2022-11-30:
If we go w/semantic version we need to clarify how we know experimental and unstable.

dkalowsk, 2022-11-30:
Just was made aware of this issue.  The idea of versioning the APIs can be a really powerful tool for development.  Why the limit API versions to just Doxygen?   What if you could use the compiler to enforce and notify updates to end users?

As an end user I would want to see something like the following, which could make my life in tracking updates a lot easier.

```
include/drivers/i2c.h:

#include <zephyr/api_verison.h>

#define I2C_MAJOR_REV 3  // associated with major breaks in API
#define I2C_MINOR_REV 2  // associated with minor tweaks in the API

GENERATE_API_REV(I2C, I2C_MAJOR_REV, I2C_MINOR_REV)
```

This would generate a form of a tag that can be used later for checks on the code by the compiler.  If desired this could be expanded to include API state as well, although I'd imagine continuing to use the `99` version currently in use between releases would be a consistent mechanism.

With this kind of information, it is now possible to have an application by a downstream consumer to have in code:

```
main.c

#include <zephyr/kernel.h>
#include <zephyr/api_version.h>

BUILD_ASSERT(GENERATE_API_REV(I2C,3,2) == I2C_API_REV)
BUILD_ASSERT(GENERATE_API_REV(EEPROM,3,1) > EEPROM_API_REV)
BUILD_ASSERT(GENERATE_API_REV(GPIO,1,1) == GPIO_API_REV)

int main(void) 
{
    // do something
}
```

This method has a few benefits:

- It is relatively easy for incoming PRs that change the API version to quickly show that information
- It is pretty easy for the release manager to know when an API has changed and  needs an update/mention in the release notes.  This can even be done as a  test application that the CI can be used to check.
- Allows for clear definitions of which API versions the application was written to support, an important detail for developers taking over a project.
- It provides an immediate feedback to an end user potentially stopping a silent behavioral change when doing a kernel update and the release-notes either weren't updated or paid attention to.

It's been awhile since I've used Doxygen, I believe the preprocessor value can be extracted as well to update the original proposal as well.

gmarull, 2022-12-01:
> This would generate a form of a tag that can be used later for checks on the code by the compiler. If desired this could be expanded to include API state as well, although I'd imagine continuing to [...]

This is really a poor man dependency management. I think the idea of versioned APIs doesn't make much sense because Zephyr clients specify a Zephyr version, not API versions. As soon as you specify a Zephyr version, you're locked with the APIs that come with it, you have no choice. This would be possible if APIs were versioned packages/modules.

Also, from what I heard yesterday at TSC, even a typo fix could mean increasing patch version of an API. I have never seen any package, e.g. in Python, doing something like this. From time to time, they just release and increase version according to the changes made since last release. And often, you're lucky if they follow SemVer.

If the real motivation of versioned API is to improve user experience, I think we're looking at the wrong solution. I think we should focus on providing better API designs. We have many toy APIs, or cases like serial/uart or async where we haven't been able to provide a decent API for a long time. To me, this is what creates frustration because toy APIs tend to change frequently or don't cover a wide range of use cases, forcing people to go in their own. You can version them, but the pain of breaking changes or new replacement APIs will still be there, no matter what you do.

So to summarize, I think that the addition of `@since` is valuable. At least it is proven to work reasonably well in other languages like Python. For the rest, I'd focus on good API designs and provide good release notes/migration plans when needed.

PR #61994 (2024-03-19): Automatically generate API overview documentation
This PR should address #50922. It basically:

- ~~Generate \since tags from doxgen and git (it may not be perfect, but it's a fairly comprehensive start);~~
- Manually ~~fix~~ add some based on current information from doc/develop/api.overview.rst;
- Add \version (showing stability level) based on information from doc/develop/api.overview.rst;
- Use all new info (\since and \version) to generate the new API overview page, which is basically an improved version of doxygen own "Modules index" page.

More details of each step are on the commit mes [... cut]
closed 2024-03-19 by carlescufi