zephyrproject-rtos/zephyr #50928: Create a zephyr module/fork for CMSIS-DSP

yperess, 2022-10-03:
## Introduction

The current module/fork of CMSIS is out of date since CMSIS split off the DSP library to https://github.com/ARM-software/CMSIS-DSP. We should mirror that.

### Problem description

It is no longer possible to track ToT CMSIS without having a separate module for the DSP library and set the correct module dependencies

### Proposed change

Create a fork of the CMSIS-DSP repo and add it as a Zephyr module. Then add it as a dependency in Kconfig to the current CMSIS library and update said library to ToT.

stephanosio, 2022-10-03:
Related discussion: https://github.com/zephyrproject-rtos/zephyr/pull/50012#pullrequestreview-1126273027

carlescufi, 2022-10-03:
Makes sense. This is not quite a new module request but it'd still be nice to use the template:
https://docs.zephyrproject.org/latest/contribute/external.html#submission-and-review-process

yperess, 2022-10-20:
I want to follow-up real quick from yesterday's meeting. The current CMSIS fork at https://github.com/zephyrproject-rtos/cmsis is out of date since the DSP [directory](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP) is now empty. This means we can't update CMSIS until we've also created a fork of https://github.com/ARM-software/CMSIS-DSP.

I believe I understood correctly yesterday that this makes sense and we'll be creating that fork?

yperess, 2022-10-20:
Quick update on this after discussion with @keith-zephyr.

Since the CMSIS fork cannot be updated to ToT, we have 2 options from what I can tell:

1. We create the CMSIS-DSP fork and remap `CONFIG_CMSIS_DSP` to that module. This will be the most disruptive path. Downstream users will need to be notified that DSP moved and they'll need to update their git repos to include the new module.

2. We create the CMSIS-DSP fork _AND_ a new branch (`next`) in https://github.com/zephyrproject-rtos/cmsis for ToT https://github.com/ARM-software/CMSIS_5. Developers using https://github.com/zephyrproject-rtos/cmsis/blob/master/CMakeLists.txt will get a warning saying that this repo is deprecated and that CMSIS should use another branch. After a while, deveopers will use the new cmsis branch + the new cmsis-dsp fork and we'll be able to merge the cmsis branch into `master`. We should then deprecate the `next` branch telling developers to move back to `master`.

stephanosio, 2022-10-20:
> I want to follow-up real quick from yesterday's meeting. The current CMSIS fork at https://github.com/zephyrproject-rtos/cmsis is out of date since the DSP [...]

Migrating to the new CMSIS-DSP repository will involve upgrading to a newer CMSIS-DSP version (updating build scripts, Kconfig, ...) and updating all the Zephyr-side CMSIS-DSP tests to work with the new CMSIS-DSP version. This is a huge task and requires very high level of expertise on the CMSIS-DSP library.

Since #51059 does not require upgrading to a newer CMSIS-DSP version and/or using a dedicated CMSIS-DSP repository/module, I would suggest that we get #51059 in first, without this new dedicated CMSIS-DSP repository.

yperess, 2022-10-20:
Agreed, this is more tangental.

stephanosio, 2023-04-13:
> @stephanosio @yperess what is the status of this?

@nashif https://github.com/zephyrproject-rtos/zephyr/issues/50928#issuecomment-1285936904

> Migrating to the new CMSIS-DSP repository will involve upgrading to a newer CMSIS-DSP version (updating build scripts, Kconfig, ...) and updating all the Zephyr-side CMSIS-DSP tests to work with the [...]

Nothing happened since then, mainly because there is no immediate need/user request for the CMSIS-DSP version upgrade.

XenuIsWatching, 2023-04-16:
I've actually now got an immediate need for it. I did get a fork of cmsis-dsp, cmsis-nn, and zephyr working together.
https://github.com/XenuIsWatching/CMSIS-DSP/tree/zephyr
https://github.com/XenuIsWatching/CMSIS-NN/tree/zephyr
https://github.com/XenuIsWatching/zephyr/tree/cmsis-dsp-nn-module

So far I got all the cmsis-dsp tests passing and most of the cmsis-nn tests are passing (some are giving different results)

A few hacks had to be done here and there to get it working
1) remove cmake references to cmsis-dsp and cmsis-nn in https://github.com/zephyrproject-rtos/cmsis
2) put a placeholder in west.yml for cmsis-dsp and cmsis-nn modules

I've got a few todos here and there
1) understand why some cmsis-nn are giving different results for the cmsis-nn tests
2) understand if I really understand cmake with all the find and replace I did in the modules

stephanosio, 2023-04-17:
Since it is not very clear from reading the comments in this issue, just a reminder that the "Vote to Approve creating a zephyr module/fork for CMSIS-DSP, as presented in https://github.com/zephyrproject-rtos/zephyr/issues/50928" has been approved by the TSC on 2022-10-25.

stephanosio, 2023-05-02:
The Zephyr CMSIS-DSP module repository has been created based on the content of the upstream [CMSIS-DSP repository](https://github.com/ARM-software/CMSIS-DSP): https://github.com/zephyrproject-rtos/cmsis-dsp

closed 2023-05-02 by stephanosio