zephyrproject-rtos/zephyr #50940: logging.log_output_ts64 fails on qemu_arc_hs5x

stephanosio, 2022-10-04:
**Describe the bug**

`logging.log_output_ts64` test fails on `qemu_arc_hs5x`

**To Reproduce**

```
twister -v -N -M -p qemu_arc_hs5x -s tests/subsys/logging/log_output/logging.log_output_ts64
```

**Expected behavior**

Test passes

**Logs and console output**

```
Running TESTSUITE test_log_output
===================================================================
START - test_colors
 PASS - test_colors in 0.002 seconds
===================================================================
START - test_format_ts
[00:00:01.000,000] domain/src: test

 PASS - test_format_ts in 0.001 seconds
===================================================================
START - test_level_flag
 PASS - test_level_flag in 0.001 seconds
===================================================================
START - test_levels
 PASS - test_levels in 0.002 seconds
===================================================================
START - test_no_flags
 PASS - test_no_flags in 0.001 seconds
===================================================================
START - test_no_flags_dname

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/logging/log_output/src/log_output_test.c:91: test_log_output_test_no_flags_dname: strcmp(exp_str, mock_buffer) not equal to 0

 FAIL - test_no_flags_dname in 0.002 seconds
===================================================================
START - test_raw
 PASS - test_raw in 0.001 seconds
===================================================================
START - test_ts_flag
 PASS - test_ts_flag in 0.001 seconds
===================================================================
START - test_ts_to_us
 PASS - test_ts_to_us in 0.001 seconds
===================================================================
TESTSUITE test_log_output failed.

------ TESTSUITE SUMMARY START ------

SUITE FAIL -  88.89% [test_log_output]: pass = 8, fail = 1, skip = 0, total = 9 duration = 0.012 seconds
 - PASS - [test_log_output.test_colors] duration = 0.002 seconds
 - PASS - [test_log_output.test_format_ts] duration = 0.001 seconds
 - PASS - [test_log_output.test_level_flag] duration = 0.001 seconds
 - PASS - [test_log_output.test_levels] duration = 0.002 seconds
 - PASS - [test_log_output.test_no_flags] duration = 0.001 seconds
 - FAIL - [test_log_output.test_no_flags_dname] duration = 0.002 seconds
 - PASS - [test_log_output.test_raw] duration = 0.001 seconds
 - PASS - [test_log_output.test_ts_flag] duration = 0.001 seconds
 - PASS - [test_log_output.test_ts_to_us] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
RunID: f82677ace3ccc6bffc73d68bdff4c68e
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA: cf3b10a2c015631c253162b4d4a09f118d9df12c

**Additional context**

https://github.com/zephyrproject-rtos/zephyr/actions/runs/3180928376/jobs/5185086458#step:11:1392

stephanosio, 2022-10-04:
cc @nordic-krch @ruuddw

nordic-krch, 2022-10-06:
This is quite strange. Anything added to the test make the test pass but increasing stack does not. I also tried to bisect and got to a commit which break the test: 003ee34593 which is not related at all but starts to break the test on my setup consistently.

evgeniy-paltsev, 2022-10-06:
Hmm, I've tried it on proposed commit (https://github.com/zephyrproject-rtos/zephyr/commit/cf3b10a2c015631c253162b4d4a09f118d9df12c) and on current main (40a6f08d3c) and I can't reproduce the issue locally.

E.g: environment is same as proposed:

OS: Ubuntu 20.04
Toolchain: Zephyr SDK 0.15.0
Commit SHA: https://github.com/zephyrproject-rtos/zephyr/commit/cf3b10a2c015631c253162b4d4a09f118d9df12c

```
*** Booting Zephyr OS build zephyr-v3.1.0-5889-gcf3b10a2c015  ***
Running TESTSUITE test_log_output
===================================================================
START - test_colors
 PASS - test_colors in 0.002 seconds
===================================================================
START - test_format_ts
[00:00:01.000,000] domain/src: test
 PASS - test_format_ts in 0.001 seconds
===================================================================
START - test_level_flag
 PASS - test_level_flag in 0.001 seconds
===================================================================
START - test_levels
 PASS - test_levels in 0.002 seconds
===================================================================
START - test_no_flags
 PASS - test_no_flags in 0.001 seconds
===================================================================
START - test_no_flags_dname
 PASS - test_no_flags_dname in 0.001 seconds
===================================================================
START - test_raw
 PASS - test_raw in 0.001 seconds
===================================================================
START - test_ts_flag
 PASS - test_ts_flag in 0.001 seconds
===================================================================
START - test_ts_to_us
 PASS - test_ts_to_us in 0.001 seconds
===================================================================
TESTSUITE test_log_output succeeded

------ TESTSUITE SUMMARY START ------

SUITE PASS - 100.00% [test_log_output]: pass = 9, fail = 0, skip = 0, total = 9 duration = 0.011 seconds
 - PASS - [test_log_output.test_colors] duration = 0.002 seconds
 - PASS - [test_log_output.test_format_ts] duration = 0.001 seconds
 - PASS - [test_log_output.test_level_flag] duration = 0.001 seconds
 - PASS - [test_log_output.test_levels] duration = 0.002 seconds
 - PASS - [test_log_output.test_no_flags] duration = 0.001 seconds
 - PASS - [test_log_output.test_no_flags_dname] duration = 0.001 seconds
 - PASS - [test_log_output.test_raw] duration = 0.001 seconds
 - PASS - [test_log_output.test_ts_flag] duration = 0.001 seconds
 - PASS - [test_log_output.test_ts_to_us] duration = 0.001 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
````

carlocaione, 2022-10-12:
Opened #51193 to disable this until a fix is found.

stephanosio, 2022-12-12:
This test seems to be passing on `qemu_arc_hs5x` now.

@evgeniy-paltsev @nordic-krch Should https://github.com/zephyrproject-rtos/zephyr/commit/dfbfb3ff67f1caaaf86a72335ff751260d416fc5 be reverted now?

evgeniy-paltsev, 2023-01-13:
Hi @stephanosio,

sorry for late reply. Indeed the test passes (in both upstream and internal verification).

There were multiple fixes for logging for ARC recently, though note that they were not directly addressing this issue.

Thanks for reverting test disabling.

PR #53595 (2023-01-10): Revert "test: log_output: Exclude qemu_arc_hs5x"
This reverts commit dfbfb3ff67f1caaaf86a72335ff751260d416fc5.

Fixes #50940.