zephyrproject-rtos/zephyr #50941: sample.logger.syst.catalog.deferred_cpp fails on qemu_cortex_m0

stephanosio, 2022-10-04:
**Describe the bug**

`sample.logger.syst.catalog.deferred_cpp` test fails on `qemu_cortex_m0`.

**To Reproduce**

```
scripts -v -N -M -p qemu_cortex_m0 -s samples/subsys/logging/syst/sample.logger.syst.catalog.deferred_cpp
```

**Expected behavior**

Test passes.

**Logs and console output**

```
INFO    - 1173/1173 qemu_cortex_m0            samples/subsys/logging/syst/sample.logger.syst.catalog.deferred_cpp FAILED unexpected eof (qemu 0.152s)
INFO    - /__w/zephyr/zephyr/twister-out/qemu_cortex_m0/samples/subsys/logging/syst/sample.logger.syst.catalog.deferred_cpp/handler.log
ERROR   - SYS-T RAW DATA: 020A7F0B480000000000000000002A2A2A20426F6F74696E67205A6570687972204F53206275696C64202573202573202A2A2A0A007A65706879722D76332E322E302D3137362D676366336231306132633031350000
SYS-T RAW DATA: 230A0601040000000000000000002F6E0000
SYS-T RAW DATA: 330A060104000000000000000000166E0000
SYS-T RAW DATA: 430A060104000000000000000000006E0000
SYS-T RAW DATA: 730A060104000100000000000000E96D0000
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Toolchain: Zephyr SDK 0.15.0
 - Commit SHA: cf3b10a2c015631c253162b4d4a09f118d9df12c

**Additional context**

https://github.com/zephyrproject-rtos/zephyr/actions/runs/3180928376/jobs/5185086696#step:11:1449

stephanosio, 2022-10-04:
cc @nordic-krch @dcpleung @microbuilder

dcpleung, 2022-10-04:
Here is the error message:

```
SYS-T RAW DATA: 020A7F0B460000000000000000002A2A2A20426F6F74696E67205A6570687972204F53206275696C64202573202573202A2A2A0A007A65706879722D76332E322E302D342D676635353837353532626238380000
SYS-T RAW DATA: 230A0601040000000000000000006B6E0000
SYS-T RAW DATA: 330A060104000000000000000000526E0000
SYS-T RAW DATA: 430A0601040000000000000000003C6E0000
SYS-T RAW DATA: 730A060104000100000000000000256E0000
qemu: fatal: Lockup: can't escalate 3 to HardFault (current priority -1)

R00=200008b2 R01=00000001 R02=00000007 R03=20000284
R04=200008ae R05=00000005 R06=00000001 R07=20000cae
R08=00000000 R09=00000000 R10=00000000 R11=00000000
R12=00000000 R13=20002360 R14=00002109 R15=00002110
XPSR=81000003 N--- T handler
FPSCR: 00000000
FAILED: zephyr/CMakeFiles/run_qemu /tmp/build/zephyr/CMakeFiles/run_qemu 
```

After bisecting, this is the commit which is causing the issue:

```
commit f5587552bb88fe6e57f3be1b30b8fb93fd3e0c32 (HEAD)
Author: Christopher Friedt <cfriedt@fb.com>
Date:   Sat Sep 24 13:39:42 2022 -0400

    logging: move backends to subdirectory
    
    This is mainly to reduce clutter in `subsys/logging`, but also to make
    backend management slightly easier.
    
    Signed-off-by: Christopher Friedt <cfriedt@fb.com>
```

which should not have caused the issue.

However, when I tried to use `CONFIG_NO_OPTIMIZATIONS`, it worked without issue. `CONFIG_DEBUG_OPTIMIZATIONS` and `CONFIG_SPEED_OPTIMIZATIONS` also worked fine. The error only occurs with `CONFIG_SIZE_OPTIMIZATIONS`.

This also doesn't seem to be stack size related. I have tried to enlarge the main stack size, and the log processing thread stack size. The error was still there.

dcpleung, 2022-10-04:
While debugging, the log line https://github.com/zephyrproject-rtos/zephyr/blob/d823f88e3b3478050f16431526ebdc550152a0af/samples/subsys/logging/syst/src/main.c#L52

has integer as the first argument.

When it comes to outputting the log, https://github.com/zephyrproject-rtos/zephyr/blob/7c3bf208137f5f469b3a27af0946f6ae00ac5143/subsys/logging/log_output_syst.c#L677

where `arg_tag` should be 0x05 which corresponds to `CBPRINTF_PACKAGE_ARG_TYPE_INT`, However, when I tried to step through the code, the switch matches `CBPRINTF_PACKAGE_ARG_TYPE_PTR_VOID` instead. Either GDB is playing tricks on me... or there is some toolchain issue.

dcpleung, 2022-10-06:
@stephanosio have you seen something like this before in Zephyr's ARM toolchain?

gmarull, 2022-10-07:
changed priority to high since it is impacting CI, see https://github.com/zephyrproject-rtos/zephyr/actions/runs/3204173755/jobs/5235181704

stephanosio, 2022-10-07:
> Should this be new issue or part of this one?

That looks like a completely different issue, so should be a new issue.

dcpleung, 2022-10-09:
Workaround in https://github.com/zephyrproject-rtos/zephyr/pull/51104 to unblock CI.

jhedberg, 2023-08-29:
Any update on this one?

microbuilder, 2023-09-07:
This has been fixed in QEMU, where a bug was found in the underlying timer representation used by this target.

```bash
$ scripts/twister -v -N -M -p qemu_cortex_m0 -s samples/subsys/logging/syst/sample.logger.syst.catalog.deferred_cpp
Renaming output directory to /Users/Kevin.Townsend/adi/zephyr/zephyr/twister-out.1
INFO    - Using Ninja..
INFO    - Zephyr version: zephyr-v3.4.0-3331-g2f05af891de2
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /Users/Kevin.Townsend/adi/zephyr/zephyr/twister-out/testplan.json
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue

INFO    - 1 test scenarios (1 test instances) selected, 0 configurations skipped (0 by static filter, 0 at runtime).
INFO    - 1 of 1 test configurations passed (100.00%), 0 failed, 0 errored, 0 skipped with 0 warnings in 40.25 seconds
INFO    - In total 1 test cases were executed, 0 skipped on 1 out of total 608 platforms (0.16%)
INFO    - 1 test configurations executed on platforms, 0 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /Users/Kevin.Townsend/adi/zephyr/zephyr/twister-out/twister.json
INFO    - Writing xunit report /Users/Kevin.Townsend/adi/zephyr/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /Users/Kevin.Townsend/adi/zephyr/zephyr/twister-out/twister_report.xml...
INFO    - Run completed
```

closed 2023-09-07 by microbuilder